alexpevzner/sane-airscan #1: HP DeskJet 2540 does not scan with the sane-airscan backend

tillkamppeter, 2019-12-26:
I have the HP DeskJet 2540 multi-function device which prints perfectly via AirPrint, so it is also supposed to scan via AirScan. I tried with the sane-airscan without success.
The output of avahi-discover for this device is
```
Service data for service 'Deskjet 2540 series [E7DDC7]' of type '_scanner._tcp' in domain 'local' on 3.0:
	Host HP5CB901E7DDC7.local (192.168.43.161), port 8080, TXT data: ['flatbed=T', 'button=T', 'UUID=1c852a4d-b800-1f08-abcd-5cb901e7ddc7', 'note=', 'adminurl=http://HP5CB901E7DDC7.local.', 'mdl=Deskjet 2540 series', 'mfg=HP', 'ty=Deskjet 2540 series', 'txtvers=1']
```
so its eSCL URI should be: `http://192.168.43.161:8080/eSCL`.
First, I tried simply
```
scanimage -L
```
and got no result of the airscan backend.
I added the line
```
"HP DeskJet 2540" = http://192.168.43.161:8080/eSCL
```
to the `[devices]` section of `/etc/sane.d/airscan.conf` and removed the read permissions of `/etc/sane.d/dll.d/hplip` (to avoid any interfrences of the "hpaio" SANE backend).
I also activated debug mode in `/etc/sane.d/airscan.conf` via
```
[debug]
trace = ~/tmp/airscan
enable = true
```
Now I run
```
scanimage -L
```
I get the answer
```
device `airscan:HP DeskJet 2540' is a Unknown xxxx eSCL network scanner
```
So the device gets found by the airscan backend.
The full debug output is
```
$ scanimage -L
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
statically adding: "HP DeskJet 2540"
"HP DeskJet 2540": device created
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080//eSCL/
  2: http://192.168.43.161:8080//eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities: OK
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
MDNS: initial scan finished
device `airscan:HP DeskJet 2540' is a Unknown xxxx eSCL network scanner
sane_exit() called
"HP DeskJet 2540": removed from device table
"HP DeskJet 2540": device destroyed
sane_exit(): OK
$
```
Now I run
```
simple-scan
```
wait for it to recognize the scanner and click the green "Scan" button. A window pops up saying "Failed to scan", "Unable to start scan". On the terminal I get
```
$ simple-scan
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
statically adding: "HP DeskJet 2540"
"HP DeskJet 2540": device created
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080//eSCL/
  2: http://192.168.43.161:8080//eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities: OK
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
MDNS: initial scan finished
"HP DeskJet 2540": sane_open("HP DeskJet 2540"): Success
"HP DeskJet 2540": sane_start()
"HP DeskJet 2540": state=SCAN_STARTED
"HP DeskJet 2540": state=SCAN_REQUESTING
"HP DeskJet 2540": HTTP POST http://192.168.43.161:8080/eSCL/ScanJobs
"HP DeskJet 2540": HTTP POST http://192.168.43.161:8080/eSCL/ScanJobs: Conflict
"HP DeskJet 2540": state=SCAN_CHECK_STATUS
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerStatus
"HP DeskJet 2540": HTTP GET http://192.168.43.161:8080/eSCL/ScannerStatus: OK
"HP DeskJet 2540": JOB status=Device busy
"HP DeskJet 2540": state=SCAN_DONE
"HP DeskJet 2540": state=SCAN_IDLE
"HP DeskJet 2540": sane_start(): Device busy
"HP DeskJet 2540": sane_close()
"HP DeskJet 2540": cancel requested
```
Closing the pop-up and simple-scan itself gives these additional lines on the terminal:
```
sane_exit() called
"HP DeskJet 2540": removed from device table
"HP DeskJet 2540": device destroyed
sane_exit(): OK
$
```
For me it looks like that the capabilities poll is done with a broken URI (duplicate slash). Correct URI for capabilities:
```
http://192.168.43.161:8080/eSCL/ScannerCapabilities
```
Wrong URI for capabilities:
```
http://192.168.43.161:8080//eSCL/ScannerCapabilities
```
In a browser the former shows the capabilities in XML, the latter errors.

tillkamppeter, 2019-12-26:
Log files of the airscan backend:
[scanimage-Deskjet-2540-series-[E7DDC7].log.txt](https://github.com/alexpevzner/sane-airscan/files/4003356/scanimage-Deskjet-2540-series-.E7DDC7.log.txt)
[scanimage-HP-DeskJet-2540.log.txt](https://github.com/alexpevzner/sane-airscan/files/4003357/scanimage-HP-DeskJet-2540.log.txt)
[simple-scan-Deskjet-2540-series-[E7DDC7].log.txt](https://github.com/alexpevzner/sane-airscan/files/4003358/simple-scan-Deskjet-2540-series-.E7DDC7.log.txt)
[simple-scan-HP-DeskJet-2540.log.txt](https://github.com/alexpevzner/sane-airscan/files/4003359/simple-scan-HP-DeskJet-2540.log.txt)

tillkamppeter, 2019-12-26:
The scanner capabilities can be polled from the DeskJet 2540 without any problems. Simply running
```
wget http://192.168.43.161:8080/eSCL/ScannerCapabilities
```
gives an XML file with the capabilities: 
[ScannerCapabilities.txt](https://github.com/alexpevzner/sane-airscan/files/4003365/ScannerCapabilities.txt)

tillkamppeter, 2019-12-26:
It looks like that I get the same result when I access the scanner via USB, using [ippusbxd](https://github.com/OpenPrinting/ippusbxd/). Note that here the scanner will not get auto-discovered via DNS-SD as ippusbxd [reports the devices only as printers](https://github.com/OpenPrinting/ippusbxd/issues/11), so I add the following line to the `[devices]` section of `/etc/sane.d/airscan.conf` (and commented out the others):
```
"eSCL Scanner on USB via ippusbxd" = http://localhost:60000/eSCL
```
Then I run `scanimage -L` again and get
```
$ scanimage -L
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
statically adding: "eSCL Scanner on USB via ippusbxd"
"eSCL Scanner on USB via ippusbxd": device created
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080//eSCL/
  2: http://192.168.43.161:8080//eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities: OK
MDNS: initial scan finished
device `airscan:eSCL Scanner on USB via ippusbxd' is a Unknown xxxx eSCL network scanner
sane_exit() called
"eSCL Scanner on USB via ippusbxd": removed from device table
"eSCL Scanner on USB via ippusbxd": device destroyed
sane_exit(): OK
$
```
It contains the line
```
device `airscan:eSCL Scanner on USB via ippusbxd' is a Unknown xxxx eSCL network scanner
```
so the scanner was recognized. 
Strange thing here is that even if the scanner is discovered through USB `scanimage -L` also accesses the scanner through the network.
My assumption is now the following:
The airscan backend tries to find scanners BOTH by DNS-SD and by lines in the `[devices]` section. On the DNS-SD attempt it always fails as it tries to poll the capabilities with URIs with a duplicate slash before "eSCL" (`...//eSCL/...`), so it never succeeds to poll capabilities and so only can find a scanner by the `[devices]` lines. By the `[devices]` lines it correctly discovers my scanner both via the network and via USB with ippusbxd.
Now we run "simple-scan" as before, clicking "Scan" as soon as it sees our scanner (this time via USB) and then closing the rror pop-up and the application itself.
We get on the terminal:
```
$ simple-scan
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
statically adding: "eSCL Scanner on USB via ippusbxd"
"eSCL Scanner on USB via ippusbxd": device created
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: initial scan finished
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080//eSCL/
  2: http://192.168.43.161:8080//eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080//eSCL/ScannerCapabilities: Not Found
"Deskjet 2540 series [E7DDC7]": ScannerCapabilities query: Not Found
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities: OK
"eSCL Scanner on USB via ippusbxd": sane_open("eSCL Scanner on USB via ippusbxd"): Success
"eSCL Scanner on USB via ippusbxd": sane_start()
"eSCL Scanner on USB via ippusbxd": state=SCAN_STARTED
"eSCL Scanner on USB via ippusbxd": state=SCAN_REQUESTING
"eSCL Scanner on USB via ippusbxd": HTTP POST http://localhost:60000/eSCL/ScanJobs
"eSCL Scanner on USB via ippusbxd": HTTP POST http://localhost:60000/eSCL/ScanJobs: Conflict
"eSCL Scanner on USB via ippusbxd": state=SCAN_CHECK_STATUS
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerStatus
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerStatus: OK
"eSCL Scanner on USB via ippusbxd": JOB status=Device busy
"eSCL Scanner on USB via ippusbxd": state=SCAN_DONE
"eSCL Scanner on USB via ippusbxd": state=SCAN_IDLE
"eSCL Scanner on USB via ippusbxd": sane_start(): Device busy
"eSCL Scanner on USB via ippusbxd": sane_close()
sane_exit() called
"eSCL Scanner on USB via ippusbxd": removed from device table
"eSCL Scanner on USB via ippusbxd": device destroyed
sane_exit(): OK
$
```
I hope that this helps to make this scanner working.

alexpevzner, 2019-12-26:
> For me it looks like that the capabilities poll is done with a broken 
> URI (duplicate slash). Correct URI for capabilities:

Looks like so. Fixed.

Seems that HP announces "rs" in the mDNS TXT record with leading slash, 
unlike Kyocera.

-- 

	Wishes, Alexander Pevzner (pzz@apevzner.com)

alexpevzner, 2019-12-26:
> My assumption is now the following:
> The airscan backend tries to find scanners BOTH by DNS-SD and by lines
> in the |[devices]| section. On the DNS-SD attempt it always fails as it
> tries to poll the [...]

Yes, I agree.

> Now we run "simple-scan" as before, clicking "Scan" as soon as it sees 
> our scanner (this time via USB) and then closing the rror pop-up and the 
> application itself.

I've noticed it in logs you've sent before and fixed one small thing: 
probably I should not set "duplex" to "false" in scan request when 
scanning from Platen, just skip this parameter.

Please try updated version. If it still doesn't work, please send me a 
full protocol trace, not only console log.

-- 

	Wishes, Alexander Pevzner (pzz@apevzner.com)

tillkamppeter, 2019-12-26:
First, I had attached the debug logs on the first test when I had scanned via the netwotk. After the USB test I had to leave quickly and was not able any more to add the debug logs.
Now I have built your current GIT snapshot and the duplicate slash problem is solved. My scanner (accessed via network) is discovered correctly via DNS-SD. With all lines in the `[devices]` section commented out I get one line as answer to `scanimage -L`:
```
device `airscan:Deskjet 2540 series [E7DDC7]' is a Unknown xxxx eSCL network scanner
```
simple-scan does not work though. Here is the console output of the usual session (click "Scan", close error pop-up, close app):
```
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: initial scan finished
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080/eSCL/
  2: http://192.168.43.161:8080/eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities: OK
"Deskjet 2540 series [E7DDC7]": sane_open("Deskjet 2540 series [E7DDC7]"): Success
"Deskjet 2540 series [E7DDC7]": sane_start()
"Deskjet 2540 series [E7DDC7]": state=SCAN_STARTED
"Deskjet 2540 series [E7DDC7]": state=SCAN_REQUESTING
"Deskjet 2540 series [E7DDC7]": HTTP POST http://192.168.43.161:8080/eSCL/ScanJobs
"Deskjet 2540 series [E7DDC7]": HTTP POST http://192.168.43.161:8080/eSCL/ScanJobs: Conflict
"Deskjet 2540 series [E7DDC7]": state=SCAN_CHECK_STATUS
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerStatus
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerStatus: OK
"Deskjet 2540 series [E7DDC7]": JOB status=Device busy
"Deskjet 2540 series [E7DDC7]": state=SCAN_DONE
"Deskjet 2540 series [E7DDC7]": state=SCAN_IDLE
"Deskjet 2540 series [E7DDC7]": sane_start(): Device busy
"Deskjet 2540 series [E7DDC7]": sane_close()
sane_exit() called
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
sane_exit(): OK
```
Debug log of this session:
[simple-scan-Deskjet-2540-series-[E7DDC7].log.txt](https://github.com/alexpevzner/sane-airscan/files/4003597/simple-scan-Deskjet-2540-series-.E7DDC7.log.txt)

tillkamppeter, 2019-12-26:
I have one additional feature request for the config file of the backend: There should be a way to turn off DNS-SD discovery, for example if you have configured a special configuration in the `[devices]` section and want to be sure that clients use it (and not the DNS-SD-suggested network connection).

tillkamppeter, 2019-12-26:
Scanning via USB with ippusbxd leads to the same result. Here is the console output of simple-scan:
```
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
statically adding: "eSCL Scanner on USB via ippusbxd"
"eSCL Scanner on USB via ippusbxd": device created
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv6)
MDNS: found "Deskjet 2540 series [E7DDC7]" (ipv4)
MDNS: "Deskjet 2540 series [E7DDC7]" addresses resolved:
  1: http://192.168.43.161:8080/eSCL/
  2: http://192.168.43.161:8080/eSCL/
statically adding: "Deskjet 2540 series [E7DDC7]"
"Deskjet 2540 series [E7DDC7]": device created
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerCapabilities: OK
"Deskjet 2540 series [E7DDC7]": HTTP GET http://192.168.43.161:8080/eSCL/ScannerCapabilities: OK
MDNS: initial scan finished
"eSCL Scanner on USB via ippusbxd": sane_open("eSCL Scanner on USB via ippusbxd"): Success
"eSCL Scanner on USB via ippusbxd": sane_start()
"eSCL Scanner on USB via ippusbxd": state=SCAN_STARTED
"eSCL Scanner on USB via ippusbxd": state=SCAN_REQUESTING
"eSCL Scanner on USB via ippusbxd": HTTP POST http://localhost:60000/eSCL/ScanJobs
"eSCL Scanner on USB via ippusbxd": HTTP POST http://localhost:60000/eSCL/ScanJobs: OK
"eSCL Scanner on USB via ippusbxd": state=SCAN_CHECK_STATUS
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerStatus
"eSCL Scanner on USB via ippusbxd": HTTP GET http://localhost:60000/eSCL/ScannerStatus: OK
"eSCL Scanner on USB via ippusbxd": JOB status=Device busy
"eSCL Scanner on USB via ippusbxd": state=SCAN_DONE
"eSCL Scanner on USB via ippusbxd": state=SCAN_IDLE
"eSCL Scanner on USB via ippusbxd": sane_start(): Device busy
"eSCL Scanner on USB via ippusbxd": sane_close()
sane_exit() called
"eSCL Scanner on USB via ippusbxd": removed from device table
"eSCL Scanner on USB via ippusbxd": device destroyed
"Deskjet 2540 series [E7DDC7]": removed from device table
"Deskjet 2540 series [E7DDC7]": device destroyed
sane_exit(): OK
```
and here is the debug log: 
[simple-scan-eSCL-Scanner-on-USB-via-ippusbxd.log.txt](https://github.com/alexpevzner/sane-airscan/files/4003630/simple-scan-eSCL-Scanner-on-USB-via-ippusbxd.log.txt)

tillkamppeter, 2019-12-27:
By the way, the "escl" backend (from @Ordissimo) actually scans on my device. I tested this only via USB/ippusbxd (data manually entered in config file) as Avahi discovery of the device on the network via the "escl" backend did not work.

alexpevzner, 2019-12-27:
Hi Till,
1. If you statically add a device using the same device name as Avahi returns, the static configuration will "win", and Avahi-discovered device under the same name will be ignored. Is it enough to satisfy your feature request?

2. Your last log, where scan via USB fails, looks extremely strange and different from previous failures. Backend first GETs ScannerCapabilities and correctly receives it. Then backend POSTs the scan request, and receives ScannerCapabilities again, which should not happen. Can it be that something stuck on ippusbxd?

tillkamppeter, 2019-12-27:
For (1) this would help if you have a device which gets discovered but you want to connect differently, for example via ippusbxd instead of by network. It does not serve if you for example have tons of scanners in the network and you are only interested in one, as you cannot suppress unwished scanners. Therefore there should be an option to explicitly turn off Avahi discovery, so that only your manually configured scanners get used.
About (2), I have tried @Ordissimo's backend and I can actually scan with it, also via USB with ippusbxd. ippusbxd has very verbose debug logging, so you could have a look on what is actually happening and why it works with the other backend.

alexpevzner, 2019-12-27:
On 12/27/19 11:43 PM, Till Kamppeter wrote:
> For (1) this would help if you have a device which gets discovered but
> you want to connect differently, for example via ippusbxd instead of by
> network. It does not serve if you for example have [...]

I'm now talking about the regular user perspective, not about 
developer's convenience.

I understand the problem your mentioned, and I had an idea to support a 
syntax like this, to explicitly disable particular device:

[devices]
"HP Bla-Bla-Bla series" = disable

It may be much more convenient for the normal user, that to disable auto 
discovery and then manually add all devices he or she is interested in.

Figuring out a correct eSCL URL can be quite tricky for a regular user, 
because different scanners use different TCP ports, and, in theory, URL 
path may be something different that "/eSCL" (I've never seen it on 
hardware devices, but AirSane uses path prefix that looks like UUID).

What do you think?

-- 

	Wishes, Alexander Pevzner (pzz@apevzner.com)

debiantriage, 2019-12-27:
>   I have the HP DeskJet 2540 multi-function device which prints perfectly via AirPrint, so it is also supposed to scan via AirScan.

Equating printing via AirPrint and scanning via AirScan is not quite how I would look at it. There are devices that are AirPrint-capable that do not offer an eSCL service; the Officejet 6500 E710n-z is one. I find that _uscan._tcp is a better indicator than _scanner._tcp.

-- 
Brian.

debiantriage, 2019-12-27:
>   Seems that HP announces "rs" in the mDNS TXT record with leading slash, unlike Kyocera.

It depends on the HP device. Some have `rs=eSCL`, others `rs=/eSCL `.

-- 
Brian.

alexpevzner, 2019-12-27:
Is there any devices that support Microsoft's WSD but don't support eSCL?

WSD is quite similar to eSCL, though uses very different discovery method, not currently supported by Linux. Unlike eSCL, it is even documented (well, in Microsoft style of writing documentation).

I'm considering adding WSD support, but not sure does it make a lot of sense.

alexpevzner, 2019-12-27:
I have added
    [devices]
    "some device" = disable
to explicitly disable particular annoying device

tillkamppeter, 2019-12-28:
@debiantriage, For me _uscan._tcp also looks like the better indicator, as its TXT record content makes more sense for eSCL scanning. Especially it contains "rs=(/)eSCL".

tillkamppeter, 2019-12-28:
@alexpevzner, thank you for adding the option to disable certain scanner discoveries, but I suggest to add another directive to disable DNS-SD discovery altogether for those users who get hundreds of scanners and only one is relevant for them.

alexpevzner, 2019-12-28:
@tillkamppeter, I've added options to configuration file:
`[options]
discovery = enable | disable ; default is enable
`
Actually, we had to create another issue for this purpose :-)

tillkamppeter, 2019-12-28:
@alexpevzner, thank you very much.

alexpevzner, 2019-12-29:
@tillkamppeter , thank a lot for your help fighting with this device!

The problem was in HP's XML scanner.

The following XML works (all on single long line):
`<scan:ScanSettings xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03">`

And the following doesn't (text formatted as 3 lines):
`<scan:ScannerCapabilities`
`xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03"`
`xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm">`

closed 2019-12-29 by alexpevzner