alexpevzner/sane-airscan #2: An HP ENVY 4500

debiantriage, 2020-01-01:
This device works OOTB with the airscan backend and xsane. Functions tested were

- Resolution.
- Preview.
- Colour/Grey.
- Image type.
- Selection of a portion of a preview image.

`scanimage -L`
`device 'airscan:ENVY4500' is a Airscan ENVY4500 eSCL network scanner`

(A minor typo: "a" should be "an").

What would the output have been if there was more than a single ENVY 4500 on the network?

Having to use an HP non-free plugin and non-free Epson scanner software should now be a thing of the past. I do not have access to a recent Canon device but the backend hopefully provides a way of replacing the limited Scangearmp2 from Canon. Brother do not seem to use _uscan._tcp on their machines, so airscan may not be of use there.

Cheers,

Brian.

alexpevzner, 2020-01-01:
Hi, Brian,

thank you for good news. This is HP ENVY 4500, correct?

> |scanimage -L|
> |device 'airscan:ENVY4500' is a Airscan ENVY4500 eSCL network scanner|
> 
> (A minor typo: "a" should be "an").

This "a" comes from the scanimage program, not from the backend.

> What would the output have been if there was more than a single ENVY 
> 4500 on the network?

Till Kamppeter convinced me to use device network name here instead of 
hardware model. Device network name usually defaults to hardware model, 
but must be unique across the network (this is a DNS-SD protocol 
requirement), so if there are multiple devices, either somebody will 
configure them to use different names, or devices will do it automatically.

> Having to use an HP non-free plugin and non-free Epson scanner software
> should now be a thing of the past. I do not have access to a recent
> Canon device but the backend hopefully provides a way of [...]

For HP there is at least a non-free plugin. For my Kyocera nothing 
exist. Only scan to e-mail, extremely inconvenient.

What does Brother use? Does it support WSD, or only some proprietary 
protocol?

-- 

	Wishes, Alexander Pevzner (pzz@apevzner.com)

alexpevzner, 2020-01-01:
Device added to the list, so I guess I can close the issue :-)

closed 2020-01-01 by alexpevzner

debiantriage, 2020-01-01:
On Wed 01 Jan 2020 at 06:13:46 -0800, Alexander Pevzner wrote:

> Hi, Brian,

Hello Alexander,
 
> thank you for good news. This is HP ENVY 4500, correct?

An HP ENVY 4520, actually; one of the HP ENVY 4500 series.

[Snip]

> What does Brother use? Does it support WSD, or only some proprietary 
> protocol?

I am nowhere being a scanner expert but I think a proprietary protocol
is used. None of the TXT records I have for recent devices show WDS or
eSCL. They do show a _scanner._tcp service.

Regards,

Brian.

alexpevzner, 2020-01-01:
WSD uses its own discovery method, WS-Discovery. This is Microsoft, they live in alternative universe. I heard that even Newton laws in their universe are different.

But WSD is documented, and documentation is publicly available, which is unusual for Microsoft. The protocol looks very similar to eSCL, so in theory it can be implemented with a reasonable effort.