alexpevzner/sane-airscan #3: HP LaserJet MFP M28w [sane_read(): Device busy]

thokis, 2020-01-08:
[simple-scan-HP-LaserJet-MFP-M28w-(0C56F5).log](https://github.com/alexpevzner/sane-airscan/files/4037188/simple-scan-HP-LaserJet-MFP-M28w-.0C56F5.log)

[simple-scan-HP-LaserJet-MFP-M28w-(0C56F5).tar.gz](https://github.com/alexpevzner/sane-airscan/files/4037189/simple-scan-HP-LaserJet-MFP-M28w-.0C56F5.tar.gz)

It is kinda a weird MFP. For me it is only possible to print with hpaio. sane-airscan shows the `sane_read(): Device busy` message and nothing else. Maybe you need some more information so let me know if i can help you. And big thanks to you for your work i really respect people like you

alexpevzner, 2020-01-09:
Hi @thokis,
thank you for your feedback.

I've tweaked sane-airscan here and there, please try again. Note, you will need to build from fresh sources from github

thokis, 2020-01-09:
I mark the issue as resolved and you could add the HP LaserJet Pro M28w to the working list. Scans as fast as the hpaio backend and get's discovered way faster then before. Thank you very much

closed 2020-01-09 by thokis

alexpevzner, 2020-01-09:
@thokis,

thank you very much for good news. May I ask you to gather a protocol trace anyway, for my collection? .tar file is not needed, only the .log file.

thokis, 2020-01-09:
[scanimage-HP-LaserJet-MFP-M28w-(0C56F5).log](https://github.com/alexpevzner/sane-airscan/files/4041784/scanimage-HP-LaserJet-MFP-M28w-.0C56F5.log)
[simple-scan-HP-LaserJet-MFP-M28w-(0C56F5).log](https://github.com/alexpevzner/sane-airscan/files/4041776/simple-scan-HP-LaserJet-MFP-M28w-.0C56F5.log)

scanimage still says that the device is busy when i try a actual scan but i think the tool is broken because hpaio also does not work with it. simple-scan on the other hand works like a charm.