alexpevzner/sane-airscan #4: HP Color LaserJet MFP M281fdw

debiantriage, 2020-01-10:
>    device `airscan:HP Color LaserJet MFP M281fdw (C23F2C)' is a AirScan HP Color LaserJet MFP M281fdw (C23F2C) eSCL network scanner

https://forums.linuxmint.com/viewtopic.php?f=51&t=309086

alexpevzner, 2020-01-10:
@debiantriage, it's not quite clear, what action is required here from me. Device seems to work, correct?

debiantriage, 2020-01-11:
On Fri 10 Jan 2020 at 13:12:50 -0800, Alexander Pevzner wrote:

> @debiantriage, it's not quite clear, what action is required here
> from me. Device seems to work, correct?

Indeed, the user reports the device as working. I guess the report
needn't remain open.

Cheers,

Brian.

alexpevzner, 2020-01-11:
OK, thank you, @debiantriage 

Seems that some users experience certain difficulties installing the driver. May be, I should somehow clarify the documentation, how do you think?

closed 2020-01-11 by alexpevzner

debiantriage, 2020-01-12:
On Sat 11 Jan 2020 at 12:57:56 -0800, Alexander Pevzner wrote:

> OK, thank you, @debiantriage 
> 
> Seems that some users experience certain difficulties installing the
> driver. May be, I should somehow clarify the documentation, how do
> you think?

Some users will always have difficulties, even when given full
instructions. Your docs are fine.

Cheers,

Brian.