alexpevzner/sane-airscan #5: Gentoo: Build fails, sandbox violation

joakim-tjernlund, 2020-01-19:
Gentoo sandbox error:
```
>>> Install media-gfx/sane-airscan-0.9.10 into /var/tmp/portage/media-gfx/sane-airscan-0.9.10/image
make -j8 -s DESTDIR=/var/tmp/portage/media-gfx/sane-airscan-0.9.10/image install 
 * ACCESS DENIED:  open_wr:      /etc/sane.d/airscan.conf
cp: cannot create regular file '/etc/sane.d/airscan.conf': Permission denied
make: *** [Makefile:58: install] Error 1
 * ERROR: media-gfx/sane-airscan-0.9.10::jockes_overlay failed (install phase):
 *   emake failed
 * 
 * If you need support, post the output of `emerge --info '=media-gfx/sane-airscan-0.9.10::jockes_overlay'`,
 * the complete build log and the output of `emerge -pqv '=media-gfx/sane-airscan-0.9.10::jockes_overlay'`.
 * The complete build log is located at '/var/tmp/portage/media-gfx/sane-airscan-0.9.10/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/media-gfx/sane-airscan-0.9.10/temp/environment'.
 * Working directory: '/var/tmp/portage/media-gfx/sane-airscan-0.9.10/work/sane-airscan-0.9.10'
 * S: '/var/tmp/portage/media-gfx/sane-airscan-0.9.10/work/sane-airscan-0.9.10'
 * --------------------------- ACCESS VIOLATION SUMMARY ---------------------------
 * LOG FILE: "/var/log/sandbox/sandbox-4.log"
 * 
VERSION 1.0
FORMAT: F - Function called
FORMAT: S - Access Status
FORMAT: P - Path as passed to function
FORMAT: A - Absolute Path (not canonical)
FORMAT: R - Canonical Path
FORMAT: C - Command Line

F: open_wr
S: deny
P: /etc/sane.d/airscan.conf
A: /etc/sane.d/airscan.conf
R: /etc/sane.d/airscan.conf
C: cp -n airscan.conf /etc/sane.d 
 * --------------------------------------------------------------------------------

joakim-tjernlund, 2020-01-19:
After hacking DESTDIR one gets:
```
>>> Completed installing media-gfx/sane-airscan-0.9.10 into /var/tmp/portage/media-gfx/sane-airscan-0.9.10/image

 * Final size of build directory: 928 KiB
 * Final size of installed tree:  108 KiB

 * One or more compressed files were found in docompress-ed directories.
 * Please fix the ebuild not to install compressed files (manpages,
 * documentation) when automatic compression is used:
 * 
 *   /usr/share/man/man5/sane-airscan.5.gz

 * QA Notice: Pre-stripped files found:
 * /usr/lib64/sane/libsane-airscan.so.1

joakim-tjernlund, 2020-01-19:
CFLAGS are hardcoded:
CFLAGS	= -O2 -g -W -Wall -Werror -fPIC
and has Werror forced on, this is not good for distributions
No LDFLAGS

joakim-tjernlund, 2020-01-19:
Something like this maybe:
```
CFLAGS	+= -fPIC
# The workaround is to prevent our backend's shared object from being
# unloaded when not longer in use, and these magical options do it
# by adding NODELETE flag to the resulting ELF shared object
LDFLAGS += -Wl,-z,nodelete

install: all
	install -m 644 -D airscan.conf $(DESTDIR)/$(PREFIX)$(CONFDIR)/airscan.conf
	install -m 644 -D dll.conf $(DESTDIR)/$(PREFIX)$(CONFDIR)/dll.d/airscan/dll.conf
	install -D -t $(DESTDIR)/$(PREFIX)/$(LIBDIR)/sane $(BACKEND)
	install -m 644 -D $(MANPAGE) $(DESTDIR)/$(PREFIX)/$(MANDIR)/man5/$(MANPAGE)

joakim-tjernlund, 2020-01-19:
Some more fixes, looks like this now:
```
LDFLAGS += -Wl,-z,nodelete

all:	$(BACKEND) test

$(BACKEND): Makefile $(SRC) airscan.h airscan.sym
	$(CC) -o $(BACKEND) -shared $(SRC) $(CFLAGS) $(LDFLAGS)

install: all
	install -v -m 644 -D airscan.conf $(DESTDIR)/$(PREFIX)$(CONFDIR)/airscan.conf
	install -v -m 644 -D dll.conf $(DESTDIR)/$(PREFIX)$(CONFDIR)/dll.d/airscan/dll.conf
	install -v -D -t $(DESTDIR)/$(PREFIX)/$(LIBDIR)/sane $(BACKEND)
	install -v -m 644 -D $(MANPAGE) $(DESTDIR)/$(PREFIX)/$(MANDIR)/man5/$(MANPAGE)
clean:
	rm -f test $(BACKEND) tags

test:	$(BACKEND) test.c
	$(CC) -o test test.c $(BACKEND) -Wl,-rpath . $(CFLAGS) $(LDFLAGS)

joakim-tjernlund, 2020-01-19:
I think xFLAGS like below will work for all
```
CFLAGS ?= -O2 -g -W -Wall -Werror
CFLAGS += -fPIC
CFLAGS += `pkg-config --cflags --libs avahi-client`
CFLAGS += `pkg-config --cflags --libs avahi-glib`
CFLAGS += `pkg-config --cflags --libs libjpeg`
CFLAGS += `pkg-config --cflags --libs libsoup-2.4`
CFLAGS += `pkg-config --cflags --libs libxml-2.0`

LDFLAGS ?= 
LDFLAGS += -Wl,-z,nodelete
LDFLAGS += -Wl,--version-script=airscan.sym

alexpevzner, 2020-01-19:
You may use PREFIX instead of DESTDIR. It has exactly the same purpose.

I don't agree regarding -Werror. This code should compile without warnings. If it doesn't, I'd better fix the code.

LDFLAGS is meaningless, because everything is compiled and linked by a single gcc call.

I'm not very familiar with Gentoo packaging system. If it has some specific requirements, please let me know, I'll rearrange the Makefile in order to support these requirements.

alexpevzner, 2020-01-19:
I've added DESTDIR support

alexpevzner, 2020-01-19:
@joakim-tjernlund , if you want me to make CFLAGS and LDFLAGS overrideable from the outside, I want to better understand, how it will be used

joakim-tjernlund, 2020-01-20:
It is a user/distribution choice whether to build with debug/fortiy/LTO etc. and the way
that is done is via xFLAGS. These should always be user settable.
Ideally you should only enforce flags you need for correct operation. 

Also, whether to strip bins or not, compression of docs/man pages

joakim-tjernlund, 2020-01-20:
BTW, maybe install --compare works for you rather than cp -n ?

joakim-tjernlund, 2020-01-20:
Forgot, LDFLAGS is not meaningless. A user/distribution will typically pass linker flags 
via LDFLAGS. For maximum compatibility you should mimic the the built in make rules.
Same for CPPFLAGS

joakim-tjernlund, 2020-01-20:
w.r.t Werror, different compilers product different warning, often fals warnings too. This will break
the build needlessly. Werror is a good development flag though

alexpevzner, 2020-01-20:
@joakim-tjernlund ,

I still don't agree about -Werror. If build with warnings will not fail, nobody will complain until problem will prove itself in runtime. I know that different compilers produce different warnings, but in general, different compilers has different bugs, and this is the same situation, it requires my personal attention. Though, I don't expect a lot of surprises here, this software already builds on 10 different distros, with different compiler bugs.

joakim-tjernlund, 2020-01-20:
And then there are your deps, any of you libs may develop a warning.
But Werror is least important ATM. What about the other stuff?

alexpevzner, 2020-01-20:
install --compare "in some cases, do not modify the destination at all". So its behaviour is not documented. cp -n  does exactly what desired: doesn't override existent file (which may contain user's settings). This is my responsibility though to ensure that old version of configuration file will always be compatible with newer version of software.

alexpevzner, 2020-01-20:
Is there any written recommendations on interfaces that Makefile should implement to make Gentoo users happy?

joakim-tjernlund, 2020-01-20:
hmm, not that I know. However, it is expect to follow the autotools rules.

joakim-tjernlund, 2020-01-20:
I think you will find lots of other uses will be happy as well. Look at the ARCH pkg, they have
skipped your install rule and hand code the install.

joakim-tjernlund, 2020-01-20:
Here is a new suggestion, will get very close. What do you think?
LIBS is a new variable that users can pass extra libs with.
```
diff --git a/Makefile b/Makefile
index 3a88e7d..0e14e3e 100644
--- a/Makefile
+++ b/Makefile
@@ -23,13 +23,13 @@ SRC = \
        airscan-zeroconf.c \
        sane_strstatus.c
 
-CFLAGS = -O2 -g -W -Wall -Werror -fPIC
+CFLAGS ?= -O2 -g -W
+CFLAGS += -Wall -Werror
 CFLAGS += `pkg-config --cflags --libs avahi-client`
 CFLAGS += `pkg-config --cflags --libs avahi-glib`
 CFLAGS += `pkg-config --cflags --libs libjpeg`
 CFLAGS += `pkg-config --cflags --libs libsoup-2.4`
 CFLAGS += `pkg-config --cflags --libs libxml-2.0`
-CFLAGS += -Wl,--version-script=airscan.sym
 
 # Merge DESTDIR and PREFIX
 PREFIX := $(abspath $(DESTDIR)/$(PREFIX))
@@ -50,25 +50,27 @@ endif
 # The workaround is to prevent our backend's shared object from being
 # unloaded when not longer in use, and these magical options do it
 # by adding NODELETE flag to the resulting ELF shared object
-CFLAGS += -Wl,-z,nodelete
+LDFLAGS ?=
+LDFLAGS += -Wl,-z,nodelete
+LDFLAGS += -Wl,--version-script=airscan.sym
 
 all:   $(BACKEND) test
 
 $(BACKEND): Makefile $(SRC) airscan.h airscan.sym
        -ctags -R .
-       gcc -o $(BACKEND) -shared $(SRC) ${CFLAGS}
+       $(CC) -o $(BACKEND) -shared -fPIC $(CFLAGS) $(LDFLAGS) $(SRC) $(LIBS)
 
 install: all
        mkdir -p $(PREFIX)$(CONFDIR)
        mkdir -p $(PREFIX)$(CONFDIR)/dll.d
        cp -n airscan.conf $(PREFIX)$(CONFDIR)
        cp -n dll.conf $(PREFIX)$(CONFDIR)/dll.d/airscan
-       install -s -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
-       mkdir -p $(PREFIX)/$(MANDIR)/man5
-       gzip <$(MANPAGE) > $(PREFIX)$(MANDIR)/man5/$(MANPAGE).gz
+       install -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
+       install -D $(PREFIX)/$(MANDIR)/man5/
+       mkdir -p $(MANPAGE) $(PREFIX)/$(MANDIR)/man5/$(MANPAGE)
 
 clean:
        rm -f test $(BACKEND) tags
 
 test:  $(BACKEND) test.c
-       gcc -o test test.c $(BACKEND) -Wl,-rpath . ${CFLAGS}
+       $(CC) -o test test.c $(CFLAGS) $(LDFLAGS) $(BACKEND) -Wl,-rpath . $(LIBS)

joakim-tjernlund, 2020-01-20:
ouch, typos(I cannot build on this machine so not tested), new version:
```
--- a/Makefile
+++ b/Makefile
@@ -23,13 +23,13 @@ SRC = \
        airscan-zeroconf.c \
        sane_strstatus.c
 
-CFLAGS = -O2 -g -W -Wall -Werror -fPIC
+CFLAGS ?= -O2 -g -W
+CFLAGS += -Wall -Werror
 CFLAGS += `pkg-config --cflags --libs avahi-client`
 CFLAGS += `pkg-config --cflags --libs avahi-glib`
 CFLAGS += `pkg-config --cflags --libs libjpeg`
 CFLAGS += `pkg-config --cflags --libs libsoup-2.4`
 CFLAGS += `pkg-config --cflags --libs libxml-2.0`
-CFLAGS += -Wl,--version-script=airscan.sym
 
 # Merge DESTDIR and PREFIX
 PREFIX := $(abspath $(DESTDIR)/$(PREFIX))
@@ -50,25 +50,26 @@ endif
 # The workaround is to prevent our backend's shared object from being
 # unloaded when not longer in use, and these magical options do it
 # by adding NODELETE flag to the resulting ELF shared object
-CFLAGS += -Wl,-z,nodelete
+LDFLAGS ?=
+LDFLAGS += -Wl,-z,nodelete
+LDFLAGS += -Wl,--version-script=airscan.sym
 
 all:   $(BACKEND) test
 
 $(BACKEND): Makefile $(SRC) airscan.h airscan.sym
        -ctags -R .
-       gcc -o $(BACKEND) -shared $(SRC) ${CFLAGS}
+       $(CC) -o $(BACKEND) -shared -fPIC $(CFLAGS) $(LDFLAGS) $(SRC) $(LIBS)
 
 install: all
        mkdir -p $(PREFIX)$(CONFDIR)
        mkdir -p $(PREFIX)$(CONFDIR)/dll.d
        cp -n airscan.conf $(PREFIX)$(CONFDIR)
        cp -n dll.conf $(PREFIX)$(CONFDIR)/dll.d/airscan
-       install -s -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
-       mkdir -p $(PREFIX)/$(MANDIR)/man5
-       gzip <$(MANPAGE) > $(PREFIX)$(MANDIR)/man5/$(MANPAGE).gz
+       install -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
+       install -m 644 -D $(MANPAGE) $(PREFIX)/$(MANDIR)/man5/$(MANPAGE)
 
 clean:
        rm -f test $(BACKEND) tags
 
 test:  $(BACKEND) test.c
-       gcc -o test test.c $(BACKEND) -Wl,-rpath . ${CFLAGS}
+       $(CC) -o test $(CFLAGS) $(LDFLAGS) test.c $(BACKEND) -Wl,-rpath . $(LIBS)

alexpevzner, 2020-01-20:
ARCH people contacted me and didn't ask to change something :-)

The primary target of my install rules is a common user who runs make install without risk to break his or her system or override his or her settings. Package maintainers may override my make install as they wish.

alexpevzner, 2020-01-20:
Your Makefile doesn't work, unfortunately. make CFLAGS=something overrides CFLAGS completely

What is the reason to add extra libs?

joakim-tjernlund, 2020-01-20:
> ARCH people contacted me and didn't ask to change something :-)
> 
> The primary target of my install rules is a common user who runs make install without risk to break his or her system or override his or her settings. Package maintainers may override my make install as they wish.

Goal was not to break the lonely dev builds, but add something that will work for packages too.
Any quality SW has a working install target, check your dependant libs.

joakim-tjernlund, 2020-01-20:
> Your Makefile doesn't work, unfortunately. make CFLAGS=something overrides CFLAGS completely
Got to look at that from home then.
Does 'CFLAGS=something make' also override?
> 
> What is the reason to add extra libs?

Most common I think is to add in some extra debug lib. It could also be
that a dependant lib is split and one needs to add the splitted libs. 

Question: Would other software be able to link to your lib to do something useful?
If so, a header file should also be installed

alexpevzner, 2020-01-20:
I's not a library to link to, it's a dynamically loaded plugin.

If dependant library is split, it should be handled by its pkg-config file

joakim-tjernlund, 2020-01-20:
This works for me:
```
From 4598b080d8d5f3d2464259e32e36a39f1929b9a2 Mon Sep 17 00:00:00 2001
From: Joakim Tjernlund <Joakim.Tjernlund@infinera.com>
Date: Tue, 21 Jan 2020 00:52:24 +0100
Subject: [PATCH] Makefile: respect CFLAGS, LDFLAGS and LIBS

Also adjust install not to compress man page

---
 Makefile | 32 +++++++++++++++++++-------------
 1 file changed, 19 insertions(+), 13 deletions(-)

diff --git a/Makefile b/Makefile
index 3a88e7d..8b99927 100644
--- a/Makefile
+++ b/Makefile
@@ -23,13 +23,20 @@ SRC	= \
 	airscan-zeroconf.c \
 	sane_strstatus.c
 
-CFLAGS	= -O2 -g -W -Wall -Werror -fPIC
-CFLAGS += `pkg-config --cflags --libs avahi-client`
-CFLAGS += `pkg-config --cflags --libs avahi-glib`
-CFLAGS += `pkg-config --cflags --libs libjpeg`
-CFLAGS += `pkg-config --cflags --libs libsoup-2.4`
-CFLAGS += `pkg-config --cflags --libs libxml-2.0`
-CFLAGS += -Wl,--version-script=airscan.sym
+CFLAGS ?= -O2 -g -W -Wall -Werror
+
+AIR_LIBS   = $(shell pkg-config --libs avahi-client)
+AIR_LIBS  += $(shell pkg-config --libs avahi-glib)
+AIR_LIBS  += $(shell pkg-config --libs libjpeg)
+AIR_LIBS  += $(shell pkg-config --libs libsoup-2.4)
+AIR_LIBS  += $(shell pkg-config --libs libxml-2.0)
+
+AIR_FLAGS  = $(shell pkg-config --cflags  avahi-client)
+AIR_FLAGS += $(shell pkg-config --cflags  avahi-glib)
+AIR_FLAGS += $(shell pkg-config --cflags  libjpeg)
+AIR_FLAGS += $(shell pkg-config --cflags  libsoup-2.4)
+AIR_FLAGS += $(shell pkg-config --cflags  libxml-2.0)
+AIR_FLAGS += -Wl,--version-script=airscan.sym
 
 # Merge DESTDIR and PREFIX
 PREFIX := $(abspath $(DESTDIR)/$(PREFIX))
@@ -50,25 +57,24 @@ endif
 # The workaround is to prevent our backend's shared object from being
 # unloaded when not longer in use, and these magical options do it
 # by adding NODELETE flag to the resulting ELF shared object
-CFLAGS += -Wl,-z,nodelete
+AIR_FLAGS += -Wl,-z,nodelete
 
 all:	$(BACKEND) test
 
 $(BACKEND): Makefile $(SRC) airscan.h airscan.sym
 	-ctags -R .
-	gcc -o $(BACKEND) -shared $(SRC) ${CFLAGS}
+	$(CC) -o $(BACKEND) -shared -fPIC $(AIR_FLAGS) $(CFLAGS) $(LDFLAGS) $(LIBS) $(AIR_LIBS) $(SRC)
 
 install: all
 	mkdir -p $(PREFIX)$(CONFDIR)
 	mkdir -p $(PREFIX)$(CONFDIR)/dll.d
 	cp -n airscan.conf $(PREFIX)$(CONFDIR)
 	cp -n dll.conf $(PREFIX)$(CONFDIR)/dll.d/airscan
-	install -s -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
-	mkdir -p $(PREFIX)/$(MANDIR)/man5
-	gzip <$(MANPAGE) > $(PREFIX)$(MANDIR)/man5/$(MANPAGE).gz
+	install -D -t $(PREFIX)$(LIBDIR)/sane $(BACKEND)
+	install -m 644 -D -t $(PREFIX)/$(MANDIR)/man5/ $(MANPAGE)
 
 clean:
 	rm -f test $(BACKEND) tags
 
 test:	$(BACKEND) test.c
-	gcc -o test test.c $(BACKEND) -Wl,-rpath . ${CFLAGS}
+	$(CC) -o test test.c $(BACKEND) $(CFLAGS) $(AIR_FLAGS) $(CFLAGS) $(LDFLAGS) $(LIBS) $(AIR_LIBS)  -Wl,-rpath .
-- 
2.24.1

alexpevzner, 2020-01-21:
OK, it is much better.

I don't want to hardcode uncompressed man page. Most of distros I know use compressed man pages. Hardcoding uncompressed man page will be regression for already supported distros.

So it should be either parameter or automatic guess.

joakim-tjernlund, 2020-01-21:
> OK, it is much better.
> 
> I don't want to hardcode uncompressed man page. Most of distros I know use compressed man pages. Hardcoding uncompressed man page will be regression for already supported distros.

Yes, but the compress themself if they want to and select compression method too. Gentoo has bz2,
not gz
> 
> So it should be either parameter or automatic guess.
No, just leave it uncompressed.

joakim-tjernlund, 2020-01-21:
Just a note, your cp -n rather than install does not set privs so the file will just inherit these, whatever it may be.

joakim-tjernlund, 2020-01-21:
Qs:
-  What is /etc/sane.d/dll.d/airscan/dll.conf? Is it used on Linux?
- Could ZeroConf be a compile config option? Can't use it and don't want it installed.

alexpevzner, 2020-01-21:
/etc/sane.d/dll.d/airscan/dll.conf instructs sane-dll (man sane-dll) to load sane-airscan

Build without avahi libraries is not supported, but auto-discovery can be disabled in airscan.conf

joakim-tjernlund, 2020-01-23:
No changes to Makefile committed yet, are you not happy still ?

alexpevzner, 2020-01-23:
I'm sorry. Give me a bit of time, please. I'm exceptionally busy with another project now

joakim-tjernlund, 2020-02-05:
ping?

alexpevzner, 2020-02-05:
Coming soon, really.

BTW, do you really use sane-airscan with some device? Does it work for you?

joakim-tjernlund, 2020-02-09:
I tried on my homw printer/scanner but it turned out that my Canon MG5500 doesn't support
airscan :(

Now I am just waiting for you to fix the Make system and release it so I can submit an ebuild to Gentoo

PS.
     To be complete you should also include CPPFLAGS

alexpevzner, 2020-02-11:
Hi, @joakim-tjernlund 

I've added some variables, but probably not all that you expect :-)

Please have a look and give me some feedback. Sorry for delay.

Regarding man page compression, now you have COMPRESS variable, you can set it to "gzip", "bzip2" or "" for no compression

alexpevzner, 2020-02-11:
Apple claims that Canon MG5500 series does support airscan:
https://support.apple.com/en-us/HT201311

The following command:
`avahi-browse -rt _uscan._tcp`
does it show your device? Note, your computer and device needs to be connected to the same LAN segment

joakim-tjernlund, 2020-02-11:
avahi-browse -rt _uscan._tcp does not show anything

joakim-tjernlund, 2020-02-11:
I do have these though:
```
avahi-browse -rt _scanner._tcp
+   eth0 IPv4 Canon MG5500 series                           _scanner._tcp        local
=   eth0 IPv4 Canon MG5500 series                           _scanner._tcp        local
   hostname = [2BF972000000.local]
   address = [192.168.0.81]
   port = [8612]
   txt = ["mac=D8:49:2F:2B:F9:72" "scannerAvailable=1" "uuid=61099C" "mdl=MG5500 series" "mfg=Canon" "ty=Canon MG5500 series" "adminurl=http://2BF972000000.local." "txtvers=1"]

avahi-browse -rt _canon-bjnp1._tcp
+   eth0 IPv4 Canon MG5500 series                           _canon-bjnp1._tcp    local
=   eth0 IPv4 Canon MG5500 series                           _canon-bjnp1._tcp    local
   hostname = [2BF972000000.local]
   address = [192.168.0.81]
   port = [8611]
   txt = ["mac=D8:49:2F:2B:F9:72" "Fax=F" "Scan=T" "Duplex=T" "Color=T" "UUID=00000000-0000-1000-8000-D8492F2BF972" "usb_CMD=" "usb_MDL=MG5500 series" "usb_MFG=Canon" "adminurl=http://2BF972000000.local." "pdl=application/octet-stream" "product=(Canon MG5500 series)" "ty=Canon MG5500 series" "priority=20" "qtotal=1" "note=" "rp=auto" "txtvers=1"]

alexpevzner, 2020-02-11:
Is it all it advertises? I don't see even IPP protocol here...

joakim-tjernlund, 2020-02-11:
No, there is more, ipp is there:
```
=   eth0 IPv4 Canon MG5500 series                           Internet Printer     local
   hostname = [2BF972000000.local]
   address = [192.168.0.81]
   port = [631]
   txt = ["mac=D8:49:2F:2B:F9:72" "kind=document,envelope,photo" "PaperMax=legal-A4" "Fax=F" "Scan=T" "Duplex=T" "Color=T" "URF=V1.3,CP1,PQ4-5,RS600,SRGB24,W8,OB9,OFU0,DM3,IS1" "UUID=00000000-0000-1000-8000-D8492F2BF972" "usb_CMD=URF" "usb_MDL=MG5500 series" "usb_MFG=Canon" "adminurl=http://2BF972000000.local./index.html?PAGE_AAP" "pdl=application/octet-stream,image/urf,image/jpeg" "product=(Canon MG5500 series)" "ty=Canon MG5500 series" "priority=15" "qtotal=1" "note=" "rp=ipp/print" "txtvers=1"]

joakim-tjernlund, 2020-02-12:
> Hi, @joakim-tjernlund
> 
> I've added some variables, but probably not all that you expect :-)
> 
> Please have a look and give me some feedback. Sorry for delay.
> 
> Regarding man page compression, now you have COMPRESS variable, you can set it to "gzip", "bzip2" or "" for no compression

So I had a look now and not what I requested. Frankly, I am tired of this now, you obviously don't know Makefiles/build conventions and don't want my advise either.

alexpevzner, 2020-02-13:
You requested certain make parameters to be settable from outside, and now you have it. Few parameters could be missed, though.

I have partially accepted your advices and now I've allocated some time to solve this issue. So with your constructive feedback this issue can be resolved in a couple of iterations, I believe.

Nobody knows Makefile conventions, because these conventions are not documented anywhere. Even between autotools and cmake, there are a lot of differences.

Please note, I must be careful, because I don't want to break other 10 supported distros.

joakim-tjernlund, 2020-02-13:
Why did you throw out my patch? What did it break?
Tell me what what broke/is bad in my patch

alexpevzner, 2020-02-13:
My version is very similar to yours, except local variable names and some details.

One incorrect thing in your patch, sources should not be placed after flags, it breaks .so linking on Ubuntu (due to gcc bug, but I have to live with it). I must be careful with things like these.

Also, I don't want to unconditionally install unstripped .so and uncompressed man. For man page compression I've offered you a solution, for stripping binaries, ideally I'd generate a separate debuginfo packages for systems that support them, but it requires more effort (because all systems a different at this aspect)

joakim-tjernlund, 2020-02-13:
> My version is very similar to yours, except local variable names and some details.

No, it is still broken. It still force your view of the world upon everyone
> 
> One incorrect thing in your patch, sources should not be placed after flags, it breaks .so linking on Ubuntu (due to gcc bug, but I have to live with it). I must be careful with things like these.

So fix/deal with that without forcing all into the same corner. 

> Also, I don't want to unconditionally install unstripped .so and uncompressed man. For man page compression I've offered you a solution, for stripping binaries, ideally I'd generate a separate [...]

That is a user/distribution choice. What if I want to install with debug syms?
of build with clang or pass linker flags ?
All of this should be "configurable" without having to hack the build system.

alexpevzner, 2020-02-13:
You can build with clang - use CC

There is no separate linker pass. You can use CFLAGS to pass any flags. If you think, you also need CPPFLAGS and LDFLAGS, OK, no problem, I can add it. Though I can't understand, what linker flags you may want to pass.

Regarding stripping binaries, have no idea how to make it configurable. 

So is my understanding correct, adding CPPFLAGS and LDFLAGS is enough to satisfy the issue?

alexpevzner, 2020-02-13:
I've added CPPFLAGS and LDFLAGS

joakim-tjernlund, 2020-02-13:
> You can build with clang - use CC
>
> There is no separate linker pass. You can use CFLAGS to pass any flags. If you think, you also need CPPFLAGS and LDFLAGS, OK, no problem, I can add it. Though I [...]

Honestly, I am not sure. When I saw you had just throw away my work and added something
else that was broken(again) I gave up. Hopefully I will find some motivation to look at
it again in detail, or you could revert your stuff, add in my work and then adjust on top on that.

alexpevzner, 2020-02-13:
I didn't throw away you work, I just didn't apply it mechanically

joakim-tjernlund, 2020-02-13:
> I didn't throw away you work, I just didn't apply it mechanically

No, not really. You botched my intent with those changes.
Again, you don't know what constitutes a good Makefile but think you do.

If you still want my help, revert your stuff and make changes on top on my changes
in a branch, once done you can squash all into one commit on master.

alexpevzner, 2020-02-13:
This is not constructive, sorry

closed 2020-02-13 by alexpevzner

alexpevzner, 2020-03-09:
Hi @joakim-tjernlund ,

according to specs, your device seems to support WSD (WS-Scan) protocol. This is a XML-based protocol, developed by Microsoft, it is quite similar to eSCL.

Recently, I've made a fork of sane-airscan with a purpose to add WSD support. I want to invite you to participate in testing of this backend:
https://github.com/alexpevzner/sane-airscan-wsd

Discovery is not completely implemented yet, but manual configuration should work, more or less. As it is very difficult to guess a correct URL (WS-Scan uses its own discovery protocol, having nothing in common with DNS-SD), this is better to start from testing a discovery tool:
https://github.com/alexpevzner/airscan-discover

On success, this tool prints few lines, suitable for adding to the configuration file. It looks like this:
> $ ./airscan-discover
> [devices]
>   "Kyocera ECOSYS M2040dn" = http://192.168.1.102:9095/eSCL
>   "Kyocera ECOSYS M2040dn" = http://192.168.1.102:5358/DeviceService/, wsd

If it doesn't, I need to see its output with `-d` option.

Please note, for sane-airscan-wsd the configuration file is temporary named as `/etc/sane.d/airscan-wsd.conf`

--
Best regards, Alexander Pevzner.

joakim-tjernlund, 2020-03-10:
Hi Alexander, I am very busy ATM but will (try) to remember to check this out later.

Jocke

On Mon, 2020-03-09 at 09:38 -0700, Alexander Pevzner wrote:

Hi @joakim-tjernlund<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjoakim-tjernlund&data=02%7C01%7Cjoakim.tjernlund%40infinera.com%7Cf97f290cfb424201d86908d7c4485035%7C285643de5f5b4b03a1530ae2dc8aaf77%7C1%7C0%7C637193687190328881&sdata=kgGk2uVuP1bfw7cmJN98cEUuYgtSKuaTO0cdHVYiG6k%3D&reserved=0> ,

according to specs, your device seems to support WSD (WS-Scan) protocol. This is a XML-based protocol, developed by Microsoft, it is quite similar to eSCL.

Recently, I've made a fork of sane-airscan with a purpose to add WSD support. I want to invite you to participate in testing of this backend:
https://github.com/alexpevzner/sane-airscan-wsd<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Falexpevzner%2Fsane-airscan-wsd&data=02%7C01%7Cjoakim.tjernlund%40infinera.com%7Cf97f290cfb424201d86908d7c4485035%7C285643de5f5b4b03a1530ae2dc8aaf77%7C1%7C0%7C637193687190338876&sdata=kz3C5pA4O%2Fn%2Bqlj%2FCMJST%2FfYREAVTBOMmznAkS%2BGVJY%3D&reserved=0>

Discovery is not completely implemented yet, but manual configuration should work, more or less. As it is very difficult to guess a correct URL (WS-Scan uses its own discovery protocol, having nothing in common with DNS-SD), this is better to start from testing a discovery tool:
https://github.com/alexpevzner/airscan-discover<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Falexpevzner%2Fairscan-discover&data=02%7C01%7Cjoakim.tjernlund%40infinera.com%7Cf97f290cfb424201d86908d7c4485035%7C285643de5f5b4b03a1530ae2dc8aaf77%7C1%7C0%7C637193687190338876&sdata=7mXd8XtWIQqlI1TT%2FaXFbKwtN%2F1zeiAy0mQajsJxKYU%3D&reserved=0>

On success, this tool prints few lines, suitable for adding to the configuration file. It looks like this:

$ ./airscan-discover
[devices]
"Kyocera ECOSYS M2040dn" = http://192.168.1.102:9095/eSCL<https://nam03.safelinks.protection.outlook.com/?url=http%3A%2F%2F192.168.1.102%3A9095%2FeSCL&data=02%7C01%7Cjoakim.tjernlund%40infinera.com%7Cf97f290cfb424201d86908d7c4485035%7C285643de5f5b4b03a1530ae2dc8aaf77%7C1%7C0%7C637193687190348873&sdata=7xCcUDJM2dz1zJXDem2sJ34leTc6cI%2FP%2BU9EOnenOk8%3D&reserved=0>
"Kyocera ECOSYS M2040dn" = http://192.168.1.102:5358/DeviceService/<https://nam03.safelinks.protection.outlook.com/?url=http%3A%2F%2F192.168.1.102%3A5358%2FDeviceService%2F&data=02%7C01%7Cjoakim.tjernlund%40infinera.com%7Cf97f290cfb424201d86908d7c4485035%7C285643de5f5b4b03a1530ae2dc8aaf77%7C1%7C0%7C637193687190348873&sdata=Lo3chcwJ9N9MGjk9I%2BmpC6%2FUpPIP1X4kWpoL3S8sDAU%3D&reserved=0>, wsd

If it doesn't, I need to see its output with -d option.

Please note, for sane-airscan-wsd the configuration file is temporary named as /etc/sane.d/airscan-wsd.conf

alexpevzner, 2020-03-10:
OK, thank you!