alexpevzner/sane-airscan #6: A Canon TS6100 series issue

debiantriage, 2020-01-27:
Hello Alexander,

I am involved with a user trying to use sane-airscan with a Canon TS6100. Please see [https://forums.linuxmint.com/viewtopic.php?f=90&t=310364](url). Scanning appears to happen but we get 

>   vulcan@vulcan-Inspiron-1545 ~ $ scanimage -d 'airscan:Canon TS6100 series' > image.pnm scanimage: relocation error: /usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1: symbol jpeg_crop_scanline, version LIBJPEG_8.0 not defined in file libjpeg.so.8 with link time reference 

You may want to keep an eye on what is going on in this Mint thread.

Cheers,

Brian.

alexpevzner, 2020-01-27:
Hi, Brian,

What build do they use, Debian or Ubuntu?

It seems to be a compatibility issue between libjpeg used for compilation and libjpeg, actually installed on a system.

openSUSE Build Service doesn't support Mint. May be, somebody willing to maintain a native Mint package?

debiantriage, 2020-01-27:
On Mon 27 Jan 2020 at 10:05:40 -0800, Alexander Pevzner wrote:

> Hi, Brian,
> 
> What build do they use, Debian or Ubuntu?
> 
> It seems to be a compatibility issue between libjpeg used for
> compilation and libjpeg, actually installed on a system.
> 
> openSUSE Build Service doesn't support Mint. May be, somebody willing
> to maintain a native Mint package?

You could be correct but Mint 19.x is basically Ubuntu bionic, for which
you, very nicely, provide a Debian file. However, I do not actually know
how the user got sane-airscan on his system - Debian file or compilation.
Compatibility seems a likely avenue for exploration.

I'll press on and see where it goes.

Incidentally, IMHO, sane-airscan should join the collection of official
SANE backends. But it hasn't! Will it?

Cheers,

Brian.

alexpevzner, 2020-01-27:
>Incidentally, IMHO, sane-airscan should join the collection of official SANE backends. But it hasn't! Will it? Cheers, Brian.

Well. I hope so. https://gitlab.com/sane-project/backends/issues/202

There is an alternative backend, sane-escl. It doesn't support ADF, it doesn't support cancel, it doesn't work on my Kyocera device. But it was released 2 weeks before my backend, so now it is in the official collection. I still got no official answer from SANE team. Nobody likes to say "no" to anybody.

debiantriage, 2020-01-28:
On Mon 27 Jan 2020 at 11:28:59 -0800, Alexander Pevzner wrote:

> >Incidentally, IMHO, sane-airscan should join the collection of official SANE
> >backends. But it hasn't! Will it?

> > Cheers, Brian.

> Well. I hope so. https://gitlab.com/sane-project/backends/issues/202
>
> There is an alternative backend, sane-escl. It doesn't support ADF, it
> doesn't support cancel, it doesn't work on my Kyocera [...]

I am only a user so have no position regarding the technical merits of
one backend over the other. All I can say is that sane-airscan worked
out of the box with my HP ENVY 4500. sane-escl took quite a bit of
adjustment (the support was good) to get satisfactory scans. Using
it with Xsane is still not quite right (it is with sane-airscan).

Anyway, what is against having both backends as SANE offerings? Both
get tested and users have a choice. These are cutting-edge softwares;
both deserve widespread exposure. The time of release shouldn't
matter.

Cheers,

Brian.

alexpevzner, 2020-01-28:
I wrote sane-airscan first of all to solve my own problems with my new Kyocera MFP. When I was in the middle of development, sane-escl was announced. I've looked to it and decided to continue.

Obviously, as work is done, and I believe, done good, I want to share my work with another people. After all, if this software will not be used, sooner or later it will die.

Meanwhile, sane-escl doesn't work with my MFP. It scans one page, and then device remains in busy state for about 1 minute. This is major issue, there are many minor issues too.

Regarding common users, I don't think having a choice between two backends that do basically the same has any value for a common user. As for me, it only overcomplicates the life. 

Anyway, it is not a my decision to include or not my backend into the official SANE distribution.

debiantriage, 2020-01-28:
On Tue 28 Jan 2020 at 10:47:46 -0800, Alexander Pevzner wrote:

> I wrote sane-airscan first of all to solve my own problems with my new
> Kyocera MFP. When I was in the middle of development, sane-escl was
> announced. I've looked to it and decided to continue.

I have a three-line bash script that is a frontend to eSCL. It is
likely not a brilliantly written script (and it uses curl) but I
wrote it to see whether it could be done. TBH, it suits me as much
as a fully-fledged backend, but I would not publicise it.

> Obviously, as work is done, and I believe, done good, I want to share
> my work with another people. After all, if this software will not be
> used, sooner or later it will die.

That would be a shame.
 
> Meanwhile, sane-escl doesn't work with my MFP. It scans one page, and
> then device remains in busy state for about 1 minute. This is major
> issue, there are many minor issues too.

As I implied, I really have no horse in this race. But, for you, there
is no contest when it comes to your Kyocera MFP.

> Regarding common users, I don't think having a choice between two
> backends that do basically the same has any value for a common user.
> As for me, it only overcomplicates the life. 

They do not do the same thing. You have given an example above. I'll
mention epson and epson2 as choices offered in SANE.
 
> Anyway, it is not a my decision to include or not my backend into the
> official SANE distribution.

I understand that. Maybe I will contact sane-devel to express my view.

Cheers,

Brian.

alexpevzner, 2020-01-29:
Hi @debiantriage 

seems that Mint 18.1 is actually Ubuntu 16.04, not 18.04. I've added Ubuntu 16.04 package and really noticed that so old libjpeg misses a couple of functions that I did use, so I had to slightly patch my code too.

So it's better now not to break poor 18.1 system by installing foreign  libjpeg-turbo8 but rather try the updated package. 

If it helps, I think I should add to my README few recommendations for the Mint users, what Ubuntu package to use depending on their Mint version. I need Mint community feedback to do so.

debiantriage, 2020-01-29:
On Tue 28 Jan 2020 at 22:49:03 -0800, Alexander Pevzner wrote:

> Hi @debiantriage 

Hello again, Alexander,
 
> If it helps, I think I should add to my README few recommendations for the
> Mint users, what Ubuntu package to use depending on their Mint version. I
> need Mint community feedback to do so.

I am Debian user (and, I believe, a member of the Debian Printing
Team). My involvement with Mint printing/scanning issues is something
I do because I find it interesting. But I am far more familiar with
Debian things rather than with Mint things.

Recommendations for Mint users in your README doesn't seem a bad idea.
Basically, I think xUbuntu_16.04/ would cover Mint 18.x; xUbuntu_18.04/
would suit Mint 19.x. It seems that Mint 20 will be based on Ubuntu
20.04 LTS (Focal Fossa) and is due to be released later this year.

Cheers,

Brian.

alexpevzner, 2020-01-29:
Heh, I'm Fedora user. For me, Debian, Ubuntu and Mint all have the same face :-)

BTW, if SANE won't include my backend, I can imagine that distros will include it as a separate package. For eSCL scanner owners it will be even convenient: they will not have to install a giant sane-backends package with a million of backends, enabled by default and as such very slow at start.

So if Debian will consider inclusion of my backend into their distro as a separate package, I would be very glad.

alexpevzner, 2020-01-29:
Issue seems to be solved by providing Ubuntu 16.04 package, compatible with Mint 18.x

closed 2020-01-29 by alexpevzner

debiantriage, 2020-01-29:
On Wed 29 Jan 2020 at 04:16:22 -0800, Alexander Pevzner wrote:

> Issue seems to be solved by providing Ubuntu 16.04 package, compatible
> with Mint 18.x

Indeed. Thank you for your involvement and the elegagant solution.
Another scanner working with sane-airscan to add to the list.

Cheers,

Brian.