alexpevzner/sane-airscan #7: Issue with POST URL for scanning

markosjal, 2020-02-04:
I have been working on a project for the last year or more for a specific scanner. It is to emulate eSCL functionality on a scanner. 

I recently tried to streamline some code and tested with your SANE back-end.

Now when I POST a file I see it is POSTED from the SANE back-end to 

/eSCL/ScanJobs

My software is responding with "Moved Permanently" but I also see that all queries like ScannerStatus and ScannerCapabilities all use the trailing slash

I tested POSTing an xml file with CURL and I can see that using no trailing slashdoes not work but using a trailing slash does work.  

BTW thanks for the mention of the reverse engineering  part at the bottom of the main project page. This has been a long lonely road with only occassional access to real eSCL devices , let alone a Mac.

alexpevzner, 2020-02-04:
> My software is responding with "Moved Permanently" but I also see that all queries like ScannerStatus and ScannerCapabilities all use the trailing slash

Excuse me, I don't understand that. To query, for example, scanner capabilities, I use URL like this:

HTTP GET http://192.168.1.102:9095/eSCL/ScannerCapabilities

without any trailing slash.

BTW, if you plan to use my backend to test your eSCL server, you may want to know that my backend is able to write a very detailed protocol trace to disk. See "reporting bugs" section of the README.md file for information how to enable this feature.

alexpevzner, 2020-02-10:
Seems I can close this issue

closed 2020-02-10 by alexpevzner