alexpevzner/sane-airscan #8: More of a request than an issue.

markosjal, 2020-02-06:
I am looking for wireshark (or other)  traces of successful eSCL/AirScan scans. If anyone can help please . 

I am looking for all exchanges between scanner and client from scanner detection all the way through to client loading the image. 

I have no eSCL devices here. I would prefer to see it from a Mac to an "officially supported" scanner but will take what I can get.

You can send them to markosjal at the email with a G.

alexpevzner, 2020-02-06:
While working on sane-airscan. I've gathered a small collection of eSCL traces. You can find it here: https://github.com/alexpevzner/eSCL-protocol-traces

I think, I can close the issue, though we still can continue communication here.

closed 2020-02-06 by alexpevzner

markosjal, 2020-02-09:
Thanks for the traces, unfortunately I do not see the raw exchanges there completely

I was able to understand enough with them to see more about what was happening with my scanner Backend to eSCL converter to finally get an image into Simple Scan and XSane tonight for the first time!  This has been the one problem I have had and mostly for lack of sample exchanges , clients and devices to test with. I have had the web GUI and other features going for some time

I am working on getting this on GitHub or somewhere (hopefully up somewhere  can understand). Licensng is being changed to probably MIT  https://opensource.org/licenses/MIT

So I now am even closer to a more functional interface that adapts the following "abandoned" scanners to escl by way of an apache web server and PHP, and testing on a Raspberry Pi 3

ION AirCopy
ION AirCopy E-Post Edition
IScan Fly
Halo Magic Scanner
Mustek iScan Air / S400W
Century CPS-A4WF
Transcription Patri Kun A4 Wi-Fi Portable Scanner
転写パットリくん A4 Wi-Fiポータブルスキャナー

Literally wreless, small, portable, rechargable, battery powered document scanners that can be passed from cubicle to cubicle or room to room. Both of mine were purchased on eBay for about $25 USD over 3 years ago originally for my android tablet.

It also has its own Web GUI so you can scan on any device with a browser (restoring compatibility to iPad and iPhone) , and is easily adaptable to any comand line scanner. I have seen pyton and other scripts to use non-SANE scanning from a command line, where the scanners were not SANE compatible.. I also use imagemagick to generate PDFs and PNGs as the scanner only supports JPG.

So now anyone that can write a command line scanner for their otherwise unsupported scanner, by using these two tools, they can make it SANE compatible. These particular scanners use telnet commands for all functions, but I am using the binary that was previously at http://bastel.duckdns.com/~public 
I have binaries for Raspberry Pi 2/3 , X86_32 and X86_64, as well as original source.

No luck with Mopria however and that seems imoportant. Not sure where those problems come from but it looks to me like in Wireshark after the POST is sent to scan from Mopria, Mopria checks ScannerStatus (for working, processing, or something)  a few times before the "Created 201" comes back and mopria just says "Scan Failed" , instead of waiting. So Mopria must be looking for something in ScannerStatus that it is not finding as I have suspected for some time now. 

The other ssue I a dealing with is how I can get Mopria to report "Stopped" when scanner is off but this client shows no status whatsoever, as if no scanner isinstalled in scanimage -L. Maybe this is correct behavior for SANE?

More info:
Raspberry Pi  connects to scanners access point when scanner is turned on. With wireless interface connected to scanner and wired interface to LAN  Raspberry P acts as a go-getween between Web GUI and eSCL/ArScan commands from clients.

<LAN Clienrts><Raspberry Pi Wired LAN interface> <Raspberry Pi><Raspberry Pi Wireless interface><Scanner>

markosjal, 2020-02-11:
I wanted to give you a big THANK YOU, and a pat on the back. This solution works flawlessly with my solution under Linux. Other than I can not get inSane or SaneDroid to work, it takes my scanners to the next level combined with my own software. 

The only issue I see is that I must start the scanner before Simple Scan, but I can change that in my software. As I am trying to emulate a real scanner I believe a real scanner would not produce any ScannerStatus when turned off , so my software does not. I have also tested ScannerStatus with 

     '<pwg:State>Stopped</pwg:State>'

this seems to tell Mopria that scanner is offline but then again with Linux, I must restart simple scan to then get it to see the scanner. I can see that SimpleScan, and SANE seem to check the scnner at start - up.

What a valuable tool , that makes my effort of the last several months more valuable, becuse I can really use it in SANE. It also opens up a whole new method of creating a scanner pseudo Back-end for those of us who would otherwise be clueless. Emulating eSCL is far easier than writing a SANE back-end.

alexpevzner, 2020-02-11:
Real scanner normally is not turned off, but sleeping. In this sleeping state it responds to all ScannerStatus and similar requests.

My scanner, while sleeping, returns status "Idle". When scanning is requested, it wakes up and performs requested job, then goes to sleep again.

alexpevzner, 2020-02-11:
Note, I only request ScannerStatus in a case of error. I neither request it before scan attempt not after successful scan attempt. The only purpose for me to query  ScannerStatus is to return a meaningful error code after failure (i.e. to distinguish between "ADF empty", "paper jammed", "scanner busy" and "general I/O failure" situations)