alexpevzner/sane-airscan #9: The HP Officejet-Pro-6970

debiantriage, 2020-02-06:
It seems that this device also has a scanner unit that works with sane-airscan.

https://forums.linuxmint.com/viewtopic.php?f=51&t=311452

Skip page 1 if you do not want all the debugging conversation.

Cheers,

Brian.

closed 2020-02-06 by debiantriage

alexpevzner, 2020-02-06:
BTW, the presence of rs=eSCL in _uscan._tcp is not indication of ESCL. The presence of _uscan._tcp itself is indication of ESCL. rs parameter may contain arbitrary string, it's meaning is base URL for the ESCL protocol (relative to the scanner hostname).

Most of devices really use eSCL, but it is not required by protocol. The AirSane program (eSCL server on a top of SANE drivers) use very different scheme, and it's author claims that it works with Apple devices (and it works with my backend too, but doesn't work with sane-escl)

debiantriage, 2020-02-06:
I forgot to say that your Ubuntu 16.04 .deb file came in very useful again. Indeed, your provision of binary packages with the openSUSE Build Service is a real advantage for users. Pointing a user to it beats having to ask her/him to compile a package.

Cheers,

Brian.

alexpevzner, 2020-02-06:
This is actually very nice, that SUSE provides such a service for many distros. Without it it would be much harder for me to provide binaries for many platforms.

I see, you actively promote my driver! :-) Thank a lot for this work! Actually, I'd better prefer people to add my repo to their repo list rather that to download the package directly: at this case they will receive automatic updated.

debiantriage, 2020-02-06:
On Thu 06 Feb 2020 at 10:28:47 -0800, Alexander Pevzner wrote:

> BTW, the presence of rs=eSCL in _uscan._tcp is not indication of ESCL.
> The presence of _uscan._tcp itself is indication of ESCL. rs parameter
> may contain arbitrary string, it's meaning is base URL for the ESCL
> protocol (relative to the scanner hostname).

Thanks for that. If I have understood correctly, I should be telling a
user that merely having a _uscan._tcp output indicates that sane-airscan
(and sane-escl) are suitable backends to use with that scanner.

Please do not forget that I have no real grasp of scanning and, in the
case of eSCL (which I am very taken with), documentation is sparse.

So may I ask: SCL is, I think. Scanner Control Language. What does "e"
stand for? "extended"? "enhanced"? And the "u" in uscan? I've been
seeing it as "universal".

> Most of devices really use eSCL, but it is not required by protocol.
> The AirSane program (eSCL server on a top of SANE drivers) use very
> different scheme, and it's author claims that it works with Apple
> devices (and it works with my backend too, but doesn't work with
> sane-escl)

I will have think about this and may get back to you later.

Thanks again,

Brian.

alexpevzner, 2020-02-06:
Yes, the presence of _uscan._tcp is enough.

_uscan._tcp parameters are very similar to _ipp._tcp. _ipp uses rp, _uscan uses rs ("p" is printer, "s" is scanner).

I have no idea what does the name eSCL mean. All answers must be in unpublished document from Apple. I'd be glad to have a chance to look to this document :-)

debiantriage, 2020-02-06:
On Thu 06 Feb 2020 at 10:35:36 -0800, Alexander Pevzner wrote:

> This is actually very nice, that SUSE provides such a service for many
> distros. Without it it would be much harder for me to provide binaries
> for many platforms.

Without it I doubt I would be doing this work. Users come in many
shapes and sizes. Telling them to install x.deb is a lot easier than
pointing to compilation instructions (no matter how straightforward
you or I think they are).
 
> I see, you actively promote my driver! :-) Thank a lot for this work!
> Actually, I'd better prefer people to add my repo to their repo list
> rather that to download the package directly: at this case they will
> receive automatic updated.

Please bear in mind that I am, to some extent, imposing on users when
I ask them to install sane-airscan in response to a scanning problem
they have. Adding a repro is an extra burden to require at the start
of a triaging process and can complicate the process if the user is
co-operative but not familiar with repro addition.

However, you have a good point and it was something that went through
my mind today. I shall try to remember to point out the existence of
your repro when I can be sure basic scanning works with sane-airscan.

Regards,

Brian.

alexpevzner, 2020-02-06:
May be it would be better after users have installed .deb, using direct link, and impressed how nice everything works now, to offer them an opportunity to install repo as well.

Though sane-airscan is a quite stable now, time to time I release updates. It would be better if users will receive them. Though not all users ever bother to update their system, I know.

The next planned major update will bring the following feature. Currently, I download entire image from scanner and only after that begin to return scanned data to the client program. Actually it would be much better to start returning scanned data as soon as it comes, even partially. It may greatly improve user's experience on devices able to return scanned parts early, because otherwise user may have no indication that something useful happens.

Till told me that his device takes 9 minutes to scan at high resolution! Who will wait 9 minutes without any visible progress indicator?

markosjal, 2020-02-16:
I thought the rs= means "resource"

alexpevzner, 2020-02-16:
@markosjal,

Initially I also thought so. Actually, in IPP printer advertising, the similar parameter called "rp". So "r" may be from "resource", "p" is printer, "s" is scanner. But it only a guess, though.