alexpevzner/sane-airscan #14: HP ENVY 5540 series

debiantriage, 2020-03-02:
I couldn't be bothered to debug HPLIP when there is an easier technique available:

https://ubuntuforums.org/showthread.php?t=2437814

closed 2020-03-02 by debiantriage

alexpevzner, 2020-03-02:
There is a complain that Brother MFC-L2710DW doesn't work well, but this person doesn't respond to my messages...

https://www.phoronix.com/forums/forum/software/distributions/1151191-ubuntu-20-04-lts-could-better-support-scanners-compliant-with-apple-airscan

debiantriage, 2020-03-02:
On Mon 02 Mar 2020 at 08:21:12 -0800, Alexander Pevzner wrote:

> There is a complain that Brother MFC-L2710DW doesn't work well, but this person doesn't respond to my messages...
> 
> https://www.phoronix.com/forums/forum/software/distributions/1151191-ubuntu-20-04-lts-could-better-support-scanners-compliant-with-apple-airscan

I suppose you are referring to

  > I have tested both with my Brother MFC-L2710DW and sane-escl is
  > more stable, sane-airscan scans once then can't find the scanner
  > again.

One response to this is to invite a bug report and participation in the
debugging procedure. Anyway, wouldn't even a single successful scan count
as "it works"? I leave that to you to decide?

Cheers,

Brian.