alexpevzner/sane-airscan #18: HP LaserJet Pro MFP M521dn support

petrjac, 2020-04-03:
With *HP LaserJet Pro MFP M521dn* a get following error for `$ scanimage -L`
```
file airscan-devops.c: line 71 (devopt_choose_colormode): assertion failed: (wanted < NUM_OPT_COLORMODE)
Neúspěšně ukončen (SIGABRT) (core dumped [obraz paměti uložen])
```

In config `[devices]` section of `/etc/sane.d/airscan.conf` I have:
```
"HP 521" = http://172.22.7.38:80/eSCL
```

### Device info
`$ curl -s http://m521-2:80/eSCL/ScannerStatus | xmllint --format -`
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!-- THIS DATA SUBJECT TO DISCLAIMER(S) INCLUDED WITH THE PRODUCT OF ORIGIN. -->
<scan:ScannerStatus xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:scan="http://schemas.hp.com/imaging/escl/2011/02/08" xmlns:pwg2="http://www.pwg.org/schemas/2010/12/sm" xsi:schemaLocation="http://schemas.hp.com/imaging/escl/2011/02/08 CloudScan.xsd">
  <pwg2:Version>1.0</pwg2:Version>
  <pwg2:State>Idle</pwg2:State>
  <scan:AdfState>ScannerAdfEmpty</scan:AdfState>
</scan:ScannerStatus>
```
### System info
```
$ scanimage --version
scanimage (sane-backends) 1.0.27; backend version 1.0.27

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan

$ apt-cache policy sane-airscan 
sane-airscan:
  Instalovaná verze: 0.9.17+54.1
  Kandidát:          0.9.17+54.1
  Tabulka verzí:
 *** 0.9.17+54.1 100
        100 /var/lib/dpkg/status
```

Please let me know if I can get some more information (*coredump* maybe?)

alexpevzner, 2020-04-03:
Hi,
may I ask you to gather protocol trace?
In the /etc/sane.d/airscan.conf, uncomment the following lines:
> [debug]
> trace = ~/airscan/trace
> enable = true

Collected traces will be in the ~/airscan/trace directory. .tag files are not needed for now, only .log files.

You may send it me directly by e-mail: pzz@apevzner.com

alexpevzner, 2020-05-13:
Fixed in sane-airscan-wsd branch, which has been merged here

closed 2020-05-13 by alexpevzner