alexpevzner/sane-airscan #22: Brother MFC-M2750DW support

Keruspe, 2020-04-30:
Hi,

Thanks for this project! My scanner is detected, but when I scan an A4 document, the result if the full page, 4 times smaller, using only one fourth of the resulting A4 image.

Not sure if this is some settings that needs tweaking or if it's harder than this?

alexpevzner, 2020-04-30:
Hi @Keruspe,

What program do you use to scan (I mean, xsane/simple-scan/scanimage/something else)?

Keruspe, 2020-04-30:
simple-scan

Keruspe, 2020-04-30:
I can provide debug logs or whatever can help if needed

alexpevzner, 2020-04-30:
Did you try the following before scanning:  Document->Preferences->Page Size->A4?

Keruspe, 2020-04-30:
I think I’ve been looking for such a setting but somehow didn’t find it. I’ll retry that later and will keep you posted!

I wish it’s just a pebkac on my side :)

alexpevzner, 2020-04-30:
OK, let me know even if everything is OK. At least, I'll update a list of tested devices :-)

Keruspe, 2020-04-30:
Hmm, safe issue after selecting A4 manually

Keruspe, 2020-04-30:
Tries with A5 and A6 just as random tests:
With A5 I end up with my full document occupying maybe 75% of the doc
With A6 my document is incomplete

alexpevzner, 2020-04-30:
I didn't quite understood, sorry.

A4 works normally, correct?

Can you explain in more details, what's wrong with A5 and A6?

Keruspe, 2020-04-30:
Let me sum this up:

My input document is A4

When configured to scan an A4 document, I end up with my full document occupying ~25% of the final document.
When configured to scan an A5 document, I end up with my full document occupying ~75% of the final document.
When configured to scan an A6 document, I end up with a truncated document.

alexpevzner, 2020-04-30:
OK. May I ask you to send me logs?

To obtain logs, you need to uncomment the following section in the **/etc/sane.d/airscan.conf** file:

    [debug]
    trace = ~/airscan/trace
    enable = true

Logs will be in the ~/airscan/trace directory, which will be created automatically

alexpevzner, 2020-04-30:
Forgot to mention. This is better to send me these logs directly by e-main: pzz@apevzner.com

Keruspe, 2020-04-30:
Done

Keruspe, 2020-04-30:
I can easily test patches or try to debug this if you can point me to where I should be looking

alexpevzner, 2020-04-30:
The log was slightly truncated. Please, try again. And send me the .tar file too.

P.S. I can't understand from Brother specs, is this is A3 or A4 machine?

Keruspe, 2020-04-30:
It supports a « letter » format that is slightly bigger that A4 (a couple
of millimeters) but no A3

alexpevzner, 2020-04-30:
It seems it scans at 200 DPI, though 300 DPI is requested.

Can you play with scanning on different resolutions?

Keruspe, 2020-04-30:
When requesting 600 DPI, the result is even smaller, and when requesting 150 DPI the result is truncated

alexpevzner, 2020-04-30:
This scanner doesn't support 150, it supports only 100, 200, 300 and 600. Here simple-scan lies. So for experiments its better to use xsane, it offers real resolutions.

May I ask you to try your scanner with Vuescan for Linux or with Mopria for Android?

Keruspe, 2020-04-30:
VueScan gives the correct output

alexpevzner, 2020-04-30:
With correct actual resolution?

Keruspe, 2020-04-30:
Seems so, yeah. It's using PDF and not JPG though if that changes anything

Keruspe, 2020-04-30:
Just noticed sane-backends now ship with an escl backend (in 1.0.29) and gave it a try. It works fine too, like vuescan

alexpevzner, 2020-04-30:
I know about sane-escl, I'm actually in contact with its author.

sane-escl blocks in sane_start() until image is available, and then returns from sane_get_parameters() actual image parameters.

sane-airscan returns from sane_start() immediately, so sane_get_parameters() returns expected image parameters. If actual image has a different size that expected (typically it is, +/- 1 pixel, probably due to rounding errors within scanner firmware), then image size is adjusted. This magic allows me to send CANCEL request before scan completion.

So with sane-escl image will look correct, but actual resolution, I believe, will be 200 DPI. You can check by image size, A4 at 300 DPI should be about 2480x3507 pixels.

I'm trying to understand why Brother doesn't honour requested image parameters.

May I ask you to build and test the following sane-airscan branch:

https://github.com/alexpevzner/sane-airscan-wsd

Keruspe, 2020-04-30:
coredump

```
#0  0x00007fffe024a7a6 in netif_addr_get () at ../airscan-netif.c:56
        idx = 6
        ifa = 0x7fffc434be50
        ifp = 0x7fffc434c130
        list = 0x0
        addr = 0x7fffc433b390
#1  0x00007fffe024ff25 in wsdd_netif_update_addresses (initscan=true) at ../airscan-wsdd.c:1141
        addr_list = 0x100000001
        addr = 0x7fffe0242934 <eloop_fdpoll_set_mask+116>
        diff = {added = 0x7fffd5d5f700, removed = 0x7fffd5d5edc0}
#2  0x00007fffe024fd7a in wsdd_start_stop_callback (start=true) at ../airscan-wsdd.c:1207
No locals.
#3  0x00007fffe02422ad in eloop_thread_func (data=0x0) at ../airscan-eloop.c:103
        i = 2
#4  0x00007ffff7e4081d in g_thread_proxy (data=0x774c00) at ../glib-2.64.2/glib/gthread.c:807
        thread = 0x774c00
#5  0x00007ffff6c2154f in start_thread () from /usr/x86_64-pc-linux-gnu/lib/libpthread.so.0
No symbol table info available.
#6  0x00007ffff6ddd7a3 in clone () from /usr/x86_64-pc-linux-gnu/lib/libc.so.6
No symbol table info available.
```

alexpevzner, 2020-04-30:
Oops! Fixed (I hope)

Keruspe, 2020-04-30:
Heh, fixed it too ^^

With the coredump fixed, the wsd branch seems to work fine

alexpevzner, 2020-04-30:
The only difference, sane-airscan sends scan request XML in a readable form, while wsd branch formats it into a long single line. Seems that Brother's XML parser is totally broken.

Don't know, should I backport this fix here. wsd branch will be released in few days by pushing all changes to the legacy sane-airscan.

I would appreciate, if you'll test it in WSD mode too. Your scanner seems to support WSD.

To do so, please add the following lines to configuration file:

[options]
protocol = manual

After that, the device should appear in the list twice, once in eSCL mode and once in WSD.

Keruspe, 2020-05-01:
Works fine in both modes

alexpevzner, 2020-05-01:
Thank a lot for your help!

So I'll add you devices to the list of supported devices in the wsd branch, and will close the issue as soon as wsd branch will be released.

alexpevzner, 2020-05-13:
Fixed in sane-airscan-wsd branch, which has been merged here

closed 2020-05-13 by alexpevzner