alexpevzner/sane-airscan #24: Three working devices

debiantriage, 2020-05-05:
Canon Lide 400 scanner
https://ubuntuforums.org/showthread.php?t=2442546

HP OfficeJet Pro 9010 series
https://bugs.launchpad.net/hplip/+bug/1876607

LaserJet Pro MFP 148fdw
https://answers.launchpad.net/hplip/+question/690428

Regards,

Brian.

closed 2020-05-05 by debiantriage

alexpevzner, 2020-05-05:
Thank a lot!

This is first pure scanner in the list. Previously there were only MFPs.

And this the first OpenSUSE user. I maintained SUSE packages mostly in respect to SUSE who provides us with the OpenSUSE build services, but before this time never got feedback from any SUSE user.

alexpevzner, 2020-05-05:
One more thing. This guy with Canon Lide 400, his scanner probably goes to sleep and stops announcing itself via DNS-SD. This is why discovery works unstable.

May be, adding device to /etc/sane.d/airscan.conf will help him. Also he can try to tweak sleep settings in scanner configuration.

I don't have account on Ubuntu forum to tell it him directly.

debiantriage, 2020-05-06:
I passed your advice on. The OP replied with

> I added device to /etc/sane.d/airscan.conf and also to escl.conf. Now both escl and airscan are always recognized, no matter what I do (rebooting, shutdown/startup, unplugging, plugging etc). So the problem is basically solved. I sometimes get double entries for escl and airscan, and when the scanner is unplugged there is a 'ghost entry' for escl, but these don't bother me. Pixma, which is always recognized, still doesn't work with scanner programs, but I don't really need it as escl and airscan always work.
Will now mark this thread as solved.

alexpevzner, 2020-05-06:
Lets get it working perfectly. "Anything Else Is A Compromise" (c) Miele :-)

With sane-airscan this is possible to disable automatic discovery, and it makes sense to do, if discovery is not reliable:

    [options]
    discovery = disable

Alternatively, if name of manually added device will **exactly** match the name of discovered device, manually added will always win.

And if user wants to use shorter name, that automatic discovery returns, the following trick is also possible:

    [devices]
    Kyocera ECOSYS M2040dn = disable             ; disable the discovered device
    My Kyocera = http://192.168.1.102:9095/eSCL/ ; add it manually under different name

Hotplug obviously will not work with manually configured devices, as sane-airscan doesn't probe these devices until they are actually opened.