alexpevzner/sane-airscan #27: Could airscan-discover find the ipp-over-usb scanner

tangyanli, 2020-05-14:
Hi,
I have a multi-functional printer which supports ipp-over-usb function, and the printer has inside scanner.
When I plug the printer into USB port and set the airprint = on, I can send data to the printer by IPP URI. And when I set the airscan.conf as below, the airscan scan the picture well.
***
[devices]
My Scanner = http://127.0.0.1:60000/eSCL
***

But when I use the airscan-discover tool to find the supported scanner, the tool cannot output the ipp-on-usb scanner information.
Is there some method to discover the ipp-on-usb scanner information?

alexpevzner, 2020-05-14:
Hi,
what is your Linux distro? And what ippusbxd version do you use?

tangyanli, 2020-05-14:
Debian 10 + ippusbxd_1.34-2

tangyanli, 2020-05-14:
Ubuntu20.04 + ippusbxd_1.34-2 has the same phenomenon.

alexpevzner, 2020-05-14:
Please, show me output of the following command:

    avahi-browse -rt -a

For now let it be on Ubuntu

tangyanli, 2020-05-15:
Hi,
When I output the "avahi-browse -rt -a" information, I found the phenomenon has a little difference with yesterday.
When I startup Ubuntu with network cable plugging in, the airscan-discover can find the ipp-over-usb infor; but when I reboot Ubuntu without network cable plugging in, the airscan-discover cannot output the ipp-over-usb infor.

The attachment file is output information for "avahi-browse -rt -a" and "airscan-discover -d"

[Startup_Ubuntu_withNetworkCable.log](https://github.com/alexpevzner/sane-airscan/files/4631786/Startup_Ubuntu_withNetworkCable.log)
[Startup_Ubuntu_withoutNetworkCable.log](https://github.com/alexpevzner/sane-airscan/files/4631787/Startup_Ubuntu_withoutNetworkCable.log)

alexpevzner, 2020-05-15:
Hi,

on Ububtu, the problem seems to be in ippusbxd. I'll investigate it better and will contact you a bit later.

On Debian, there is additional problem with the old version on Avahi. It is explained on ippusbxd README: https://github.com/OpenPrinting/ippusbxd

tangyanli, 2020-05-15:
PPS:
Actually, On Debian, the Avahi is the newest version: avahi-daemon_0.8-1_xx.deb

alexpevzner, 2020-05-15:
Hi,

may I ask you to try to replace ippusbxd with the alternative package:
https://github.com/OpenPrinting/ipp-usb

You probably need to do the following:
1. Install Go compiler
2. Clone and enter https://github.com/OpenPrinting/ipp-usb directory
3. make
4. Uninstall ippusbxd
5. As root, call make install in the ipp-usb directory
7. Unplug and plug back USB cable, or reboot, whatever is simpler

tangyanli, 2020-05-15:
OK，I will test the ipp-usb next week.

tangyanli, 2020-05-18:
Hi,
I failed to build the ipp-usb.

Error info: go: github.com/OpenPrinting/goipp@v0.0.0-20200517195333-da79ff13b3a4: Get "https://proxy.golang.org/github.com/%21open%21printing/goipp/@v/v0.0.0-20200517195333-da79ff13b3a4.mod": dial tcp 172.217.160.113:443: connect: connection refused

I cann't access the 172.217.160.113:443. Does the server still exist now?

alexpevzner, 2020-05-18:
This is Google's server. Probably, some temporary error, or may be some restrictive firewall between you and Google, please try again.

If you will succeed, you actually can try to run it directly from build directory, without installation.

If not, wait a little bit, I'll create ipp-usb binary packages.

tangyanli, 2020-05-18:
I tried many times. So the binary packages would be better.

alexpevzner, 2020-05-18:
OK. At this case it will be tomorrow. I'll let you know.

tangyanli, 2020-05-18:
Hi,
I have build the ipp-usb.
TestOS: Ubuntu20.04

For ippusbxd:
Unplug and plug back USB cable ten times, every time the IPP is advertised, but the eSCL is only advertised 3 times.

For ipp-usb (set mode = standalone):
Also unplug and plug back USB cable ten times,  every time the IPP and eSCL are both advertised.

alexpevzner, 2020-05-18:
Hi,
the packages for ipp-usb are here: https://download.opensuse.org/repositories/home:/pzz/

tangyanli, 2020-05-18:
Got it.
Test result was reported yesterday. Please refer the previous comment.

tangyanli, 2020-05-20:
Hi,
The ipp-usb works really fine.
And may I ask your help to support ARM binary version?

tangyanli, 2020-05-21:
Hi,
This is just a information.
I compiled the airscan and ipp-usb on a ARM PC.
They all work fine.
Thanks.

alexpevzner, 2020-05-21:
ARM binaries for what distro?

tangyanli, 2020-05-21:
Chinese Linux distro, deepin

alexpevzner, 2020-05-21:
@tangyanli,

For binary packaging, I use OpenSUSE build services, which support packaging for many (but not all) Linux targets,

This is a more or less accurate list of targets, supported by OpenSUSE build services: https://en.opensuse.org/openSUSE:Build_Service_supported_build_targets

If Deepin is compatible with one of listed distros and it has ARM target, I can enable it.

Alternatively, if Deepin provides packaging for user-supplied software (as many other distros do), you can try to add ipp-usb there by yourself. It should not be very difficult, an I can assist you doing this in a case of any problems (in assumption that I don't read Chinese :-)). If we will go this way, once we succeed, I can add a reference from the ipp-usb README to your 3rd party repo.

tangyanli, 2020-05-21:
That’s so great!
Deepin is compatible with Debian.
Does the OpenSUSE build services support MIPS target based on Debian?

alexpevzner, 2020-05-21:
For Debian 10 the list is following:
* i586
* x86_64
* ppc
* ppc64
* s390x
* local
* armv6l
* armv7l
* aarch64
* ppc64le
* aarch64_ilp32
* riscv64

So unfortunately, no MIPS.

Go compiler, however, supports MIPS, so outside of SuSE, compilation for MIPS should be possible

tangyanli, 2020-05-21:
What a pity.
From now on, I will build MIPS version.
I may ask your help again :-)

alexpevzner, 2020-05-21:
OK. I think, for now I can close sane-airscan issue, as resolved. If you will have some more questions, you may open a new issue.

ipp-usb is hosted here: https://github.com/OpenPrinting/ipp-usb

closed 2020-05-21 by alexpevzner

tangyanli, 2020-05-21:
All right. Thank you so much!

alexpevzner, 2020-05-21:
You are always welcome!