alexpevzner/sane-airscan #29: Canon Pixma TS 9550

debiantriage, 2020-05-16:
https://forums.linuxmint.com/viewtopic.php?f=51&t=319482

Cheers,

Brian.

alexpevzner, 2020-05-16:
Hi @debiantriage,

can you please ask the user, what does he (or she) sees in the list, eSCL or WSD?

About USB, it really works. I suggest ipp-usb instead of ippusbxd, it works **much** better:
https://github.com/OpenPrinting/ipp-usb

I'll package it in a couple of days, so it will be easy

alexpevzner, 2020-05-16:
P.S., airscan-discover, that now comes with the sane-airscan, is the simplest way to answer the question about eSCL vs WSD

debiantriage, 2020-05-16:
Sorry, I do not understand "list". Is that _scaninage -L_ ? I'll look to airscan-discover to advise the user.

USB worked for me on Debian unstable (but I will check again). It didn't appear to work on Mint 19.1, but it was only a quick test.

Coincidentally, I was about to compile ipp-usb. My ENVY 4520 is not very happy with ippusbxd when it comes to printing. A Debian package would be very useful as I am not very skilled in building software.

Cheers,

Brian.

alexpevzner, 2020-05-16:
scaninage -L is OK. airscan-discover is also OK. List of scanners, shown by xsane on startup is also OK, but it doesn't show a list, when there is only one scanner. simple-scan hides the list somewhere in settings.

So for simplicity, let it be scanimage -L or airscan-discover

ippusbxd has serious problems, and these problems are by design, not by implementation. All these problems are fixed in ipp-usb, it uses very different design, so it works absolutely reliable (this is Till's words, not mine).

I didn't package it before, because I assumed that it is mostly targeted to the distro maintainers, not to the end users. But now I see there is some demand among end users, while distro maintainers are very conservative, so I'll provide packaging

debiantriage, 2020-05-18:
> can you please ask the user, what does he (or she) sees in the list, eSCL or WSD?

No reply as yet. I will try to remember to get the _scanimage -L_ output in future.

> About USB, it really works. I suggest ipp-usb instead of ippusbxd, it works much better: https://github.com/OpenPrinting/ipp-usb

Building a Debian package on unstable turned out to be straightforward. Everything appears to work. BTW, what is the expected format for the device URI?

I intend raising the packaging of ipp-usb on the debian-printing list. The main objection  to it appears to be that it is written in Go and that Chrome OS regards it as a memory hog. I don't see this as an obstacle to having it in Debian if it is more capable than ippusbxd.

Regards,

Brian.

alexpevzner, 2020-05-18:
Uri is typically http://localhost:60000/eSCL for eSCL and http://localhost:60000/ipp/printer for IPP. Port may differ (if it is busy, ipp-usb will try to allocate another one). Once port is allocated, ipp-usb will save it to disk and try to allocate the same port the next time.

Frankly speaking, I don't quite understand ChromeOS reasoning. I've recently measured memory consumption of ipp-usb vs ippusbxd. To my surprise, RSS (Resident Set Size, i.e. amount of memory really used) of ipp-usb is even slightly less. If there are more that 1 devices connected, ipp-usb will handle them all in a single process, adding a small amount of memory per device, while ippusbxd requires a process per device.

Only on-disk space consumption is higher. But does it really matter, when even a low-end SSD has size of tens of gigabytes?

And another thing, Go apps are very safe by default. If, by my mistake, there will be buffer overflow or something similar, process will panic instead of opening security hole.

Also, sandboxed packaging now in trend, where every application comes with its own set of dynamic libraries. In this case, ipp-usb disk consumption will be very comparable to other apps, because it has very little external dependencies (direct dependencies are only libusb and libavahi-client)

debiantriage, 2020-05-18:
> Uri is typically http://localhost:60000/eSCL for eSCL and http://localhost:60000/ipp/printer for IPP. Port may differ (if it is busy, ipp-usb will try to allocate another one). Once port is allocated, ipp-usb will save it to disk and try to allocate the same port the next time.

_lpstat -l -e_ for my ENVY 4500 shows `ipp://ENVY4500%20(USB%201)._ipp._tcp.local/`
`lpstat -t` has `ENVY4500_USB_1_`.

alexpevzner, 2020-05-18:
I was speaking about HTTP-level URLs. I don't know too much about URLs, used internally by CUPS.

I've just packaged ipp-usb for a number of Linux distros, the packages for are here: https://download.opensuse.org/repositories/home:/pzz/

debiantriage, 2020-05-20:
@alexpevzner  requested
> can you please ask the user, what does he (or she) sees in the list, eSCL or WSD?

The user responded with

> device `airscan:e1:Canon TS9500 series' is a eSCL Canon TS9500 series eSCL network scanner
device `airscan:w0:CANON INC. TS9500 series' is a WSD CANON INC. TS9500 series WSD network scanner

closed 2020-05-20 by debiantriage