alexpevzner/sane-airscan #31: thread safe issue in latest version when add device from zeroconf

Ariesfish, 2020-05-22:
Hi alex,

I encountered a segment fault when running latest version on my Debian 10 (Ubuntu 18.04 too).
**I add my device into airscan.conf, and the issue happens. If there's no device configuration in airescan.conf, it seems not happen.**

After check, it crashed in zeroconf functions, where "zeroconf_device_list" seems be modified by other thread thus got NULL pointer error.

And **v0.9.1.8** is OK, the issue may be brought in due to WSD implementation. I am not quite understand entire process about device initialization in zeroconf.

Could you have a  look of it?

alexpevzner, 2020-05-22:
@Ariesfish,

I can't reproduce it. Can you obtain a call stack?

alexpevzner, 2020-05-22:
P.S. I need decoded call stack, because addresses are always different, so I will not be able to decode it by myself. So you will have to build sane-airscan and replace installed libsane-airscan.so.1 with unstripped one that you has build.

If it is not possible, try to enable debuging and protocol trace in the configuration file, and send me generated trace files and console output. I will try to localize the source of problem looking to these logs.

This is unlikely to be related to thread safety. Actually, most of the code runs on a context of a single worker thread, and only API calls run on context of caller, but they are properly interlocked with the worker thread.

Ariesfish, 2020-05-22:
@alexpevzner 
One more information, I got segment fault when plugged in network cable.

And error happens here https://github.com/alexpevzner/sane-airscan/blob/8d9843d260b054bee80b512c3a037b0d77456c5c/airscan-zeroconf.c#L1066
, the name is null.

Now I have left my office, and I will try to reproduce it at home but not be sure, because I haven't scanner on hand. If not reproduced, I'll report you next Monday.

alexpevzner, 2020-05-22:
I hope I've fixed it. New release (0.99.2) was created (it will take some time, though, before binaries will be available at SUSE).

Please, notify me, does this fix works for you, so I'll either investigate deeper or close the issue.

Ariesfish, 2020-05-22:
I didn't reproduce it on my home PC, it seems occur in complex network because there are lots of WSD devices in my office LAN. I need to reproduce it next week, and check the call stack and debug log as you told.

alexpevzner, 2020-05-22:
OK, no problem.

Ariesfish, 2020-05-25:
The bug is fixed in 0.99.2. Now it works very well. Thanks very much.

Ariesfish, 2020-05-25:
Hi alex,

I tested 0.99.2, it works well and I think your fix takes effect.
But when I'm running xsane and scanning some images with my scanner, I encounted another segment fault.
I enabled debug trace and console logging in airscan.conf, and run it with gdb, and got the log when it crashed as below.
[airscan-segfault.log](https://github.com/alexpevzner/sane-airscan/files/4675386/airscan-segfault.log)

It seems that it could not resolve LBP621C and failed to remove related resolver.

alexpevzner, 2020-05-25:
Please try 0.99.3

Ariesfish, 2020-05-26:
It works well with 0.99.3. Thanks very much.

alexpevzner, 2020-05-26:
What is your scanner model? I want to add it to the list, if it is not yet there.

Ariesfish, 2020-05-26:
Canon MF746Cx

alexpevzner, 2020-05-26:
So I close this issue as resolved. Thank you for your feedback!

closed 2020-05-26 by alexpevzner