alexpevzner/sane-airscan #32: RICOH MP C3003: HTTP POST 500 Internal Server Error

dleonard0, 2020-05-26:
Scanner: Ricoh MP C3003 Color Laser Multifunction Printer
Problem: simple-scan says "Failed to scan" after selecting scanner and starting a scan
Log: 
[simple-scan-RICOH-MP-C3003.log](https://github.com/alexpevzner/sane-airscan/files/4681419/simple-scan-RICOH-MP-C3003.log)

$ simple-scan
...
RICOH MP C3003: DEVICE_STM_PROBING->DEVICE_STM_IDLE
RICOH MP C3003: sane_open("w0:RICOH MP C3003"): Success
RICOH MP C3003: sane_start()
RICOH MP C3003: DEVICE_STM_IDLE->DEVICE_STM_SCANNING
RICOH MP C3003: PROTO_OP_SCAN: submitting: attempt=0
RICOH MP C3003: HTTP POST http://10.1.1.172:53002/
RICOH MP C3003: HTTP POST http://10.1.1.172:53002/: 500 Internal Server Error
RICOH MP C3003: PROTO_OP_SCAN: decoding
RICOH MP C3003: fault code: OperationFailed
RICOH MP C3003: PROTO_OP_SCAN: decoded: status="Error during device I/O" next=PROTO_OP_CHECK delay=0
RICOH MP C3003: JOB status=Error during device I/O

In response to a POST <scan:CreateScanJobRequest> the scanner replies 500:

    <SOAP-ENV:Fault>
      <SOAP-ENV:Code>
        <SOAP-ENV:Value>SOAP-ENV:Receiver</SOAP-ENV:Value>
        <SOAP-ENV:Subcode>
          <SOAP-ENV:Value>wscn:OperationFailed</SOAP-ENV:Value>
        </SOAP-ENV:Subcode>
      </SOAP-ENV:Code>
      <SOAP-ENV:Reason>
        <SOAP-ENV:Text xml:lang="en">Service cannot perform the requested operation</SOAP-ENV:Text>
      </SOAP-ENV:Reason>
    </SOAP-ENV:Fault>

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal

$ dpkg -l sane-airscan | grep ^i
ii  sane-airscan   0.99.3+60.1  amd64        SANE backend for MFP and document scanners that

Don't know if this helps at all.. 

On the printer/scanner's configuration,  AirPrint is enabled, user authentication is disabled, and an SSL certificate is active. However there is a section that suggests that SSL is required and I don't know if that is the cause of the error.

> Device Certificate
> You must communicate data via AirPrint (IPP) after applying SSL/TLS. Configure the Certificate 1.
> Confirm the Certificate Status of Certificate 1 on the Device Certificate screen. As necessary, change the settings or upload a certificate file.
> The changes of Device Certificate settings are applied to other communication methods.

alexpevzner, 2020-05-26:
Hi @dleonard0,

your scanner manual requires WSD to be explicitly enabled on device: http://support.ricoh.com/bb_v1oi/pub_e/oi_view/0001047/0001047003/view/scanner/int/0095.htm

Have you already done it?

dleonard0, 2020-05-26:
Oh! that was it! I set Prohibit WSD Scan Command to Do not Prohibit, and now simple-scan works perfectly!
Thanks!

closed 2020-05-26 by dleonard0

alexpevzner, 2020-05-26:
It gave me a chance to add your device to the list of supported devices :-)
Have a nice day, thank you for your feedback!