alexpevzner/sane-airscan #33: xUbuntu_20.04 repository not longer available

zunbeltz, 2020-06-03:
The xUbuntu_20.04 repository has disappear from http://download.opensuse.org/repositories/home:/pzz/.
Is this a temporal error, or there is some serious problem? The driver is working quite well with my HP Envy 5640.

alexpevzner, 2020-06-03:
It probably was a temporary failure at SuSE. Fedora 32 was also unavailable. But now everything seems to be OK.

May I ask you to show me output of the following command:

    scanimage -L

I want to know what protocol is in use with your device, to update my table

zunbeltz, 2020-06-04:
HI,

It is working now again.
Here is the output

device `airscan:e0:HP ENVY 5640 series [2DC3DB]' is a eSCL HP ENVY 5640
series [2DC3DB] eSCL network scanner
device `hpaio:/net/envy_5640_series?ip=192.168.0.10&queue=false' is a
Hewlett-Packard envy_5640_series all-in-one

Thanks for your work.

Zunbeltz

alexpevzner, 2020-06-04:
Hi @zunbeltz,

thank you for your feedback!

closed 2020-06-04 by alexpevzner