alexpevzner/sane-airscan #34: Canon TR7500 Series  Error during device I/O

ubuergi, 2020-06-07:
Hi,
there seem to be invalid arguments in the POST request when scanning (c.f. [scanimage-CANON-INC.-TR7500-series.log](https://github.com/alexpevzner/sane-airscan/files/4741765/scanimage-CANON-INC.-TR7500-series.log)). The scanner actually starts scanning but scanimage fails:

$ scanimage > tst
scanimage: sane_read: Error during device I/O

I'm happy to provide more infos if need.

alexpevzner, 2020-06-07:
Hi @ubuergi,

may I ask you to fetch latest code from github, rebuild it and retry?

Note, I usually don't reinstall it when debugging, but simply replace installed `libsane-airscan.so.1` with the symbolic link to the `libsane-airscan.so.1` in my build directory. Path to installed `libsane-airscan.so.1` depends on your distro. Typically, this is `/usr/lib64/sane/libsane-airscan.so.1` for Fedora and `/usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1` for Ububtu/Debian

ubuergi, 2020-06-07:
Hi @alexpevzner,

Built from the latest source, it works flawlessly! Thx!

alexpevzner, 2020-06-07:
I've created a next, 0.99.4 release, which includes this fix.
Thank you for your feedback!

alexpevzner, 2020-06-07:
Resolved - closed

closed 2020-06-07 by alexpevzner