alexpevzner/sane-airscan #36: Canon MF746Cx Error during device I/O

tangyanli, 2020-06-11:
Hi, alexpevzner

I use the MF746Cx scannner by WSD protocol. But the sand_read() fails with error "RetrieveImageRequest: invalid response".

Could you please have a look?

[xsane-Canon-MF745C-746C.log](https://github.com/alexpevzner/sane-airscan/files/4762595/xsane-Canon-MF745C-746C.log)

tangyanli, 2020-06-11:
Canon iR-ADV 4545 has the same issue: "RetrieveImageRequest: invalid response".

Here is the log
[Uploading xsane-Canon-iR-ADV-4545-4551.log…]()

If it's necessary, I can supply more log file.

alexpevzner, 2020-06-11:
Hi @tangyanli,

please try the newest version - I hope I've fixed it

tangyanli, 2020-06-11:
alexpevzner,

I just leave my office, and I will test it on tomorrow early morning.

tangyanli, 2020-06-12:
Hi, alexpevzner

Fixed.
And thank you so much :-)

alexpevzner, 2020-06-12:
You are always welcome! :-)

One more question. I want to update my table of tested devices. There are two devices in this issue:
* Canon MF745C/746C
* Canon iR-ADV 4545/4551

Is it correct, that both are WSD-only, without eSCL?

P.S. If you yse automatic protocol selection, which is default, you will see WSD only if eSCL is not supported. In the manual mode you will see both protocols.

tangyanli, 2020-06-12:
Hi, alexpevzner

For the two devices, the eSCL and WSD are both works.
I see, the eSCL is better than WSD. For example, eSCL supports ADF Duplex, but WSD doesn't support the ADF Duplex.

alexpevzner, 2020-06-12:
ADF Duplex should work, if supported by device. I've tested it with my Kyocera, everything was OK. And your cannons also indicate ADF Duplex support in their logs.

Let's debug it on your cannons. What's wrong with it, it is not offered in the list of available modes, or it doesn't actually?

tangyanli, 2020-06-12:
For MF745C, the xsane lists the ADF duplex, and the ADF duplex works.
But the xsane doesn’t list ADF duplex for iR-ADV 4545.

tangyanli, 2020-06-12:
I check the log. Actually, iR-ADV 4545 does’t support ADF duplex. But it supports ADF duplex by eSCL. I thought this should be device specification.

alexpevzner, 2020-06-12:
Yes, I see:

    <wscn:ADFSupportsDuplex>false</wscn:ADFSupportsDuplex>

This is high-end device, they have a lot of configuration options. There is a small chance that ADF-Duplex can be enabled somewhere in scanner configuration, separately for WSD

tangyanli, 2020-06-12:
I may read the device user manual carefully :-)
So this issue is fixed, and I prefer to use the scanner by eSCL. It works so well.

alexpevzner, 2020-06-12:
OK. I've created 0.99.5 release with these fixes.

Thank you for your feedback!

closed 2020-06-12 by alexpevzner

tangyanli, 2020-06-12:
Hi, alexpevzner

It would be helpful if the 0.99.5 amd64 packages on ubuntu16.04 and Debian10 are also supplied in the repositories :-)
And thanks as always.

alexpevzner, 2020-06-12:
When you asked me to add ARM64 for `ipp-usb`, I added it globally for all packages, so for Debian 10 it rebuilds automatically.

Regarding Ubuntu 16.04, there is a same problem as for `ipp-usb`: SuSE virtual machine that builds for this combination of OS and CPU misses many essential packages (like `pkg-config`, for example, and much more), and it's not in my power to change it.

tangyanli, 2020-06-13:
hi, alexpevzner

I remember the ARM64 build problem (and all other things) you told me.
It’s just, I see the packages for AMD64 are still 0.99.4, not ask for ARM64 package...