alexpevzner/sane-airscan #37: Canon Scanner 2140 USB 3.0 is not working with RedHat 6.7

nomeemirza, 2020-06-14:
Hi Guys,

I have been struggling with this error "no sane device found". Red hat workstation is not able to establish two way communication.

Here is my configuration, 
I installed  sane packages into Linux Workstation
yum -y install sane-backends-1.0.24-100.el6.x86_64.rpm
sane-backends-libs-1.0.24-100.el6.x86_64.rpm
sane-backends-libs-1.0.24-100.el6.i386.rpm

Added the following into /etc/sane.d/canon_dr.conf
# DRG-2140
usb 0x1083 0x166e

Added into  /etc/udev.d/rules.d/50-wic.rules

ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="1083", MODE:="0666"

I really appreciate response.
Thanks,
Noman.

alexpevzner, 2020-06-14:
Hi @nomeemirza,

I'm not the author of entire SANE, I'm just the author of this (`sane-airscan`) backend.

To speak with SANE team, you should write to the `sane-devel` mail list: https://alioth-lists.debian.net/cgi-bin/mailman/listinfo/sane-devel

My backend will not work with your scanner via USB, but there is some chance that it will work over network. Please note, currently my driver is not included into the SANE package, you may download binary packages for many distros from here: https://download.opensuse.org/repositories/home:/pzz/

There is `airscan-discover` utility that comes with my backend. If it can find your scanner on a network, most likely my backend will work with it

nomeemirza, 2020-06-15:
Thank you so much for your response. I have sent an email to Sane Team.

nomeemirza, 2020-06-15:
Once I get resolve this issue I will post here.

alexpevzner, 2020-06-15:
You may still want to try my backend, using network connection

closed 2020-06-29 by alexpevzner