alexpevzner/sane-airscan #38: HP MFP m28w - unable to scan in 600dpi

jdtanner, 2020-06-18:
Hello,

This works great, but my HP MFP m28w is unable to scan in 600dpi (maximum resolution supprted). 
```
scanimage --format=png --output-file test.png --progress --resolution=600
scanimage: sane_start: Error during device I/O
```
dpi of 300 works fine. 
[ScannerCapabilities.txt](https://github.com/alexpevzner/sane-airscan/files/4800564/ScannerCapabilities.txt)

jdtanner, 2020-06-18:
Should have stated that this is the case for the latest version of HPLIP as well usinf the hpaio protocol, suggesting there might be something else going on here; timeout maybe?

jdtanner, 2020-06-18:
So, fiddling with the scanner/printer settings. If ESCL is turned of in the settings, then WSD will be picked up by Simple Scan. Using the WSD protocol works at 600dpi for this scanner! Not sure if that helps find the root cause of the problem, but there is something in the ESCL protocol that prevents scanning at 600dpi which isn;t present in the WSD protocol.

alexpevzner, 2020-06-18:
Hi!

May I ask you to gather protocol trace logs?

To do so, you need to uncomment the following lines in the `/etc/sane.d/airscan.conf`:

    [debug]
    trace = ~/airscan/trace
    enable = true

Logs will be in the `~/airscan/trace` directory, which will be created automatically

jdtanner, 2020-06-18:
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/4800669/scanimage-zeroconf.log)

jdtanner, 2020-06-18:
Hopefully the above will help. I tried scanning at 300dpi (worked) and 600dpi (failed).

alexpevzner, 2020-06-18:
I need all logs, not only zeroconf.log .tar currently not needed

jdtanner, 2020-06-18:
Sorry, please try the attached.
[scanimage-HP-LaserJet-MFP-M28w-(B45C29).log](https://github.com/alexpevzner/sane-airscan/files/4800709/scanimage-HP-LaserJet-MFP-M28w-.B45C29.log)

jdtanner, 2020-06-18:
These are the logs when the scan was successful.
[scanimage-HP-LaserJet-MFP-M28w-(B45C29).log](https://github.com/alexpevzner/sane-airscan/files/4800726/scanimage-HP-LaserJet-MFP-M28w-.B45C29.log)
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/4800727/scanimage-zeroconf.log)

alexpevzner, 2020-06-18:
The next question: will it be possible for you to build sane-airscan by yourself?

jdtanner, 2020-06-18:
I can have a go :)

jdtanner, 2020-06-18:
Worked!
[scanimage-HP-LaserJet-MFP-M28w-(B45C29).log](https://github.com/alexpevzner/sane-airscan/files/4800839/scanimage-HP-LaserJet-MFP-M28w-.B45C29.log)

alexpevzner, 2020-06-18:
Please retest the latest git version, and if everything is OK, I'll close the issue

jdtanner, 2020-06-19:
Morning, latest git works great. Thanks so much for the speedy resolution to this problem :)

alexpevzner, 2020-06-19:
Thank you too! Without your feedback it would be impossible for this project to reach its current state.

Closed - resolved.

This fix will be in release `0.99.7` (now building at SuSE)

closed 2020-06-19 by alexpevzner

BetaConnector, 2020-09-25:
@alexpevzner just as a note:
For my M28w I had to set ESCL_LOAD_RETRY_ATTEMPTS to 30 in order to scan an image with 600dpi.

ESCL_LOAD_RETRY_ATTEMPTS 20 leads to the following error:
```log
HP LaserJet MFP M28w (AE7047): HTTP using timeout: 30000 ms
HP LaserJet MFP M28w (AE7047): HTTP resolving 192.168.1.120 80
HP LaserJet MFP M28w (AE7047): HTTP trying 192.168.1.120:80
HP LaserJet MFP M28w (AE7047): HTTP DELETE http://192.168.1.120:80/eSCL/ScanJobs/xllievz5-xfsr-ibap-1019-9mtanigj: got response headers (200)
HP LaserJet MFP M28w (AE7047): HTTP DELETE http://192.168.1.120:80/eSCL/ScanJobs/xllievz5-xfsr-ibap-1019-9mtanigj: 200 OK
HP LaserJet MFP M28w (AE7047): HTTP DELETE http://192.168.1.120:80/eSCL/ScanJobs/xllievz5-xfsr-ibap-1019-9mtanigj: OK
HP LaserJet MFP M28w (AE7047): PROTO_OP_CLEANUP: decoding
HP LaserJet MFP M28w (AE7047): PROTO_OP_CLEANUP: decoded: status="Success" next=PROTO_OP_FINISH delay=0
HP LaserJet MFP M28w (AE7047): DEVICE_STM_CLEANUP->DEVICE_STM_DONE
HP LaserJet MFP M28w (AE7047): DEVICE_STM_DONE->DEVICE_STM_IDLE
HP LaserJet MFP M28w (AE7047): sane_read(): Device busy
Reading data: [|                                                                                                                                                        ] 0%  0 B   
error: No data read.
Closing device.
HP LaserJet MFP M28w (AE7047): sane_close()
HP LaserJet MFP M28w (AE7047): DEVICE_STM_IDLE->DEVICE_STM_CLOSED
HP LaserJet MFP M28w (AE7047): removed from device table
HP LaserJet MFP M28w (AE7047): closed protocol "eSCL"
HP LaserJet MFP M28w (AE7047): closed decoder: jpeg
HP LaserJet MFP M28w (AE7047): device destroyed
```

alexpevzner, 2020-09-26:
@BetaConnector,

thank a lot! I've increased the constant, it will be included into the next release.

BetaConnector, 2020-09-26:
Just being curious - do you know the reason why especially this printer needs so many attempts?

Thaks for your work here...

alexpevzner, 2020-09-26:
@BetaConnector,

Logs, attached to this case, shows, that at 300 DPI printer successfully returns an image after 5 attempts, while at 600 DPI 10 attempts is not enough (12 was enough, so I assumed that 20 is a reasonable high value).

Your device is slower for unknown for me reason, but you've investigated by yourself and 30 was enough for you.

Most of the printers doesn't return HTTP 503 in the situation like this - they simply block in HTTP GET request for quite a long time.