alexpevzner/sane-airscan #39: Brother MFC-L2710DW ADF issue

seife, 2020-06-22:
My Brother MFC-L2170DW works almost perfect with sane-airscan, in eSCL mode.

However, when scanning multiple pages from ADF, it most of the time aborts after 2 or 3 pages, with 

```
seife@server:~> scanimage --format=jpg --resolution 200 --source ADF -t 0 -l 0 -y 296 -x 215 --batch=out%d.jpgScanning infinity pages, incrementing by 1, numbering from 1
Scanning page 1
Scanned page 1. (scanner status = 5)
Scanning page 2
Scanned page 2. (scanner status = 5)
Scanning page 3
scanimage: sane_start: Document feeder out of documents
Batch terminated, 2 pages scanned
```

This is on openSUSE Leap 15.2 Beta, no matter if I try sane-airscan or sane-airscan-ng packages from OBS.

To debug the issue, I wrote a simple scanning tool in Python: https://github.com/seife/airscan-simple
When scanning directly to PDF format, the output is all pages in one file, which always works fine, but the inital implementation had the same problem for multi-file (jpg format) scanning, aborting after a few pages.
I also found out, that scanning with resolution 600dpi seems to work more reliably.
Finally I found, that adding a short wait before retrieving /NextDocument fixed the issue. Maybe the brother firmware is not fast enough and immediately requesting the next page results in a 404.
this line https://github.com/seife/airscan-simple/blob/master/airscan-simple.py#L90 fixed it for me.
I added a one-second-sleep, but scanimage works with 600dpi (I guess the conversion takes slightly longer), I'd suspect that even a shorter wait would suffice. But since the scanning takes longer than a second anyway, this seems to be fine ;-)

alexpevzner, 2020-06-22:
Hi!

Thank you for a lot of information you've gathered. I also want to look to protocol trace. To obtain it, please uncomment the following lines in the `/etc/sane.d/airscan.conf`:

    [debug]
    trace = ~/airscan/trace
    enable = true

Traces will be in the `~/airscan/trace` directory, which will be created automatically. `.tar` files are not needed for now.

You may either attach them here, or send me directly to pzz@apevzner.com

If you decide to send it directly, please notify me here too in case e-mail will fall to the spam folder

seife, 2020-06-22:
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/4812435/scanimage-zeroconf.log)
[scanimage-Brother-MFC-L2710DW-series.log](https://github.com/alexpevzner/sane-airscan/files/4812437/scanimage-Brother-MFC-L2710DW-series.log)

Traces with debug.
Note I'll be offline next two days, so responses will lag a bit ;-)
Thanks for caring.

alexpevzner, 2020-06-22:
How many pages left in the feeder?

seife, 2020-06-22:
5 or more
It scans the first two, then the error appears, then the rest of the pages are just "spooled through" without actual scanning.
The exact same happens if, with my python script I remove the "sleep 1 second" thing.
It always scans all pages (10 or more) with PDF mode, which retrieves all pages with one http://.../NextDocument request.

I could try what a "ScannerStatus" request shows before the NextDocument request, maybe there is a "not ready" message or something. But only on Wednesday as I'm out of office now ;-)

alexpevzner, 2020-06-22:
ScannerStatus very cleanly shows that ADF is out of documents:

    <pwg:State>Processing</pwg:State>
    <scan:AdfState>ScannerAdfEmpty</scan:AdfState>

Obviously it is firmware bug.

Adding 1-second delay will slow-down scanning on devices that can do >60 pages per minute. Here I need to think a little bit...

alexpevzner, 2020-06-22:
I've added a delay with upper bound based on a time taken by the preceding NextDocument request, so Brother should work, while very fast scanners will not be slowed down,

Please fetch the latest source and retest, before I'll close this issue

seife, 2020-06-23:
I had to test this immediately after returning home ;-)
It works beautifully, scanned 16 pages without problem, and the idea with the dynamic delay depending on scanner speed is just brilliant (I'm a bit envious that I did not think of that ;-))
Thanks for fixing this.

For the supported devices list:
Brother MFC-L2710-DW works in eSCL mode, it does not work in WSD mode.

alexpevzner, 2020-06-23:
Idea with adaptive delay is obvious for everybody who was writing networking code for all his or her professional life :-)

What is wrong with WSD? It is not offered in the list or offered, but it doesn't work?

According to zeroconf log, your device supports WSD.  However, if `protocol = auto` in the `airscan.conf`, the driver will choose protocol automatically, preferring eSCL over WSD, if it has a choice. If you set `protocol = manual`, both variants should appear in the list.

seife, 2020-06-23:
It is offered and when selected manually it did not work (don't know the exact error).
Now I tried it again and it works, both ADF and WSD.

The difference is only last time I used sane-airscan-ng-1590598308-0.x86_64.rpm, today I used sane-airscan-ng-1592844898-0.x86_64.rpm (I'm building those in my $HOME in OBS for ease of installing, and for testing I built them locally, thus the "-0" build number). The old version is commit 8392745727, the new version is from today.

But anyway, it seems fixed with WSD now, too :-)

alexpevzner, 2020-06-23:
-ng currently is a little bit outdated, and I don't very often push there fixes from the stable branch.

So I think I can close the issue, and add your device to the list.

BTW, I've just created `0.99.8` release with this fix included

closed 2020-06-23 by alexpevzner