alexpevzner/sane-airscan #41: Incorrect FSF address in LICENSE and COPYING

zdohnal, 2020-06-24:
Hi,

during package review rpmlint found out that LICENSE and COPYING files contain the old address of FSF:

```
59 Temple Place, Suite 330, Boston, MA  02111-1307  USA
```

The current address is:

```
51 Franklin Street, Fifth Floor, Boston, MA  02110-1301, USA
```

According [common tips](https://fedoraproject.org/wiki/Common_Rpmlint_issues#incorrect-fsf-address) it should be reported to the project at least.
If it is an issue in your opinion, then there is a [current text](https://www.gnu.org/licenses/old-licenses/gpl-2.0.txt) of GPLv2 license.

alexpevzner, 2020-06-24:
Fixed, thanks.

BTW, the same is true for the sane-backends project. I've borrowed these texts from there :-)

closed 2020-06-24 by alexpevzner

zdohnal, 2020-06-24:
Well, sane-backends went through package review long time ago, so no one got to it after that :D . But since sane-airscan will be a new package, I would like to have it in a top shape :) .
I'll report it to SANE too.