alexpevzner/sane-airscan #43: Various warnings from Covscan

zdohnal, 2020-06-24:
Hi,

during the package creation I ran RH internal coverity scan called covscan on sane-airscan. It is basically aggregation of static analysis tools or compilers with static analysis, and with many compilation warnings enabled e.g. gcc, cppcheck, shellcheck, clang etc.

We have a filter for issues whose one of our teams marked as important - e.g. memleaks, double frees, buffer overflows. Covscan didn't show any of those errors, but several not critical ones - here is 
[the log](https://github.com/alexpevzner/sane-airscan/files/4825094/covscan.txt) .

I'm filing the issue to let you know - if you mark it as false positive or unimportant, it's okay.

alexpevzner, 2020-06-24:
I like static analysers, so I'll review this report carefully. However it will take some time. Thanks!

alexpevzner, 2020-06-24:
Please, retest. Not all of these warnings were false-positive :-)

zdohnal, 2020-06-26:
Half of warnings are gone, here are the current result (if I didn't botched pulling the current sources from github) 
[covscan.log](https://github.com/alexpevzner/sane-airscan/files/4835485/covscan.log)

alexpevzner, 2020-06-29:
Seems that I've killed them all.

In most cases, clang analyser interprets C pointer aliasing rules stricter that language itself. For example, if I need to purge a linked list of structures of particular type, I can loop until list is empty, fetching first element and calling its "destructor", while destructor takes care of removing the element from the list (list is accessible by the pointer from the element). Clang correctly guesses that "destructor" frees the object, but cannot guess that list get modified this way and claims that memory can be used after freed.

I believe this is correct, because C allows for pointers of the same type to be aliased. Hope, clang code generator treats pointer aliasing in more pessimistic way that code analyser :-)

One place I can't understand though, and will appreciate if you'll have a look.

In the `airscan-http.c` I had to `#ifdef` out `http_uri_free()` function body (line 119), because otherwise analyser claims that memory is used after freed in the function `http_query_free()` at line 811:

    http_data_unref(q->cached->request_data);

The problematic line in the `http_uri_free()` is following, commenting it out helps equally well as `#ifdef`ing entire function body:

    g_free(uri);

If I comment out most of the body of the `http_query_free()`, analyser will complain to `g_free(q)` at the end of the `http_query_free()` function, so seems it somehow thinks that `q` and `q->uri` are related memory regions, which is not true

zdohnal, 2020-06-30:
> The problematic line in the `http_uri_free()` is following, commenting it out helps equally well as `#ifdef`ing entire function body:
>
> ```
> g_free(uri);
> ```
>
> If I comment out most of the body of the [...]

It seems as a false positive, but I'll CC Kamil, who is covscan maintainer.

@kdudka Kamil, would you mind helping us with covscan results from the last comment?

What warning are you speaking about? I see only two warnings about use-after-free - the first I don't see where the assigning the next value happens. For the second, I don't know about the second.

As I told in the initial comment, there can be false positives in the report - you can just tell me and I'll ignore them next time, there's no need to rewrite the code for false positive.

I ran covscan on the current master and here is the only warning:

```
Error: CPPCHECK_WARNING:
sane-airscan-0.99.8/airscan-trace.c:217: warning[objectIndex]: The address of local variable 'hdr' might be accessed at non-zero index.
#  215|       chsum = 0;
#  216|       for (i = 0; i < sizeof(hdr); i ++) {
#  217|->         chsum += ((char*) &hdr)[i];
#  218|       }
#  219|       sprintf(hdr.checksum, "%6.6o", chsum & 0777777);
```

kdudka, 2020-06-30:
Yes, the clang's use-after-free seems to be a false positive.  As I understand it, commit b912bcc955401b2abe16dda5ebd868ef154ace86 fixes it?

The `warning[objectIndex]` seems to be a bug in cppcheck that it ignores the `(char*)` cast and sees the code as `(&hdr)[i]`, which would indeed be a bug as described in the diagnostic message.

alexpevzner, 2020-06-30:
@kdudka 

I want to ask you too look to the commit 1de0479

kdudka, 2020-06-30:
It seems to be a bug in clang.  I was able to isolate a minimal example out of it:
```c
#include <stddef.h>
#include <stdlib.h>

#define OUTER_STRUCT(member_p, struct_t, field)                            \
    ((struct_t*)((char*)(member_p) - ((ptrdiff_t) &(((struct_t*) 0)->field))))

struct ll_node {
    void               *etc;
    struct ll_node     *ll_next;
};

struct http_query {
    void               *ptr;
    void               *uri;
    struct ll_node      list_node;
};

struct ll_node http_query_list;

void main()
{
#ifdef RUNNABLE
    struct http_query *obj = malloc(sizeof *obj);
    http_query_list.ll_next = &obj->list_node;
    obj->uri = NULL;
#endif
    struct ll_node *node = http_query_list.ll_next;
    node->ll_next = node;

    struct http_query *q = OUTER_STRUCT(node, struct http_query, list_node);
    free(q->uri);
    free(q);
}
```

As you say, clang thinks that `q` and `q->uri` both refer to the same object:
```
% clang --analyze -Xanalyzer -analyzer-output=text clang-bug.c
clang-bug.c:32:5: warning: Attempt to free released memory
    free(q);
    ^~~~~~~
clang-bug.c:31:5: note: Memory is released
    free(q->uri);
    ^~~~~~~~~~~~
clang-bug.c:32:5: note: Attempt to free released memory
    free(q);
    ^~~~~~~
1 warning generated.
```

If `q` is explicitly allocated and `q->uri` initialized, it results in even more confusing diagnostic:
```
% clang --analyze -Xanalyzer -analyzer-output=text clang-bug.c -DRUNNABLE
clang-bug.c:31:5: warning: Argument to free() is offset by 16 bytes from the start of memory allocated by malloc()
    free(q->uri);
    ^    ~~~~~~
clang-bug.c:31:5: note: Argument to free() is offset by 16 bytes from the start of memory allocated by malloc()
    free(q->uri);
    ^    ~~~~~~
1 warning generated.
```
Yet valgrind does not see any problem in the minimal example:
```
% gcc clang-bug.c -DRUNNABLE
% valgrind ./a.out
==28260== Memcheck, a memory error detector
==28260== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==28260== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==28260== Command: ./a.out
==28260== 
==28260== 
==28260== HEAP SUMMARY:
==28260==     in use at exit: 0 bytes in 0 blocks
==28260==   total heap usage: 1 allocs, 1 frees, 32 bytes allocated
==28260== 
==28260== All heap blocks were freed -- no leaks are possible
==28260== 
==28260== For lists of detected and suppressed errors, rerun with: -s
==28260== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

alexpevzner, 2020-07-05:
Hi,

this issue can probably be closed as "resolved", correct?

kdudka, 2020-07-07:
Fine by me.  A colleague of mine is going to have a look at the clang bug ... but it will take time and does not block you anyway.

alexpevzner, 2020-07-07:
OK, thank you!

closed 2020-07-07 by alexpevzner

zdohnal, 2020-07-13:
Hi @alexpevzner ,

I'm sorry for no response, I was on vacation for a week and half - yes, it is as Kamil said, the issue is resolved now from sane-airscan side.

@kdudka thank you for the help and @alexpevzner thank you for looking into issues and fixing them!

alexpevzner, 2020-07-20:
Hi @zdohnal,

how are you?

Are you blocked by something with your effort of `sane-airscan` integration into Fedora, or just busy for now?

zdohnal, 2020-07-21:
Hi @alexpevzner ,

sorry, I was busy till yesterday with stuff which was gathered during my vacation. I'll put review request today (hopefully).

zdohnal, 2020-07-21:
@alexpevzner 

Here is srpm and spec file https://zdohnal.fedorapeople.org/sane-airscan/ for review. Please let me know if you have any further concerns.
As I told before, I divided the project into main package - sane-airscan (contains airscan-discover) - and subpackage - libsane-airscan (contains shared library). When you tell somebode to install sane-airscan, both packages are installed, otherwise libsane-airscan will be brought in when sane-backends-drivers-scanners will be installed (this hasn't been implemented yet, I'll add it to the sane-backends subpackage after sane-airscan will get into repos.)

alexpevzner, 2020-07-21:
@zdohnal 

Looks pretty good. Thank a lot!