alexpevzner/sane-airscan #45: How to find out that my device is unsupported?

zdohnal, 2020-06-29:
Hi,

I have HP LaserJet M1536 dnf multifunction device, which is available via mDNS with service type ```_scanner._tcp```.

I ran ```scanimage -L```, but my device wasn't found - here are the logs 
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/4825366/scanimage-zeroconf.log) .

It seems my device doesn't support WSD or ESCL, am I correct? It would be possible to add debug logging there? See the pull request

alexpevzner, 2020-06-29:
This device supports WSD protocol, but offers it for printing only, not for scanning

zdohnal, 2020-06-30:
Damn, I hoped the device would have been capable of driverless scanning :( ... so I would need to go to the office to test sane-airscan properly or buy a new device.
So I still need to rely on HP to release binary plugin for scanning.

Thank you for the information!

closed 2020-06-30 by zdohnal

alexpevzner, 2020-06-30:
This is actually very strange, that HP bothered to add WSD to this device, but didn't add WS-Scan. WS-Print by itself looks quite useless, Windows perfectly prints to any IPP printer.

May be, there is some well-hidden switch on device's web console, that enables WS-Scan?

zdohnal, 2020-06-30:
I'll check it again, but I think I see only WS-discovery, Bonjour, SLP, LPD, LLMNR, DHCPv6 and IPv4. Only thing I have disabled is IPv6 support.

alexpevzner, 2020-06-30:
Well, although chances that it will work are very low, you may try to hack sane-airscan a little bit:

In `airscan-wsdd.c`, function `wsdd_message_parse_endpoint`, after lines 590-591:

    msg->is_scanner = !!strstr(val, "ScanDeviceType");
    msg->is_printer = !!strstr(val, "PrintDeviceType");

Add something like `msg->is_scanner |= msg->is_printer`