alexpevzner/sane-airscan #47: How can i disable sane-airscan?

Kowa92, 2020-06-29:
Hi,
i wanted to know how i could do to disable this backend when i run the `saneairscan -L` command. Is there an option in his config file?
Also, how can i uninstall this package once i have compiled using the binaries with `make install`?
Thanks

alexpevzner, 2020-06-29:
On 6/29/20 9:14 PM, Kowa92 wrote:

> i wanted to know how i could do to disable this backend when i run the 
> |saneairscan -L| command. Is there an option in his config file?

Comment out (with the # character) the airscan line in the 
/etc/sane.d/dll.d/airscan file

> Also, how can i uninstall this package once i have compiled using the 
> binaries with |make install|?

I've just added |make uninstal| -- use with care :-)

Note, it intentionally leaves the /etc/sane.d/airscan.conf file, so you 
will have to remove it manually, if you don't need it anymore.

-- 

	Wishes, Alexander Pevzner (pzz@apevzner.com)

Kowa92, 2020-06-30:
Great! :)

> I've just added |make uninstal| -- use with care :-) Note, it intentionally leaves the /etc/sane.d/airscan.conf file, so you will have to remove it manually, if you don't need it anymore.

It starts from 0.99.8 version of sane-airscan, right?

alexpevzner, 2020-06-30:
No, it will start from not yet released 0.99.9. But if you build from sources, just upgrade sources, and `make uninstall` will work (if you use precompiled binary packages, you can always uninstall, using your package manager)

alexpevzner, 2020-06-30:
BTW, what is wrong with sane-airscan, so disabling and uninstalling it is a major feature for you? :-)

Kowa92, 2020-07-01:
> No, it will start from not yet released 0.99.9. But if you build from sources, just upgrade sources, and `make uninstall` will work (if you use precompiled binary packages, you can always uninstall, using your package manager)

Ok, which is the command for upgrade sources using git? (I don't know git well.. sry)

Unfortunately, i cannot use precompiled binary packages because i installed sane-airscan on RaspberryPi..

Kowa92, 2020-07-01:
> BTW, what is wrong with sane-airscan, so disabling and uninstalling it is a major feature for you? :-)

There is nothing wrong .. I simply don't want to have one driver the next day for a single device (they could be confusing) :)

alexpevzner, 2020-07-01:
`git pull`

I've added Debian 10/armv7 build. Is it enough for Raspberry Pi?

Kowa92, 2020-07-02:
You have just added it! :O Some days ago there wasn't an  _armhf.deb package! :)

Good work man! Thanks for all! ^^

closed 2020-07-02 by Kowa92

alexpevzner, 2020-07-02:
I've also added Raspbian 9 and 10. Have no idea what is better for Raspberry.

There are a lot of targets I can add at OBS, but I don't want to overload their build servers without need, so among obviously popular targets, I'm adding others on demand.

Kowa92, 2020-07-04:
What do you mean with OBS?

alexpevzner, 2020-07-04:
SUSE Open Build. Service