alexpevzner/sane-airscan #49: Brother MFC-L2720DW image distortion with xsane if left offset is not 0

realnickel, 2020-07-08:
My Brother MFC-L2720DW is detected by airscan in WSD mode.
When scanning a full flatbed work field the image is fine, but if a cropped area is defined distortion occurs.
Distortion appears only if offset from the left side is non zero. If any other offsets are set (top, bottom or right sides) the image is always fine.

How to reproduce: xsane main menu -> standard options; geometry -> top-left x; 0.0 -> any other value; scan 
If some debug log are needed I'm willing to provide them.

alexpevzner, 2020-07-08:
Hi,

I need an example of broken image

realnickel, 2020-07-08:
[image]

[image]

[image]

alexpevzner, 2020-07-08:
Thank a lot!

I think, I have enough information to investigate this issue. I will come back to you later, when I will have some update.

realnickel, 2020-07-08:
Thank you!
Recently I discovered that the same issue is reappearing for Samsung M337x 387x 407x Series device...
So might it be worth renaming the issue in more generic way, say "image distortion with xsane if left offset is not 0" ?

alexpevzner, 2020-07-08:
Fixed in Git, next release coming soon.

This bug is hardware-independent, but it was hard to reproduce with eSCL, while easy to reproduce with WSD. In eSCL mode, image clipping in most cases handled by hardware. However in WSD mode, as a workaroung for hardware that doesn't clip properly, it is implemented purely in software, and implementation was broken.

Thank you for reporting this bug. What is interesting, nobody was complaining before :-)

realnickel, 2020-07-09:
Thank you for your rapid reaction!
Now left offset is treated correctly with no distortion, but a new bug appeared: if normal scanning is performed without prior "preview scanning" xsane crashes.
Offset settings are the same as were mentioned on previous screenshots.
File types checked: jpeg and postscript.
Please refer to attached xsane logs.

[xsane_preview.log](https://github.com/alexpevzner/sane-airscan/files/4895326/xsane_preview.log)
[xsane_not_previewed_jpeg.log](https://github.com/alexpevzner/sane-airscan/files/4895327/xsane_not_previewed_jpeg.log)
[xsane_not_previewed.log](https://github.com/alexpevzner/sane-airscan/files/4895328/xsane_not_previewed.log)

alexpevzner, 2020-07-09:
It was my stupid mistake. Please, retest.

alexpevzner, 2020-07-09:
Fixed, please retest

realnickel, 2020-07-10:
Works perfect (at least for a while :) ).
I didn't manage to reappear any of issues mentioned above.
Thanks a lot!

closed 2020-07-10 by realnickel

alexpevzner, 2020-07-10:
Great!

I see you've tested with several devices. May I know their names and scan modes (eSCL/WSD) to update my list?

realnickel, 2020-07-10:
Yes, ofcourse!
The first is mentioned Brother MFC-L2720DWR (WSD mode) which was tested in both flatbed and ADF modes. The only limitation : it manifests and scans only in Color mode (no B/W or grayscale) despite of its vendor Android application which has B/W and grascale options (though may be software implemented).

The second is a Samsung M337x 387x 407x Series (WSD mode), exact model is ProXpress M3870FD. That works fine, after firmware was updated though (V4.00.01.04 APR-09-2013 -> V4.00.02.20 MAY-27-2020).
Before the upgrade ADF scan caused device reboot.
   
The third one available is OKI-MB472 (eSCL), model no. N22502B, but it doesn't work well. It is succesfully discovered, it reports its capabilities in eSCL mode, but fails to scan in both preview and final modes (no matter flatbed or ADF). Even with latest firmware (A01.79_0_4).
I guess, I'll file a separate issue on that with debug info.

alexpevzner, 2020-07-10:
Yes, please file a separate issue for OKI-MB472, I want to look to the logs.

I would like also to look to the Brother's log, in case missed Grayscale support is my mistake (which is unlikely, though).

alexpevzner, 2020-07-13:
Hi,

I glad to see my driver included into the official AltLinux distro :-)

You may consider upgrade to 0.9.9, which is basically includes changes that goes as a separate patch in AltLinux.

Also, you don't need to put meson into "requires" - it is required only for building, and only if you actually use it (the "official" and supported build method is a plain Makefile)

alexpevzner, 2020-07-13:
And yet another thing. You may want to consider inclusion of another program, written by me: https://github.com/OpenPrinting/ipp-usb

It gives access to IPP-over-USB devices. Many modern devices support this protocol.

In short, it is a modern replacement of the `ippusbxd` that really works

evgeni, 2020-07-15:
FWIW, I also have a Brother MFC-L2720DW and it does "only" scan in Color mode (with sane-airscan-0.99.9-67.1.x86_64 on Fedora 32 from the OBS repo linked in the readme).

What kind of logs would you need to debug this?

alexpevzner, 2020-07-15:
As usually, I need protocol trace. Uncomment the following lines in the `/etc/sane.d/airscan.conf`:

    [debug]
    trace = ~/airscan/trace
    enable = true

Logs will be in the `~/airscan/trace` directory, which will be created automatically. `.tar` file is not needed, console logs are also not needed (they are mirrored to the protocol trace)

evgeni, 2020-07-15:
[perl-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/4924691/perl-zeroconf.log)
[perl-Brother-MFC-L2720DW-series.log](https://github.com/alexpevzner/sane-airscan/files/4924692/perl-Brother-MFC-L2720DW-series.log)

there you go :)

alexpevzner, 2020-07-15:
This printed doesn't announce ability to scan in Grayscale:

    <wscn:PlatenColor>
        <wscn:ColorEntry>RGB24</wscn:ColorEntry>
    </wscn:PlatenColor>

And the same for ADF:

    <wscn:ADFColor>
        <wscn:ColorEntry>RGB24</wscn:ColorEntry>
    </wscn:ADFColor>

This is hard to say what can I do at this case.

evgeni, 2020-07-15:
I have the Brother iPrint&Scan app on my Android phone, which allows to select "Color", "Color (fast)" and "Grayscale", would a network dump of the communication help?

Also, this is WSD mode, is there a way to test/force eSCL? `airscan-discover` does not list eSCL though (just thought that it might be there given README lists the 2710 and 2750 having eSCL).

alexpevzner, 2020-07-15:
I believe this app performs conversion to Grayscale in software, but if network dump is available, I will look to it. However, most likely Brother app will use Brother's proprietary protocol instead of the WSD.

You may set protocol selection mode to `manual` in the `/etc/sane.d/airscan.conf`:

    [options]
    protocol = manual

However in the default `auto` mode driver prefers eSCL, if both protocols are available. So if it doesn't offer it, the device doesn't support eSCL. I've looked to the `perl-zeroconf.log`, there is no traces of the eSCL support.

evgeni, 2020-07-15:
Yeah, I tried `protocol = manual` and no eSCL showed up. Well, was worth a try.

I'll see that I dump the traffic from the app and if it's using WSD provide you with pcaps.

alexpevzner, 2020-07-15:
OK

stapelberg, 2020-07-20:
@evgeni do you have avahi running/working? In my tests it was required for eSCL, which was then preferred.

FWIW, I have a 27**50**DW (not a 27**20**DW) and can scan grayscale out of the box no problem. Maybe they fixed it with the newer model. Or maybe a firmware update helps?

alexpevzner, 2020-07-20:
@stapelberg,

Yes, @evgeni runs avahi, I can see it the supplied `perl-zeroconf.log`. The device advertises itself as IPP printer (`_ipp._tcp`), but doesn't advertise itself as eSCL scanner (`_uscans._tcp`)

2750DW was reported as supporting eSCL and WSD as well as Color and Grayscale modes. Actually, devices with similar model name and appearance may have very different stuffing under the hood.

stapelberg, 2020-07-21:
Gotcha. Side note: what’s the difference between _uscan._tcp and _uscan**s**._tcp?

alexpevzner, 2020-07-21:
_uscan == http, _uscans == https

alexpevzner, 2020-07-23:
Hi @evgeni,

I've implemented Grayscale emulation for devices like yours. If you can build `sane-airscan`, you may want to test it now.

evgeni, 2020-07-23:
Ooh, cool, will definitely give it a try!

evgeni, 2020-07-27:
The conversion works like a charm!