alexpevzner/sane-airscan #53: build system issues

andyrtr, 2020-07-12:
What's the preferred way to build sane-airscan - the simple Makefile or the meson based build?

The Makefile doesn't create the symlink libsane-airscan.so -> libsane-airscan.so.1.
The Meson based build doesn't allow to pass SYSCONFDIR and puts config them under /usr/etc, it doesn't split dll.d/airscan to the dll.d directory.

alexpevzner, 2020-07-12:
The official way is `Makefile`. Meson build was contributed by my users, and I simply keep it here, trying not to break it ocassionally.

I can add symlink creation, but I have a strong impression that it actually is not needed for any purpose.

I will look how to remove libjpeg and libpng from airscan-discover dependencies

alexpevzner, 2020-07-12:
`Makefile` updated, so `airscan-discover` is not linked against jpeg/png libraries anymore

leigh123linux, 2020-07-31:
> The official way is `Makefile`. Meson build was contributed by my users, and I simply keep it here, trying not to break it ocassionally.

This Pr should address the meson build https://github.com/alexpevzner/sane-airscan/pull/58

alexpevzner, 2020-07-31:
Closed - fixed

closed 2020-07-31 by alexpevzner