alexpevzner/sane-airscan #54: The Canon Image Class MF244dw: only partial success

debiantriage, 2020-07-18:
Hello Alex,

Please see https://forums.linuxmint.com/viewtopic.php?f=51&t=325094

> ty=Canon MF240 Series

> scanimage -L
device `airscan:w1:Canon Canon MF240 Series' is a WSD Canon Canon MF240 Series WSD network scanner

The user gets xsane running but there is a device I/O error. The same happens with simple-scan and scanimage. Have I recommended the correct package?

Cheers,

Brian.

alexpevzner, 2020-07-18:
Hi Brain,

yes, the package is correct.

Please, look to the manual, page 198: https://content.etilize.com/User-Manual/1035940046.pdf

Seems that WSD scan needs to be enabled in the printer's web console, and it is disabled by default.

alexpevzner, 2020-07-18:
P.S. I'm very busy today. If it simple advise wouldn't help, I can pay more attention to this issue later

debiantriage, 2020-07-21:
Success! Thank you for focusing my attention on this enabling feature in some Canon devices. It's good to be able to say "goodbye" to the scangearmp frontend on modern devices.

Cheers,

Brian.

alexpevzner, 2020-07-21:
It would be nice if device owner could create a short instruction how to enable WSD on his device, similar to footnote [5] of my README. It could help other users of similar devices.

colin-williams, 2020-08-03:
Hello,

Debian testing user here.Found this when I was having trouble with sane on my device. Unable to make my MF244dw scan, but here are some instructions on how to enable WSD scan.

 You can do it from the browser or from the device LCD display. Here are instructions for both.

From device LCD screen

Menu
System Managment Settings
Network Settings
TCP/IP Settings
WSD Settings
WSD Scan Settings
Use WSD Scan
On

Web browser method

Visit IP address of printer device for web console.
Choose System Managment Mode > Login
Choose Settings > Registration
Choose System Managment Settings > Network Settings
Choose TCP/IP settings
Choose Edit WSD settings
Choose Use WSD scanning

However after enabling this setting I am unable to scan with xsane. @debiantriage @alexpevzner can you suggest next steps?

alexpevzner, 2020-08-03:
Hi @colin-williams,

have you enabled `Use Multicast Discovery` as well?

colin-williams, 2020-08-04:
@alexpevzner 

After messing with it for a couple hours, I'm able to scan a single page from the auto document feeder using simple-scan. The devices continues to grab and scan pages but after displaying the first page simple-scan doesn't append the other pages. Xsane doesn't scan any documents at all.

When you say multicast discovery, I see that mDNS is enabled.

I can share a video of the behavior if you like.

alexpevzner, 2020-08-04:
Please, enable and send me protocol logs. In the `/etc/sane.d/airscan.conf` file uncomment the following lines:

    [debug]
    trace = ~/airscan/trace
    enable = true

Logs will be in the `~/airscan/trace` directory, which will be created automatically.

And please tell me, what `sane-airscan` version do you use

colin-williams, 2020-08-04:
Hi again @alexpevzner . I had a look an it looks like I had installed the ubuntu version vs debian package. I installed the debian package. When I started xsane again after installing the sane-airscan_0.99.11+69.2_amd64.deb package, I saw the scanner now also list WSD as an option. Also listed is a non WSD option. I chose the WSD option. I can scan and it looks like I can use ADF but need to tell it number of pages.

I'm not sure if simple-scan is using xsane or sane-airscan. After above I launched simple-scan. Again chose ADF, scan. Same behavior as mentioned previously. It displays one page, continues scanning and then says cancelled. 

Anyhow working much better. Will try to see if simple-scan writes the logs also.

colin-williams, 2020-08-04:
@alexpevzner after rebooting issues with simple-scan also resolved. Then it seems to be functioning well. Terribly sorry to waste your time. It's too bad sane-airscan wasn't installed along with sane from the start...

debiantriage, 2020-08-04:
On Mon 03 Aug 2020 at 20:54:15 -0700, colin-williams wrote:

> @alexpevzner after rebooting issues with simple-scan also resolved.
> Then it seems to be functioning well. Terribly sorry to waste your
> time. It's too bad sane-airscan wasn't installed along with sane from
> the start...

Hello @colin-williams,

The next best thing is for sane-airscan to be packaged for Debian and
this has been done. It is in the NEW queue and hopefully will soon enter
unstable. The wiki will be updated to cover its use.

closed 2020-08-07 by alexpevzner