alexpevzner/sane-airscan #55: Canon PIXMA MG3600 Series

debiantriage, 2020-07-20:
https://forums.linuxmint.com/viewtopic.php?f=51&t=325428

> device `pixma:04A9178A' is a CANON Canon PIXMA MG3600 Series multi-function peripheral
device `escl:http://127.0.0.1:60000' is a ESCL MG3600 series flatbed scanner
device `airscan:e0:MG3600 series' is a eSCL MG3600 series eSCL network scanner

Scanning did not originally work until sane-airscan was installed.

alexpevzner, 2020-07-20:
Looks like it works with `sane-escl` and was never tested with `sane-airscan`'. Why it didn't work initially is not very clean

debiantriage, 2020-07-20:
On Mon 20 Jul 2020 at 05:56:10 -0700, Alexander Pevzner wrote:

> Looks like it works with `sane-escl` and was never tested with
> `sane-airscan`'. Why it didn't work initially is not very clean

You win some; you lose some. :)

Thanks,

Brian.

alexpevzner, 2020-07-20:
We should spread the rumor, that installing `sane-airscan` magically cures even unrelated problems :-)

debiantriage, 2020-07-21:
I said: 

> You win some; you lose some.

Looks like we won this one. The user now responds:

> eSCL MG3600 series ' this is the one that works for me many thanks again

Cheers,

Brian.

closed 2020-07-21 by debiantriage