alexpevzner/sane-airscan #56: Brother MFC-L2750DW: BW mode throws an IO error in both simple-scan and xsane

garrett, 2020-07-21:
Firstly: Huge things for this project to add support to so many more scanners! I'm happy mine works (albeit in color mode only).

My MFC-L2750DW scans perfectly fine in color, in both simple-scan and xsane. However, when I select text mode (BW) in simple-scan or gray in xsane, the scanner sounds like it works, but throws an IO error.

- simple-scan shows a dialog stating: "Failed to scan: Error communicating with the scanner"
- xsane shows a dialog stating: "Error during read: Error during device I/O."

This is on Fedora 32 with:
```
sane-backends-libs-1.0.30-1.fc32.x86_64
xsane-0.999-35.fc32.x86_64
xsane-common-0.999-35.fc32.x86_64
sane-airscan-0.99.10-68.1.x86_64
```

garrett, 2020-07-21:
I enabled debugging and manual selection in the /etc/sane.d/airscan.conf file. I still only see eSCL (even though WSD should also show up for this model, from what I understand).

Here's the log:

```
Brother MFC-L2750DW series: device created
Brother MFC-L2750DW series: DEVICE_STM_OPENED->DEVICE_STM_PROBING
Brother MFC-L2750DW series: using protocol "eSCL"
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScannerCapabilities
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScannerCapabilities: 200 OK
Brother MFC-L2750DW series: new decoder: jpeg
Brother MFC-L2750DW series: DEVICE_STM_PROBING->DEVICE_STM_IDLE
Brother MFC-L2750DW series: sane_open("e0:Brother MFC-L2750DW series"): Success
Brother MFC-L2750DW series: sane_start()
Brother MFC-L2750DW series: DEVICE_STM_IDLE->DEVICE_STM_SCANNING
Brother MFC-L2750DW series: PROTO_OP_SCAN: submitting: attempt=0
Brother MFC-L2750DW series: HTTP POST http://192.168.1.7/eSCL/ScanJobs
Brother MFC-L2750DW series: HTTP POST http://192.168.1.7/eSCL/ScanJobs: 201 Created
Brother MFC-L2750DW series: PROTO_OP_SCAN: decoding
Brother MFC-L2750DW series: PROTO_OP_SCAN: decoded: status="Success" next=PROTO_OP_LOAD delay=0
Brother MFC-L2750DW series: PROTO_OP_LOAD: submitting: attempt=0
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f/NextDocument
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f/NextDocument: 200 OK
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoding
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoded: status="Success" next=PROTO_OP_LOAD delay=0
Brother MFC-L2750DW series: PROTO_OP_LOAD: submitting: attempt=0
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f/NextDocument
Brother MFC-L2750DW series: Unexpected image format
Brother MFC-L2750DW series: sane_read(): Error during device I/O
Brother MFC-L2750DW series: sane_close()
Brother MFC-L2750DW series: cancel requested
Brother MFC-L2750DW series: JOB status=Operation was canceled
Brother MFC-L2750DW series: cancel skipped as job is almost done
Brother MFC-L2750DW series: DEVICE_STM_CANCEL_REQ->DEVICE_STM_CANCEL_REQ_DONE
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f/NextDocument: 404 Not Found
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoding
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoded: status="Success" next=PROTO_OP_CLEANUP delay=0
Brother MFC-L2750DW series: DEVICE_STM_CANCEL_REQ_DONE->DEVICE_STM_CLEANUP
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: submitting: attempt=0
Brother MFC-L2750DW series: HTTP DELETE http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f
Brother MFC-L2750DW series: HTTP DELETE http://192.168.1.7/eSCL/ScanJobs/74994f81-cb30-11ea-873a-3c2af46a6f0f: 404 Not Found
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: decoding
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: decoded: status="Success" next=PROTO_OP_FINISH delay=0
Brother MFC-L2750DW series: DEVICE_STM_CLEANUP->DEVICE_STM_DONE
Brother MFC-L2750DW series: DEVICE_STM_DONE->DEVICE_STM_CLOSED
Brother MFC-L2750DW series: removed from device table
Brother MFC-L2750DW series: closed protocol "eSCL"
Brother MFC-L2750DW series: closed decoder: jpeg
Brother MFC-L2750DW series: device destroyed
```

garrett, 2020-07-21:
Here's a successful scan in color mode:

```
Brother MFC-L2750DW series: device created
Brother MFC-L2750DW series: DEVICE_STM_OPENED->DEVICE_STM_PROBING
Brother MFC-L2750DW series: using protocol "eSCL"
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScannerCapabilities
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScannerCapabilities: 200 OK
Brother MFC-L2750DW series: new decoder: jpeg
Brother MFC-L2750DW series: DEVICE_STM_PROBING->DEVICE_STM_IDLE
Brother MFC-L2750DW series: sane_open("e0:Brother MFC-L2750DW series"): Success
Brother MFC-L2750DW series: sane_start()
Brother MFC-L2750DW series: DEVICE_STM_IDLE->DEVICE_STM_SCANNING
Brother MFC-L2750DW series: PROTO_OP_SCAN: submitting: attempt=0
Brother MFC-L2750DW series: HTTP POST http://192.168.1.7/eSCL/ScanJobs
Brother MFC-L2750DW series: HTTP POST http://192.168.1.7/eSCL/ScanJobs: 201 Created
Brother MFC-L2750DW series: PROTO_OP_SCAN: decoding
Brother MFC-L2750DW series: PROTO_OP_SCAN: decoded: status="Success" next=PROTO_OP_LOAD delay=0
Brother MFC-L2750DW series: PROTO_OP_LOAD: submitting: attempt=0
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f/NextDocument
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f/NextDocument: 200 OK
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoding
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoded: status="Success" next=PROTO_OP_LOAD delay=0
Brother MFC-L2750DW series: PROTO_OP_LOAD: submitting: attempt=0
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f/NextDocument
Brother MFC-L2750DW series: HTTP GET http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f/NextDocument: 404 Not Found
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoding
Brother MFC-L2750DW series: PROTO_OP_LOAD: decoded: status="Success" next=PROTO_OP_CLEANUP delay=0
Brother MFC-L2750DW series: DEVICE_STM_SCANNING->DEVICE_STM_CLEANUP
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: submitting: attempt=0
Brother MFC-L2750DW series: HTTP DELETE http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f
Brother MFC-L2750DW series: HTTP DELETE http://192.168.1.7/eSCL/ScanJobs/b6835bf6-cb30-11ea-873b-3c2af46a6f0f: 404 Not Found
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: decoding
Brother MFC-L2750DW series: PROTO_OP_CLEANUP: decoded: status="Success" next=PROTO_OP_FINISH delay=0
Brother MFC-L2750DW series: DEVICE_STM_CLEANUP->DEVICE_STM_DONE
Brother MFC-L2750DW series: DEVICE_STM_DONE->DEVICE_STM_IDLE
Brother MFC-L2750DW series: sane_read(): End of file reached
```

Both of these logs are while scanning with simple-scan.

alexpevzner, 2020-07-21:
Hi!

to deal with this case, I need a full protocol trace. To enable it, uncomment the following lines in the `/etc/sane.d/airscan.conf` file:

    [debug]
    trace = ~/airscan/trace
    enable = true

Traces will be in the `~/airscan/trace` directory, which will be created automatically. I need both .log and .tag files (don't put anything confidential into the scanner; .tar file will contain an image)

Everything that goes to console mirrored to the protocol trace, so you don't need to attach console logs when protocol traces are supplied

garrett, 2020-07-21:
I've tried scanning again. (It's a quickstart guide for my keyboard; it's mainly black and white but has splashes of color.)
[airscan.zip](https://github.com/alexpevzner/sane-airscan/files/4953985/airscan.zip)

I see there's a JPEG of a black and white scan in the tar file, so I guess it does successfully scan and then hits the error. (Which is enough for the SANE or the scanning apps to think it didn't work at all.)

alexpevzner, 2020-07-21:
This is actually more interesting case: image looks monochrome, but actually encoded as 24-bit RGB.

Probably I'll need to implement 24->8 bit resampling. It will also enable Grayscale mode simulation for Brother MFC-L2720DWR

alexpevzner, 2020-07-22:
Hi @garrett ,

I've impelemted RGB24->Grayscale8 image resampling.

if building from sources is not a problem for you, you can test it now.

garrett, 2020-07-22:
I created a toolbox container to test this out (since I'm on Fedora 32 Silverblue), did a make install, and tested it with simple-scan (also in the toolbox).

It works really well! Thanks!

alexpevzner, 2020-07-29:
Fixed in 0.99.11

closed 2020-07-29 by alexpevzner