alexpevzner/sane-airscan #60: [WSD] HP LaserJet MFP M130fw fails with 'PNG: unexpected EOF'

zdohnal, 2020-08-10:
Hi Alex,

an user in Fedora [here](https://bugzilla.redhat.com/show_bug.cgi?id=1867126) has a problem with scanning via WSD with HP LaserJet MFP M130fw. The following debug log [scanimage-HP-LaserJet-MFP-M130fw-(E700D6).log](https://github.com/alexpevzner/sane-airscan/files/5050820/scanimage-HP-LaserJet-MFP-M130fw-.E700D6.log) shows an error during PNG decoding - expected size doesn't match with one of acquired PNG file.

I'm not sure why - but InputMediaSize in request and MediaFrontImageInfo in response seems fishy. InputMediaSize seems like an initial size, but the result image comes with size from MediaFrontImageInfo.

```
00:00:01.239: HP LaserJet MFP M130fw (E700D6): Starting scan, using the following parameters:
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   source:         Flatbed
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   colormode_emul: Color
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   colormode_real: Color
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   tl_x:           0 mm
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   tl_y:           0 mm
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   br_x:           209.98 mm
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   br_y:           296.93 mm
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   image size:     8267x11690
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   image X offset: 0
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   image Y offset: 0
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   x_resolution:   300
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   y_resolution:   300
00:00:01.239: HP LaserJet MFP M130fw (E700D6):   format:         png
```

Request:
```
...
    <a:Action>http://schemas.microsoft.com/windows/2006/08/wdp/scan/CreateScanJob</a:Action>
  </s:Header>
  <s:Body>
...
          <scan:InputSize>
            <scan:InputMediaSize>
              <scan:Width>8267</scan:Width>
              <scan:Height>11690</scan:Height>
            </scan:InputMediaSize>
          </scan:InputSize>
...
```

Response:
```
    <wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.microsoft.com/windows/2006/08/wdp/scan/CreateScanJobResponse</wsa:Action>
  </SOAP-ENV:Header>
  <SOAP-ENV:Body>
...
      <WSDScannerService:ImageInformation>
        <WSDScannerService:MediaFrontImageInfo>          <WSDScannerService:PixelsPerLine>2480</WSDScannerService:PixelsPerLine>          <WSDScannerService:NumberOfLines>3507</WSDScannerService:NumberOfLines>          <WSDScannerService:BytesPerLine>7440</WSDScannerService:BytesPerLine>
        </WSDScannerService:MediaFrontImageInfo>
      </WSDScannerService:ImageInformation>
...
        <WSDScannerService:InputSize>
          <WSDScannerService:InputMediaSize>
            <WSDScannerService:Width>8267</WSDScannerService:Width>
            <WSDScannerService:Height>11690</WSDScannerService:Height>
          </WSDScannerService:InputMediaSize>
        </WSDScannerService:InputSize>
...
```

Output:
```
00:00:47.202: HP LaserJet MFP M130fw (E700D6): Image received with the following parameters:
00:00:47.202: HP LaserJet MFP M130fw (E700D6):   content type:   image/png
00:00:47.202: HP LaserJet MFP M130fw (E700D6):   frame format:   RGB
00:00:47.202: HP LaserJet MFP M130fw (E700D6):   image size:     2480x3507
00:00:47.202: HP LaserJet MFP M130fw (E700D6):   color depth:    8
```

Log during finding the device is here 
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5050884/scanimage-zeroconf.log) .

Would you mind looking into it?

Thank you in advance,

Zdenek

alexpevzner, 2020-08-10:
Hi Zdenek,

image sizes in InputMediaSize and in MediaFrontImageInfo are in different units: first is in 1/1000 of inch and second is in real pixels, so this is OK.

Something got wrong with decoding of the received PNG. There are two `scanimage-HP-LaserJet-MFP-M130fw-(E700D6).*` files: `.log` and `.tar`. I need both to deal with issues like this.

alexpevzner, 2020-08-13:
Hi Zdenek,

there seems to be enough fixes to issue the next 0.99.14 release, but this bug blocks me.

Don't know what to do: to wait a little bit more, or release as is.

zdohnal, 2020-08-13:
@alexpevzner Tom has a vacation this week, he will answer after weekend at best, so no need to wait for this bug.

zdohnal, 2020-08-17:
Here is the png file 
[image]

zdohnal, 2020-08-17:
I tried to run it with ```identify -verbose``` (ends with error code 1, no message). If it is run without ```-verbose```, it gets following output:

```
$ identify 00000000.png 
00000000.png PNG 2480x3507 2480x3507+0+0 8-bit sRGB 1781760B 0.000u 0:00.000
```

Then I ran ```pngcheck```:

```
75>>>pngcheck -vctfp7 00000000.png
File: 00000000.png (1781760 bytes)
  chunk IHDR at offset 0x0000c, length 13
    2480 x 3507 image, 24-bit RGB, non-interlaced
  chunk IDAT at offset 0x00025, length 7173
    zlib: deflated, 32K window, default compression
  chunk IDAT at offset 0x01c36, length 2484
  chunk IDAT at offset 0x025f6, length 6247
  chunk IDAT at offset 0x03e69, length 8192
  chunk IDAT at offset 0x05e75, length 8192
  chunk IDAT at offset 0x07e81, length 8192
  chunk IDAT at offset 0x09e8d, length 8192
  chunk IDAT at offset 0x0be99, length 5863
  chunk IDAT at offset 0x0d58c, length 8192
  chunk IDAT at offset 0x0f598, length 8192
  chunk IDAT at offset 0x115a4, length 8192
  chunk IDAT at offset 0x135b0, length 8192
  chunk IDAT at offset 0x155bc, length 8192
  chunk IDAT at offset 0x175c8, length 8192
  chunk IDAT at offset 0x195d4, length 8192
  chunk IDAT at offset 0x1b5e0, length 8192
  chunk IDAT at offset 0x1d5ec, length 1792
  chunk IDAT at offset 0x1dcf8, length 8192
  chunk IDAT at offset 0x1fd04, length 8192
  chunk IDAT at offset 0x21d10, length 8192
  chunk IDAT at offset 0x23d1c, length 8192
  chunk IDAT at offset 0x25d28, length 8192
  chunk IDAT at offset 0x27d34, length 8192
  chunk IDAT at offset 0x29d40, length 1364
  chunk IDAT at offset 0x2a2a0, length 8192
  chunk IDAT at offset 0x2c2ac, length 8192
  chunk IDAT at offset 0x2e2b8, length 8192
  chunk IDAT at offset 0x302c4, length 8192
  chunk IDAT at offset 0x322d0, length 8192
  chunk IDAT at offset 0x342dc, length 3530
  chunk IDAT at offset 0x350b2, length 8192
  chunk IDAT at offset 0x370be, length 8192
  chunk IDAT at offset 0x390ca, length 8192
  chunk IDAT at offset 0x3b0d6, length 8192
  chunk IDAT at offset 0x3d0e2, length 8192
  chunk IDAT at offset 0x3f0ee, length 8192
  chunk IDAT at offset 0x410fa, length 7171
  chunk IDAT at offset 0x42d09, length 8192
  chunk IDAT at offset 0x44d15, length 8192
  chunk IDAT at offset 0x46d21, length 8192
  chunk IDAT at offset 0x48d2d, length 8192
  chunk IDAT at offset 0x4ad39, length 8192
  chunk IDAT at offset 0x4cd45, length 8192
  chunk IDAT at offset 0x4ed51, length 4510
  chunk IDAT at offset 0x4fefb, length 8192
  chunk IDAT at offset 0x51f07, length 8192
  chunk IDAT at offset 0x53f13, length 8192
  chunk IDAT at offset 0x55f1f, length 8192
  chunk IDAT at offset 0x57f2b, length 8192
  chunk IDAT at offset 0x59f37, length 530
  chunk IDAT at offset 0x5a155, length 8192
  chunk IDAT at offset 0x5c161, length 8192
  chunk IDAT at offset 0x5e16d, length 8192
  chunk IDAT at offset 0x60179, length 1439
  chunk IDAT at offset 0x60724, length 8192
[... 162 lines cut ...]
  chunk IDAT at offset 0x18edd7, length 8192
  chunk IDAT at offset 0x190de3, length 7212
  chunk IDAT at offset 0x192a1b, length 8192
  chunk IDAT at offset 0x194a27, length 8192
  chunk IDAT at offset 0x196a33, length 8192
  chunk IDAT at offset 0x198a3f, length 8192
  chunk IDAT at offset 0x19aa4b, length 8192
  chunk IDAT at offset 0x19ca57, length 8192
  chunk IDAT at offset 0x19ea63, length 2550
  chunk IDAT at offset 0x19f465, length 8192
  chunk IDAT at offset 0x1a1471, length 8192
  chunk IDAT at offset 0x1a347d, length 8192
  chunk IDAT at offset 0x1a5489, length 8192
  chunk IDAT at offset 0x1a7495, length 8192
  chunk IDAT at offset 0x1a94a1, length 8192
  chunk IDAT at offset 0x1ab4ad, length 5029
  chunk IDAT at offset 0x1ac85e, length 8192
  chunk IDAT at offset 0x1ae86a, length 8192
  chunk IDAT at offset 0x1b0876, length 8192
  chunk IDAT at offset 0x1b2882, length 8192:  EOF while reading data
ERRORS DETECTED in 00000000.png
```
It seems IEND chunk is missing, but I don't know why. The picture is cropped too.

When I ran it in ```ghex```, I can see only signature, and IHDR and IDAT chunks - in other png files I can see some xml stuff, but not in this one.

alexpevzner, 2020-08-17:
Actually, about last 1/3 of PNG image is truncated (985 lines of 3507).

There is some chance that bug is in my MIME multipart decoder, though I've reviewed its code and so far found nothing suspicious. I've asked Tom to retest with the latest version (I've added appropriate instrumentation there).

I think, we can continue discussion in the RedHat bugzilla -- it will reduce the round-trip time.

zdohnal, 2020-08-18:
Tom tested version 0.99.14 and it works for him. So it is fixed.

zdohnal, 2020-08-18:
Thanks for the fix!

closed 2020-08-18 by zdohnal

alexpevzner, 2020-08-18:
I'd be glad to know, how I managed to fix it...

Well, now I have #65 issue that definitely points to multipart decoder, and these two issues can be related. For that, I have a whole set of data to reproduce, so I believe multipart decoder will be really fixed soon

zdohnal, 2020-08-18:
I tried to bisect the commits between versions 0.99.13 and 0.99.14 and sent to Tom for testing, but now he is not able to reproduce even with 0.99.12.

But he updated the system - so probably something under airscan fixed the issue.

alexpevzner, 2020-08-18:
Looks like firmware bug. May be, scanner was out of memory at that moment, which caused image truncation

I investigated and hopefully fixed #65, it has nothing in common with image truncation

If this problem will ever repeat with 0.99.14 (or, better, with the next after .14 version) I will be able to tell more