alexpevzner/sane-airscan #64: An HP M478f-9f

debiantriage, 2020-08-16:
https://ubuntuforums.org/showthread.php?t=2448857

> ty=HP Color LaserJet Pro M478f-9f

An interesting device from a printing point of view. It has msword doc and docx as accepted PDLs.

> paavo@Paavo:~$ scanimage -L
device `airscan:e0:HP Color LaserJet Pro M478f-9f [B1E168]' is a eSCL HP Color LaserJet Pro M478f-9f [B1E168] eSCL network scanner
paavo@Paavo:~$ airscan-discover
[devices]
HP Color LaserJet Pro M478f-9f [B1E168] = http://192.168.1.6:8080/eSCL/, eSCL
HP Color LaserJet Pro M478f-9f [B1E168] = https://192.168.1.6:443/eSCL/, eSCL
HP Color LaserJet Pro M478f-9f [B1E168] = http://192.168.1.6:53048, WSD
HP Color LaserJet Pro M478f-9f [B1E168] = http://[Fe80::ead8:d1FF:Feb1:e169%253]:53048, WSD
paavo@Paavo:~$ 

So - scanimage -L does not give the same as what airscan-discover gives. WSD is shown with airscan-discover.

closed 2020-08-16 by debiantriage

alexpevzner, 2020-08-16:
Really interesting device.

`airscan-discover` simply prints all that it has discovered, while `sane-airscan` by default takes a responsibility to choose the "best" variant. I don't think regular user wants to see its device listed many times (eSCL/WSD, IP4/IP6, Ethernet/WiFi, HTTP/HTTPS and so on). USB is the special exception: if used together with the `ipp-usb`, USB connection will be listed separately, because some users may prefer USB for its privacy or Network for its speed.

BTW, `ippusbxd` is now officially deprecated in the favor of `ipp-usb`: https://openprinting.github.io/OpenPrinting-News-August-2020/

So if user has none of them, it is better to recommend to install `ipp-usb`, it is much more reliable

debiantriage, 2020-08-17:
On Sun 16 Aug 2020 at 13:08:12 -0700, Alexander Pevzner wrote:

> Really interesting device.
>
> `airscan-discover` simply prints all that it has discovered, while
> `sane-airscan` by default takes a responsibility to choose the "best"
> variant. I don't think regular [...]

I am a little lost now. Does the HP Color LaserJet Pro M478f-9f not have
a WSD mode shown by airscan discover?

> BTW, `ippusbxd` is now officially deprecated in the favor of
> `ipp-usb`:
> https://openprinting.github.io/OpenPrinting-News-August-2020/

Thanks for the link. As the user who submitted the RFP in Debian, I am
pleased about that.

> So if user has none of them, it is better to recommend to install
> `ipp-usb`, it is much more reliable

Agreed! Note that ipp-usb is recommended by cups-daemon in Debian. I am
also thinking that sane-utils or libsane1 should recommend it too.

Cheers,

Brian.

alexpevzner, 2020-08-17:
> I am a little lost now. Does the HP Color LaserJet Pro M478f-9f not have
a WSD mode shown by airscan discover?

sane-airscan understands that both protocols (eSCL and WSD) are offered by the same physical device, and, by default, automatically chooses eSCL and hides WSD. airscan-discover has a different behavior: it simply dumps all alternatives that it has discovered.

This default can be changed by editing configuration file, but this is a kind of advanced options.

debiantriage, 2020-08-17:
On Mon 17 Aug 2020 at 09:44:40 -0700, Alexander Pevzner wrote:

> a WSD mode shown by airscan discover?
>
> sane-airscan understands that both protocols (eSCL and WSD) are
> offered by the same physical device, and, by default, automatically
> chooses eSCL and hides WSD. [...]

I understand that. But airscan-discover surely has WSD as an alternative.
Yet it is not shown in the table at https://github.com/alexpevzner/sane-airscan.

Cheers,

Brian.

alexpevzner, 2020-08-17:
This is because we know that the device supports WSD, but didn't actually test if it works or not.

You may ask user to enable manual protocol selection and to test:

    [options]
    protocol = manual

I usually hesitate to ask. Mostly because for non-geek users it is a really hard job.

debiantriage, 2020-08-17:
On Mon 17 Aug 2020 at 10:05:28 -0700, Alexander Pevzner wrote:

> This is because we know that the device supports WSD, but didn't
> actually test if it works or not.

Got it! I've been a bit slow in my understanding today.

> You may ask user to enable manual protocol selection and to test:
> 
>     [options]
>     protocol = manual
> 
> I usually hesitate to ask. Mostly because for non-geek users it is a
> really hard job.

I would normally not ask because it is an imposition on the user to go
beyond solving the problem faced (and often solved).

Regards,

Brian.

alexpevzner, 2020-08-17:
> I would normally not ask because it is an imposition on the user to go
beyond solving the problem faced (and often solved).

Yes, this is what I mean. Some users, however, are glad to play a little bit, and this is useful, because increases testing coverage.