alexpevzner/sane-airscan #65: sane_read(): Error during device I/O

fhangel, 2020-08-18:
Hi Alex

First of all, this is a great idea and I can't believe it is not included with most distros so thank you for making it available.

Unfortunatelly, I'm getting an error with my Ubunut 20.04 installation using a Dell C2665dnf Color MFP. The scanner is easily discovered and I can even start the scanning process from xsane (and Ubunutu's Document scanner as well) but once the scanner finishes its job I get the error message, "**Error during device I/O**" as you can see in the image below. Based on the logs, it seem like the data is received from the scanner but it can't be interpreted correctly "**RetrieveImageRequest: invalid response**"

[image]

Trace file: https://drive.google.com/file/d/1HGrJmgagkcZMWhBYyjeQoGteDlCgcLC6/view?usp=sharing
Log File: [xsane-Dell-Dell-C2665dnf-Color-MFP.log](https://github.com/alexpevzner/sane-airscan/files/5088402/xsane-Dell-Dell-C2665dnf-Color-MFP.log)

Any help with this would be very much appreciated.

alexpevzner, 2020-08-18:
Hi Freddy,

thank you for reporting this issue!

Just in case, if I will need to ask you to test something, are you able to build sane-airscan from sources, or prefer that I will prepare binary packages for testing (which is not difficult for me, but sometimes works slow)?

And may I look to the `xsane-zeroconf.log` file too?

alexpevzner, 2020-08-18:
> First of all, this is a great idea and I can't believe it is not included with most distros so thank you for making it available.

Forgot to tell, it already included into the Arch Linux and will come with the new releases of Debian, Ubuntu and Fedora. For the older distros I will continue to support my own packaging until they are complely obsolete

fhangel, 2020-08-18:
Hi Alex, thanks for the quick reply. I'm happy to build from source if that helps.

Here is the zeroconf log: [xsane-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5090686/xsane-zeroconf.log)

alexpevzner, 2020-08-18:
Hi Freddy,

I still decided to rebuild by myself.

Please, download and install the `sane-airscan-unstable` (instead of sane-airscan_0.99.xx) package from the usual location: https://download.opensuse.org/repositories/home:/pzz/

I hope I've fixed this issue. Please retest and confirm.

fhangel, 2020-08-19:
Hi Alex, 

I'm happy to confirm that it is working now. 

Thank you for your amazing work.

closed 2020-08-19 by fhangel

alexpevzner, 2020-08-19:
You may now upgrade to 0.99.15, if you wish.