alexpevzner/sane-airscan #70: HP DeskJet 5000 series. Brother DCP-L2550DW.

debiantriage, 2020-08-29:
Hello Alex,

It's not often the opportunity to do a Brother crops up.

https://forums.linuxmint.com/viewtopic.php?f=51&t=328780

Fairly straightforward:

>  simple-scan "airscan:e0:HP DeskJet 5000 series [D9B468] (USB)"
    Can you scan? 
       Yes I am able to scan.

https://forums.linuxmint.com/viewtopic.php?f=51&t=328894

Not so straightforward, but the user was very enthusiastic.

> When I tried scanning, there were two variants in the drop down menu both with the printer/scanner name. One was prefaced with ESCL and the other with eSCL. Scanning worked with both variants.

Regards,

Brian.

closed 2020-08-29 by debiantriage

alexpevzner, 2020-08-31:
Hi Brian,

this is very interesting, how the story with printing part of Brother DCP-L2550DW has finished.

With some devices, ipp-usb stucks in retrieving HTTP response, and it requires a special workaround. 3 parts of device (printing, scanning and web interface) may behave very differently in this aspect (most often, printing and scanning works, while web interface hangs, but there are variants).

In the nearest future (probably, in the next release), these "quirks" will be configurable, so it will be possible to play with them on a user machine without need to rebuild ipp-usb. But for now, it requires patching the source.

I'd be glad to look to this problem, if user can make experiments and send logs. To begin, I need him to reproduce the printing problem and send me fresh logs from `/var/log/ipp-usb/*`

debiantriage, 2020-09-01:
> this is very interesting, how the story with printing part of Brother DCP-L2550DW has finished.

There have been a few more posts since you wrote that. Alex.  I am unlikely to pursue this issue any further.

Cheers,

Brian.