alexpevzner/sane-airscan #71: An Xerox WorkCentre 3025

tex, 2020-08-30:
Hello Alex, great project!

Is there anything more I can do to help you to support this scanner?

```
milan@nixos> airscan-discover
[devices]
  Xerox WorkCentre 3025 (XRX9C934E3E92AF) = http://192.168.1.171:8018/wsd/scan, WSD
milan@nixos> scanimage -d "airscan:w0:Xerox WorkCentre 3025 (XRX9C934E3E92AF)" -o test.jpg
scanimage: sane_read: Error during device I/O
```

Wireshark capture:
[sane-airscan-workcentre3025.zip](https://github.com/alexpevzner/sane-airscan/files/5146834/sane-airscan-workcentre3025.zip)

alexpevzner, 2020-08-30:
Hi Milan,

I hope I've fixed support of your device. May I ask you to rebuild and retest?

tex, 2020-08-31:
Wonderful!!! It works! Gray and color, different DPIs, perfect! You can add it to your list of supported scanners :-)

closed 2020-08-31 by tex

tex, 2020-08-31:
May I ask you to create a new tag / version so I can update NixOS package (will create merge request so everyone will be able to enjoy this good progress ;-) )?

alexpevzner, 2020-08-31:
Yes, I will in few minutes.

I wonder, how does this device scan in Windows. It sends a malformed MIME multipart "document", so I had to tweak my multipart parser to accept documents with similar bugs. AFAIK, no widely used open source multipart parsers will accept it.

alexpevzner, 2020-08-31:
Done with tag. @zdohnal, @OdyX, please, be aware about the new `0.99.16` release, that makes possible WSD scanning with Xerox WorkCentre 3025

zdohnal, 2020-09-01:
@alexpevzner thanks for the heads-up, I'll build the new version for Fedora today.