alexpevzner/sane-airscan #77: Crash with Canon Pixma TS 5055

zdohnal, 2020-10-01:
Hi,

I have an issue in [Fedora](https://bugzilla.redhat.com/show_bug.cgi?id=1882520) with Canon Pixma TS 5055, which is connected via wifi.

I have the [backtrace](https://bugzilla.redhat.com/attachment.cgi?id=1717662) for now, showing ```http_uri_str()``` returns NULL, but I don't know yet how is it possible to have a NULL pointer there.

I asked for .tar and .log files too, there will be there as soon as the reporter provides them.

The simple fix would be to check whether ```http_uri_str()``` returns NULL, but the more curious I am about how NULL got there. I tried to backtrace in the code to see where ```uri->str``` is initialized, but got stuck in ```OUTER_STRUCT```, which I failed to understand.

Would you mind looking into it?

Thank you in advance!

Zdenek

alexpevzner, 2020-10-01:
Hi,

`uri->str` cannot be `NULL`, unless there is some kind of memory corruption error here. May be, running under valgrind will help to catch it.

Checking `http_uri_str()` for `NULL` will not fix the error, it will only hide it, and it will crash somewhere else (may be even more difficult to catch).

I will try to analyse your backtrace using /dev/eyes and /dev/brain :-), but it is obviously not simple, so if you know how to reproduce or have some more info, it will help.

`OUTER_STRUCT` converts pointer to structure field to the pointer of structure that owns this field by subtracting field's byte offset within a structure from a field pointer:
* If P is pointer to struct, &P->field points to field. If P is 0, &P->field is field byte offset within a struct
*  ((ptrdiff_t) &(((struct_t*) 0)->field)) - field byte offset within a struct, casted to integer type `ptrdiff_t`. It can be replaced with `offsetof`, but for some historical reasons I don't like `offsetof`
* ((char*)(member_p) - ((ptrdiff_t) &(((struct_t*) 0)->field))) - pointer to the struct, but casted to `(char*)`
* ((struct_t*)((char*)(member_p) - ((ptrdiff_t) &(((struct_t*) 0)->field)))) - pointer to the struct, properly caseted

zdohnal, 2020-10-01:
Ad backtrace - it is mostly optimized, I asked the reported to rerun with specific binary and install needed debuginfos. If it does not help, I'll recompile airscan with -O0.
Basically you can only get lines from it :( right now.

alexpevzner, 2020-10-01:
I've probably got it.

This is very unlikely that `uri->str` is NULL, because before reaching a point of failure, it was already accessed to obtain/resolve target IP address.

Most likely, `end = strchr(host, '/');` returns `NULL`. It can only happen, if host path of URL is not IP address literal, and path part is empty, even without `/` character (i.e., http://example.com)

I can tell for sure, if will look to user's `sane-airscan` logs. I've noticed you have requested it in Fedora bigzilla.

zdohnal, 2020-10-02:
There are quite a lot of memory errors according valgrind [results](https://bugzilla.redhat.com/attachment.cgi?id=1718313). I don't think I'll manage to propose fixes for all of them today, I hope I manage to get into it next week.

Here are the airscan [logs](https://bugzilla.redhat.com/attachment.cgi?id=1718312) . The reporter noticed the issue is triggered when he is on Wifi, because his router provides a shared network printer of some kind.

alexpevzner, 2020-10-02:
I think I've finally fixed it!

zdohnal, 2020-10-07:
@alexpevzner  the reporter tested the fix and it works. Thank you again!

zdohnal, 2020-10-07:
Closed via https://github.com/alexpevzner/sane-airscan/commit/850654147750750283cd85c2adc133dbc7773f88

closed 2020-10-07 by zdohnal

alexpevzner, 2020-10-07:
Thank you for keeping me informed!

I plan to issue the next release in 1-2 days.