alexpevzner/sane-airscan #78: Scanning from https with port number fails on 0.99.16, but not on 0.99.10

fletcherw, 2020-10-01:
I have a scanner that supports eSCL (Lexmark MB2236adwe).

If I run `scanimage -d "airscan:escl:Lexmark MB2236adwe:https://192.168.0.15:443/eSCL/" > out.pnm`
on sane-airscan v0.99.10, I can successfully scan. [0.99.10-succeeded.log](https://github.com/alexpevzner/sane-airscan/files/5312956/0.99.10-succeeded.log)

If I run the same command on v0.99.16, the scan fails. [0.99.16-failed.log](https://github.com/alexpevzner/sane-airscan/files/5313012/0.99.16-failed.log)

If I omit the port number on v0.99.16, the scan succeeds (`scanimage -d "airscan:escl:Lexmark MB2236adwe:https://192.168.0.15/eSCL/" > out.pnm`) [0.99.16-succeeded.log](https://github.com/alexpevzner/sane-airscan/files/5312954/0.99.16-succeeded.log)

fletcherw, 2020-10-01:
@yetamrra reports a similar issue on an HP OfficeJet 9010

alexpevzner, 2020-10-01:
Could you please powercycle you device, retry with failing 0.99.16 and send me the log?

I have an impression that the actual problem was in a previous run, which I don't see.

alexpevzner, 2020-10-01:
I hope I've fixed it. You may fetch latest source, rebuild and retry

fletcherw, 2020-10-01:
Hi Alex, thank you for the prompt response. Using a version built from the top of tree, this works properly.

closed 2020-10-01 by fletcherw

alexpevzner, 2020-10-01:
No problem, guys, you are always welcome!

I'm currently revisiting my device discovery core to improve its behaviour in some corner cases. Hope to finish in few days. After that I will create 0.9.17 release. Your patches that enables AF_UNIX support will also be included there.