alexpevzner/sane-airscan #80: Unable to scan using HP LaserJet Pro MFP M126nw

technastic, 2020-10-02:
[brian_p](https://launchpad.net/~claremont102) (username in Launchpad) suggested me to raise an issue here.  Please check this thread (https://answers.launchpad.net/hplip/+question/693153) for more details.

alexpevzner, 2020-10-02:
Hi @technastic,

I've read launchpad discussion. Please, uncomment the following lines in the `/etc/sane.d/airscan.conf` file:

    [debug]
    trace = ~/airscan/trace
    enable = true

Then rerun your test. The `~/airscan/trace` directory will be created and some files will be written there. Please, send me `*.log` files from that directory (you can attach them to this github issue). `*.tar` files are not needed for now.

technastic, 2020-10-03:
@alexpevzner  I'm a new user of Ubuntu (and Linux in general). `/etc/sane.d/airscan.conf` file is a read-only file. How can I modify it? It would be great if you let me know the steps in detail. And what do you mean by uncomment? Are you saying me to remove `#`? 

> Then rerun your test. 

Which "test" are you talking about?

alexpevzner, 2020-10-03:
I mean, retry failing scan attempt

technastic, 2020-10-03:
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5321554/scanimage-zeroconf.log)

alexpevzner, 2020-10-03:
Please, powercycle your printer, and then show me output of the following two commands:

    avahi-browse -a -rt
    airscan-discover

I know that Brian has already asked you to do so, but as something has changed, I want to ask you to retry these steps

technastic, 2020-10-03:
I hope you mean powering off and powering on the printer, by the word 'powercycle'.

After turning off and turning on the printer, the output of commands are:

[image]

debiantriage, 2020-10-03:
I can produce the same output, but only by disconnecting the ethernet cable attached to the computer. Please take a good look at yours, Ramesh.

alexpevzner, 2020-10-04:
Two more commands please:

    ifconfig
    ping 192.168.0.107

technastic, 2020-10-04:
Output of `ifconfig`:

```
enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.105  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::4dba:a9d5:6e44:715  prefixlen 64  scopeid 0x20<link>
        ether 40:8d:5c:f6:a1:8c  txqueuelen 1000  (Ethernet)
        RX packets 3183  bytes 3655400 (3.6 MB)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 3061  bytes 323068 (323.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 441  bytes 42426 (42.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 441  bytes 42426 (42.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

Output of `ping 192.168.0.107`:

```
PING 192.168.0.107 (192.168.0.107) 56(84) bytes of data.
64 bytes from 192.168.0.107: icmp_seq=1 ttl=255 time=117 ms
64 bytes from 192.168.0.107: icmp_seq=2 ttl=255 time=1.73 ms
64 bytes from 192.168.0.107: icmp_seq=3 ttl=255 time=1.71 ms
64 bytes from 192.168.0.107: icmp_seq=4 ttl=255 time=1.83 ms
64 bytes from 192.168.0.107: icmp_seq=5 ttl=255 time=1.66 ms
64 bytes from 192.168.0.107: icmp_seq=6 ttl=255 time=1.74 ms
64 bytes from 192.168.0.107: icmp_seq=7 ttl=255 time=1.51 ms
64 bytes from 192.168.0.107: icmp_seq=8 ttl=255 time=1.67 ms
64 bytes from 192.168.0.107: icmp_seq=9 ttl=255 time=1.47 ms
64 bytes from 192.168.0.107: icmp_seq=10 ttl=255 time=1.99 ms
64 bytes from 192.168.0.107: icmp_seq=11 ttl=255 time=2.22 ms
64 bytes from 192.168.0.107: icmp_seq=12 ttl=255 time=2.00 ms
64 bytes from 192.168.0.107: icmp_seq=13 ttl=255 time=2.19 ms
64 bytes from 192.168.0.107: icmp_seq=14 ttl=255 time=1.56 ms
64 bytes from 192.168.0.107: icmp_seq=15 ttl=255 time=1.73 ms
64 bytes from 192.168.0.107: icmp_seq=16 ttl=255 time=2.08 ms
64 bytes from 192.168.0.107: icmp_seq=17 ttl=255 time=2.05 ms
64 bytes from 192.168.0.107: icmp_seq=18 ttl=255 time=1.90 ms
64 bytes from 192.168.0.107: icmp_seq=19 ttl=255 time=2.30 ms
64 bytes from 192.168.0.107: icmp_seq=20 ttl=255 time=1.42 ms
64 bytes from 192.168.0.107: icmp_seq=21 ttl=255 time=2.13 ms
64 bytes from 192.168.0.107: icmp_seq=22 ttl=255 time=2.46 ms
64 bytes from 192.168.0.107: icmp_seq=23 ttl=255 time=2.71 ms
64 bytes from 192.168.0.107: icmp_seq=24 ttl=255 time=1.70 ms
64 bytes from 192.168.0.107: icmp_seq=25 ttl=255 time=2.01 ms
64 bytes from 192.168.0.107: icmp_seq=26 ttl=255 time=1.55 ms
64 bytes from 192.168.0.107: icmp_seq=27 ttl=255 time=2.06 ms
64 bytes from 192.168.0.107: icmp_seq=28 ttl=255 time=6.41 ms
64 bytes from 192.168.0.107: icmp_seq=29 ttl=255 time=1.61 ms
64 bytes from 192.168.0.107: icmp_seq=30 ttl=255 time=1.54 ms
64 bytes from 192.168.0.107: icmp_seq=31 ttl=255 time=1.75 ms
64 bytes from 192.168.0.107: icmp_seq=32 ttl=255 time=1.61 ms
64 bytes from 192.168.0.107: icmp_seq=33 ttl=255 time=1.71 ms
64 bytes from 192.168.0.107: icmp_seq=34 ttl=255 time=1.63 ms
64 bytes from 192.168.0.107: icmp_seq=35 ttl=255 time=1.89 ms
64 bytes from 192.168.0.107: icmp_seq=36 ttl=255 time=1.72 ms
64 bytes from 192.168.0.107: icmp_seq=37 ttl=255 time=3.33 ms
64 bytes from 192.168.0.107: icmp_seq=38 ttl=255 time=1.88 ms
64 bytes from 192.168.0.107: icmp_seq=39 ttl=255 time=1.25 ms
64 bytes from 192.168.0.107: icmp_seq=40 ttl=255 time=1.45 ms
64 bytes from 192.168.0.107: icmp_seq=41 ttl=255 time=1.77 ms
64 bytes from 192.168.0.107: icmp_seq=42 ttl=255 time=1.45 ms
64 bytes from 192.168.0.107: icmp_seq=43 ttl=255 time=3.32 ms
[... 124 lines cut ...]
64 bytes from 192.168.0.107: icmp_seq=168 ttl=255 time=1.65 ms
64 bytes from 192.168.0.107: icmp_seq=169 ttl=255 time=1.77 ms
64 bytes from 192.168.0.107: icmp_seq=170 ttl=255 time=1.43 ms
64 bytes from 192.168.0.107: icmp_seq=171 ttl=255 time=2.10 ms
64 bytes from 192.168.0.107: icmp_seq=172 ttl=255 time=1.67 ms
64 bytes from 192.168.0.107: icmp_seq=173 ttl=255 time=1.68 ms
64 bytes from 192.168.0.107: icmp_seq=174 ttl=255 time=1.41 ms
64 bytes from 192.168.0.107: icmp_seq=175 ttl=255 time=1.82 ms
64 bytes from 192.168.0.107: icmp_seq=176 ttl=255 time=1.77 ms
64 bytes from 192.168.0.107: icmp_seq=177 ttl=255 time=1.37 ms
64 bytes from 192.168.0.107: icmp_seq=178 ttl=255 time=6.21 ms
64 bytes from 192.168.0.107: icmp_seq=179 ttl=255 time=1.47 ms
64 bytes from 192.168.0.107: icmp_seq=180 ttl=255 time=1.51 ms
64 bytes from 192.168.0.107: icmp_seq=181 ttl=255 time=1.68 ms
64 bytes from 192.168.0.107: icmp_seq=182 ttl=255 time=1.67 ms
64 bytes from 192.168.0.107: icmp_seq=183 ttl=255 time=1.52 ms
64 bytes from 192.168.0.107: icmp_seq=184 ttl=255 time=1.63 ms
64 bytes from 192.168.0.107: icmp_seq=185 ttl=255 time=1.31 ms
64 bytes from 192.168.0.107: icmp_seq=186 ttl=255 time=1.65 ms
64 bytes from 192.168.0.107: icmp_seq=187 ttl=255 time=4.21 ms
^C
--- 192.168.0.107 ping statistics ---
187 packets transmitted, 187 received, 0% packet loss, time 186288ms
rtt min/avg/max/mdev = 1.179/2.666/116.645/8.414 ms
```

debiantriage, 2020-10-04:
The immediate question on the table is - why did `avahi-browse -a -rt` succeed and then fail? Let's try this:

Install tcpdump. Switch off the printer. Execute

 `sudo tcpdump --interface enp1s0 -vv host 192.168.0.107 > tcpdump.log`

Now switch the printer on and wait until it finishes booting. CTRL-C to stop the command. Post tcpdump.log.

Cheers,

Brian.

alexpevzner, 2020-10-04:
It looks like multicast traffic is disabled in this network. Or like both Bonjour (DNS-SD) and WSD discovery is disabled on printer, and there is no other network devices.

@technastic, can you explain your network in detail? Is it WiFi or wired Ethernet, how many devices are connected to the network, what router do you use etc?

Did you recently change your network or your printer configuration?

technastic, 2020-10-05:
@debiantriage Output of `sudo tcpdump --interface enp1s0 -vv host 192.168.0.107 > tcpdump.log`:

```
tcpdump: listening on enp1s0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C7 packets captured
7 packets received by filter
0 packets dropped by kernel
```

[tcpdump.log](https://github.com/alexpevzner/sane-airscan/files/5325795/tcpdump.log)

technastic, 2020-10-05:
@alexpevzner I'm having a TP Link router (Model: TL-WR841N(EU), Version: 14.1). My PC is connected to the router through Ethernet cable. Couple of smartphones and the printer are connected to the Wi-Fi network (wirelessly).

I didn't change my network or printer configuration recently.

alexpevzner, 2020-10-05:
Is your TP-Link router configured as Router or as Access Point?

technastic, 2020-10-05:
@alexpevzner The TP-Link router is configured as Router.

alexpevzner, 2020-10-05:
And PC is connected to one of LAN ports (yellow), correct?

technastic, 2020-10-05:
@alexpevzner Yup, correct.

alexpevzner, 2020-10-05:
Do you use "Guest Network" feature of your router?

technastic, 2020-10-05:
I don't use Guest Network feature.

alexpevzner, 2020-10-05:
Any non-default security settings?

technastic, 2020-10-05:
[image]

alexpevzner, 2020-10-05:
Show me the Basic Security window too, please

technastic, 2020-10-05:
The screenshot is about the Basic Security. Check the screenshot properly.

alexpevzner, 2020-10-05:
OK, lets try the static configuration. Please, add the following lines to the `/etc/sane.d/airscan.conf` file:

    [devices]
    HP LaserJet Pro MFP M126nw = http://192.168.0.107:3911/scanner, WSD

Then try to scan with `simple-scan`, `xsane` or something similar

technastic, 2020-10-05:
I've added the aforementioned lines to the `/etc/sane.d/airscan.conf` file. You can download the file from here: https://www.dropbox.com/s/0rbd4dx44sz9x5a/airscan.conf?dl=0

I tried scanning scanning using `simple-scan`. It didn't work. Refer this video: https://www.dropbox.com/s/bv0xt6cmswtldgn/12345.mp4?dl=0

Output of `simple-scan`:

```
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
zeroconf:   HP LaserJet Pro MFP M126nw = http://192.168.0.107:3911/scanner, WSD
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.0.105
WSDD:  + fe80::4dba:a9d5:6e44:715%2
WSDD: 192.168.0.105: started discovery, UDP port=52567
WSDD: fe80::4dba:a9d5:6e44:715: started discovery, UDP port=33065
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: zeroconf_device_list_get: requested
zeroconf: zeroconf_initscan_wait: requested
zeroconf: zeroconf_initscan_wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   WSD   "HP LaserJet Pro MFP M126nw"
WSDD: fe80::4dba:a9d5:6e44:715: done discovery
WSDD: 192.168.0.105: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: initial scan timer expired
zeroconf: zeroconf_initscan_wait: requested
zeroconf: zeroconf_initscan_wait: OK
HP LaserJet Pro MFP M126nw: device created
HP LaserJet Pro MFP M126nw: DEVICE_STM_OPENED->DEVICE_STM_PROBING
HP LaserJet Pro MFP M126nw: using protocol "WSD"
HP LaserJet Pro MFP M126nw: HTTP POST http://192.168.0.107:3911/scanner
HP LaserJet Pro MFP M126nw: HTTP using timeout: 5000 ms
HP LaserJet Pro MFP M126nw: HTTP resolving 192.168.0.107 3911
HP LaserJet Pro MFP M126nw: HTTP trying 192.168.0.107:3911
HP LaserJet Pro MFP M126nw: HTTP 192.168.0.107:3911: send(): Connection refused
HP LaserJet Pro MFP M126nw: HTTP POST http://192.168.0.107:3911/scanner: Connection refused
HP LaserJet Pro MFP M126nw: scanner capabilities query: Connection refused
HP LaserJet Pro MFP M126nw: DEVICE_STM_PROBING->DEVICE_STM_PROBING_FAILED
HP LaserJet Pro MFP M126nw: removed from device table
HP LaserJet Pro MFP M126nw: closed protocol "WSD"
HP LaserJet Pro MFP M126nw: device destroyed
sane_open("w0:HP LaserJet Pro MFP M126nw"): Error during device I/O
zeroconf: zeroconf_initscan_wait: requested
zeroconf: zeroconf_initscan_wait: OK
HP LaserJet Pro MFP M126nw: device created
HP LaserJet Pro MFP M126nw: DEVICE_STM_OPENED->DEVICE_STM_PROBING
HP LaserJet Pro MFP M126nw: using protocol "WSD"
HP LaserJet Pro MFP M126nw: HTTP POST http://192.168.0.107:3911/scanner
HP LaserJet Pro MFP M126nw: HTTP using timeout: 5000 ms
HP LaserJet Pro MFP M126nw: HTTP resolving 192.168.0.107 3911
HP LaserJet Pro MFP M126nw: HTTP trying 192.168.0.107:3911
HP LaserJet Pro MFP M126nw: HTTP 192.168.0.107:3911: send(): Connection refused
HP LaserJet Pro MFP M126nw: HTTP POST http://192.168.0.107:3911/scanner: Connection refused
HP LaserJet Pro MFP M126nw: scanner capabilities query: Connection refused
HP LaserJet Pro MFP M126nw: DEVICE_STM_PROBING->DEVICE_STM_PROBING_FAILED
HP LaserJet Pro MFP M126nw: removed from device table
[... 83 lines cut ...]
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: zeroconf_device_list_get: requested
zeroconf: zeroconf_initscan_wait: requested
zeroconf: zeroconf_initscan_wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   WSD   "HP LaserJet Pro MFP M126nw"

(xsane:6663): Gdk-CRITICAL **: 20:35:53.547: IA__gdk_drawable_unref: assertion 'GDK_IS_DRAWABLE (drawable)' failed

(xsane:6663): Gdk-CRITICAL **: 20:35:53.547: IA__gdk_drawable_unref: assertion 'GDK_IS_DRAWABLE (drawable)' failed
WSDD: 192.168.0.105: done discovery
WSDD: fe80::4dba:a9d5:6e44:715: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: initial scan timer expired
sane_exit() called
sane_exit(): OK
```

alexpevzner, 2020-10-05:
Is your printer connected to USB? If yes, please try to disconnect

technastic, 2020-10-05:
@alexpevzner The printer isn't connected to USB.

technastic, 2020-10-05:
@alexpevzner I don't have Type B to Ethernet cable :(

alexpevzner, 2020-10-05:
According to manual, your printer has a normal Ethernet port.

technastic, 2020-10-05:
@alexpevzner  Oh.. yeah! I forgot. Sorry. I'll go check if I have a spare Ethernet cable. Should I connect the Ethernet cable from my printer to router or PC?

alexpevzner, 2020-10-05:
To router

technastic, 2020-10-05:
@alexpevzner Connected! Now what should I do?

alexpevzner, 2020-10-05:
airscan-discover
scanimage -L

technastic, 2020-10-06:
Output of `airscan-discover`:

```
[devices]
  HP HP LaserJet Pro MFP M126nw = http://192.168.0.106:3911/scanner, WSD
  HP HP LaserJet Pro MFP M126nw = http://[fe80::5a20:b1ff:fe4f:70b4%252]:3911/scanner, WSD
```

Output of `scanimage -L`:

```
sane_init() called
loading configuration file /etc/sane.d/airscan.conf
zeroconf:   HP LaserJet Pro MFP M126nw = http://192.168.0.107:3911/scanner, WSD
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.0.105
WSDD:  + fe80::4dba:a9d5:6e44:715%2
WSDD: 192.168.0.105: started discovery, UDP port=43671
WSDD: fe80::4dba:a9d5:6e44:715: started discovery, UDP port=41950
MDNS: browse/ipv6 "HP LaserJet Pro MFP M126nw[4F70B4]": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv4 "HP LaserJet Pro MFP M126nw[4F70B4]": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6 "HP LaserJet Pro MFP M126nw[4F70B4]": AVAHI_BROWSER_NEW _scanner._tcp
MDNS: browse/ipv4 "HP LaserJet Pro MFP M126nw[4F70B4]": AVAHI_BROWSER_NEW _scanner._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
MDNS: resolve/ipv6 "HP LaserJet Pro MFP M126nw[4F70B4]": AVAHI_RESOLVER_FOUND _ipp._tcp
zeroconf: found urn:uuid:8afb247f-3032-bb47-01b6-d66e7632c3a6
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 2 (enp1s0)
zeroconf:   name:      HP LaserJet Pro MFP M126nw[4F70B4]
zeroconf:   model:     HP LaserJet Pro MFP M126nw
zeroconf:   device:    0001 (created)
WSDD: directed probe: trying if=enp1s0, addr=192.168.0.106
WSDD: HTTP POST http://192.168.0.106/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.0.106 80
WSDD: HTTP trying 192.168.0.106:80
zeroconf: zeroconf_device_list_get: requested
zeroconf: zeroconf_initscan_wait: requested
WSDD: HTTP POST http://192.168.0.106/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: got response headers (200)
WSDD: HTTP POST http://192.168.0.106/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: 200 OK
WSDD: HTTP POST http://192.168.0.106/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: OK
WSDD: HTTP POST http://192.168.0.106:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6
WSDD: HTTP resolving 192.168.0.106 3911
WSDD: HTTP trying 192.168.0.106:3911
WSDD: HTTP POST http://192.168.0.106:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: got response headers (200)
WSDD: HTTP POST http://192.168.0.106:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: 200 OK
WSDD: HTTP POST http://192.168.0.106:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: OK
zeroconf: found urn:uuid:8afb247f-3032-bb47-01b6-d66e7632c3a6
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp1s0)
zeroconf:   name:      -
zeroconf:   model:     HP HP LaserJet Pro MFP M126nw
zeroconf:   device:    0001 (found by uuid+ifindex)
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:     http://192.168.0.106:3911/scanner
WSDD: HTTP POST http://[FE80::5A20:B1FF:FE4F:70B4%252]:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6
WSDD: HTTP resolving FE80::5A20:B1FF:FE4F:70B4%2 3911
WSDD: HTTP trying [fe80::5a20:b1ff:fe4f:70b4%2]:3911
WSDD: HTTP POST http://[FE80::5A20:B1FF:FE4F:70B4%252]:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: got response headers (200)
WSDD: HTTP POST http://[FE80::5A20:B1FF:FE4F:70B4%252]:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: 200 OK
WSDD: HTTP POST http://[FE80::5A20:B1FF:FE4F:70B4%252]:3911/8afb247f-3032-bb47-01b6-d66e7632c3a6: OK
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: zeroconf_initscan_wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: HP LaserJet Pro MFP M126nw[4F70B4]: can: WSD, use: WSD
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   WSD   "HP LaserJet Pro MFP M126nw"
zeroconf:   WSD   "HP LaserJet Pro MFP M126nw[4F70B4]"
device `hpaio:/net/HP_LaserJet_Pro_MFP_M126nw?ip=192.168.0.107' is a Hewlett-Packard HP_LaserJet_Pro_MFP_M126nw all-in-one
device `hpaio:/net/hp_laserjet_pro_mfp_m126nw?ip=192.168.0.106&queue=false' is a Hewlett-Packard hp_laserjet_pro_mfp_m126nw all-in-one
device `airscan:w0:HP LaserJet Pro MFP M126nw' is a WSD HP LaserJet Pro MFP M126nw WSD network scanner
device `airscan:w1:HP LaserJet Pro MFP M126nw[4F70B4]' is a WSD HP LaserJet Pro MFP M126nw[4F70B4] WSD network scanner
sane_exit() called
zeroconf: device gone urn:uuid:8afb247f-3032-bb47-01b6-d66e7632c3a6
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp1s0)
zeroconf: device gone urn:uuid:8afb247f-3032-bb47-01b6-d66e7632c3a6
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 2 (enp1s0)
zeroconf: ZEROCONF_WSD: initial scan finished
sane_exit(): OK
```

alexpevzner, 2020-10-06:
So does it scan now?

Please, try with simple-scan or xsane, choosing "HP LaserJet Pro MFP M126nw[4F70B4]" from the list

technastic, 2020-10-06:
@alexpevzner In `simple-scan`, there are 4 options now. 
1. HP HP_LaserJet_Pro_MFP_M126nw
2. HP hp_laserjet_pro_mfp_m126nw 
3. WSD HP HP LaserJet Pro MFP M126nw
4. WSD HP LaserJet Pro MFP M126nw

Using the third option (WSD HP HP LaserJet Pro MFP M126nw), I was able to successfully scan a document and save it :)

The other 3 options, didn't work.

alexpevzner, 2020-10-06:
(1) and (2) are from hplip. I have no idea, why device discovered twice and under different names. 
(4) is `sane-airscan` device, statically configured via `/etc/sane.d/airscan.conf`. (3) is device, dynamically discovered by `sane-airscan`, using the Ethernet cable, and it works

So the problem seems to be in the TP-Link router configuration

technastic, 2020-10-06:
Any idea on what is the exact configuration which I can modify to scan using the printer wirelessly?

alexpevzner, 2020-10-06:
What has changed, so printer was working initially (at least, discovery), and stopped to work later?

May be, at that time you have connected router to the Internet?

technastic, 2020-10-06:
I didn't get you @alexpevzner

alexpevzner, 2020-10-06:
When you posted your first messages to the answers.launchpad.net, you've shown the output of `airscan-discover`, and device was there. Later, `airscan-discover` was not able to figure out the device, and `avahi-browse` too.

I'm trying to understand, what has changed in between

technastic, 2020-10-06:
Even I don't know what happened..

alexpevzner, 2020-10-06:
Can smartphone wirelessly find the printer and print to it?

technastic, 2020-10-06:
Yes, I can.

alexpevzner, 2020-10-06:
I want to look to the following TP-Link configuration pages:
* Status
* Wireless->Wireless Advanced
* Guest Network
* Security->Advanced Security

technastic, 2020-10-06:
@alexpevzner I've shared it to your email through Dropbox.

alexpevzner, 2020-10-06:
I've got it, now thinking...

alexpevzner, 2020-10-06:
Also, Network->IPTV page, please

technastic, 2020-10-06:
Added it to Dropbox. Please check. @alexpevzner

alexpevzner, 2020-10-06:
Try to disable IGMP snooping at that page

To test, it's enough to try disconnect Ethernet cable and try to find a printer with `airscan-discover`. If printer is found, you may try to scan.

technastic, 2020-10-07:
I've disabled IGMP snooping and disconnected the ethernet cable from the printer. Output of `airscan-discover`:

```
[devices]
```

One more thing to note, the IP address of the printer has changed.

technastic, 2020-10-09:
Hey @alexpevzner. I got a `sane-airscan` update. I updated the software. Some stuff which I got in terminal while updating the app:

```
Setting up sane-airscan (0.99.17+76.1) ...

Configuration file '/etc/sane.d/airscan.conf'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** airscan.conf (Y/I/N/O/D/Z) [default=N] ? d
--- /etc/sane.d/airscan.conf    2020-10-05 19:54:29.227817641 +0530
+++ /etc/sane.d/airscan.conf.dpkg-new   2020-10-08 16:34:15.000000000 +0530
@@ -12,7 +12,7 @@
 #   https://github.com/alexpevzner/airscan-discover
 
 [devices]
-HP LaserJet Pro MFP M126nw = http://192.168.0.107:3911/scanner, WSD
+#"Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL
 #"Some Unwanted Scanner" = disable
 
 # Various options
@@ -35,12 +35,18 @@
 # for this purpose:
 #   model = network     ; use network device name (default)
 #   model = hardware    ; use hardware model name
+#
+# socket_dir gives an optional path to a directory where local (UNIX) sockets
+# can be found.  If an eSCL device's URL is in the form unix://socket/eSCL/,
+# traffic will be sent through socket_dir/socket instead of TCP.  If not
+# specified, sockets will be searched for in /var/run.
 
 [options]
 #discovery = enable
 #model = network
 #protocol = auto
 #ws-discovery = fast
+#socket_dir = /var/run
 
 # Configuration of debug facilities
 #   trace = path        ; enables protocol trace and configures output
@@ -50,5 +56,5 @@
 #
 #   enable = true|false ; enable or disable console logging
 [debug]
-trace = ~/airscan/trace
-enable = true
+#trace = ~/airscan/trace
+#enable = true

Configuration file '/etc/sane.d/airscan.conf'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** airscan.conf (Y/I/N/O/D/Z) [default=N] ? y
Installing new version of config file /etc/sane.d/airscan.conf ...
```

BTW, did you find out how to fix the scanner issue?

alexpevzner, 2020-10-09:
Hi @technastic,

this issue is more general that scanner issue. Looks like your router implements some kind of firewalling between wired and wireless segments of your local network, but so far I was not able to find in the router manual how to control this firewalling.

I will spend some more time trying to understand what's going on, but currently I have no new ideas.

technastic, 2020-10-11:
I removed all my printers from the Settings app. Then downloaded hplip-3.20.3-plugin.run from https://developers.hp.com/hp-linux-imaging-and-printing/plugins
I runned the script and it was successful.

Then I typed hp-setup in terminal in my printer and added my printer.

Now printing and scanning both works :)

closed 2020-10-28 by alexpevzner