alexpevzner/sane-airscan #82: If ipv6 is disabled, the program fails with an error.

motorBMW, 2020-10-09:
If ipv6 is disabled, the program exits with an error. I don't use ipv6 in my subnet. Is it possible to disable device detection by ipv6 addresses? And another question is not on the topic. Can I add an IP address to the name of the detected device? This is quite convenient when there is a large fleet of identical devices in the same subnet.

I would really appreciate Your help!

[image]

alexpevzner, 2020-10-09:
This is probably fixed in the newest (0.99.17) release.

motorBMW, 2020-10-12:
The problem is fixed, but there is a new one. If you exclude any device from the search, the program ends with an error. I want to exclude some devices from the search. It would also be nice to add the ability to exclude devices by manufacturer, such as Xerox or HP. "Xerox*" = disable

[airscan.conf.txt](https://github.com/alexpevzner/sane-airscan/files/5365468/airscan.conf.txt)
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5365453/scanimage-zeroconf.log)

alexpevzner, 2020-10-12:
> If you exclude any device from the search, the program ends with an error. 

It seems to be a duplicate of #83. Already fixed in git (though fix is not yet confirmed) and will be released soon,

> "Xerox VersaLink B405" = eSCL

This is wrong option. On a right side must be either valid URL, optionally followed by protocol, or "disable" keyword.

> It would also be nice to add the ability to exclude devices by manufacturer, such as Xerox or HP. "Xerox*" = disable

I will think.

Isn't it enough to exclude by network name (which is usually manufacturer + model, unless explicitly configured on device), but with wildcard?

motorBMW, 2020-10-12:
> This is wrong option. On a right side must be either valid URL, optionally followed by protocol, or "disable" keyword.
Surprisingly, if I don't use this option, sane returns a message that the scanner (device) is not available. If it still managed to connect, then the scan is not stable. When this option is set, the scanner works perfectly.
> Isn't it enough to exclude by network name (which is usually manufacturer + model, unless explicitly configured on device), but with wildcard?
We have devices in our fleet that don't support WSD. For example, we will need to use hpaio for HP, but we cannot add device exceptions to hpaio that should not be displayed. Therefore, they will be duplicated when reviewing the subnet. It is also quite difficult to add all devices to the configuration file from a large zoo.

Example of a small subnet:
[scanimage.txt](https://github.com/alexpevzner/sane-airscan/files/5366172/scanimage.txt)

motorBMW, 2020-10-12:
> This is wrong option. On a right side must be either valid URL, optionally followed by protocol, or "disable" keyword.<
Surprisingly, if I don't use this option, sane returns a message that the scanner (device) is not available. If it still managed to connect, then the scan is not stable. When this option is set, the scanner works perfectly.
>Isn't it enough to exclude by network name (which is usually manufacturer + model, unless explicitly configured on device), but with wildcard?<
We have devices in our fleet that don't support WSD. For example, we will need to use hpaio for HP, but we cannot add device exceptions to hpaio that should not be displayed. Therefore, they will be duplicated when reviewing the subnet. It is also quite difficult to add all devices to the configuration file from a large zoo.

Example of a small subnet:
[scanimage.log.txt](https://github.com/alexpevzner/sane-airscan/files/5366307/scanimage.log.txt)

motorBMW, 2020-10-12:
> This is wrong option. On a right side must be either valid URL, optionally followed by protocol, or "disable" keyword.

Surprisingly, if I don't use this option, sane returns a message that the scanner (device) is not available. If it still managed to connect, then the scan is not stable. When this option is set, the scanner works perfectly.

alexpevzner, 2020-10-12:
Could you retest please with `sane-airscan-unstable` instead of 0,99.17? You can download binary packages from the usual place (https://download.opensuse.org/repositories/home:/pzz/)

'sane-airscan-unstable' includes all fixes I currently have, and if people will confirm that it works OK, I'll release it as 0.99.18, may be tomorrow.

> Surprisingly, if I don't use this option, sane returns a message that the scanner (device) is not available

If it still exist after update to unstable, I want to investigate it. Can you send me protocol traces at this case?

> Example of a small subnet:

Really small subnet :-)

I will think.

motorBMW, 2020-10-14:
> 'sane-airscan-unstable' includes all fixes I currently have, and if people will confirm that it works OK, I'll release it as 0.99.18, may be tomorrow.

The problem is fixed. Thank you very much!

closed 2020-10-14 by motorBMW

alexpevzner, 2020-10-14:
You now can replace sane-airscan-unstable with 0.99.18 version, it includes all fixes for all problems known so far

Thank you for cooperation!

alexpevzner, 2020-10-14:
P.S. If you want me not forget about filtering by manufacturer, please, create a separate issue for that