alexpevzner/sane-airscan #86: Bug: sane-airscan should support abbreviated modes

2e0byo, 2020-10-15:
I think I've found a bug inn sane-airscan (but I've not checked and it could be elsewhere in sane).

## Behaviour
When scanning with scanimage to a sane-airscan backend, `--mode Gray` works but `--mode gray` doesn't

## Expected Behaviour
Per the manual for scanimage:
>   For convenience, it is legal to abbreviate the string values as long as they remain unique.  Also, the case of the spelling doesn't  matter.   For  example,  option  setting  --mode col is identical to --mode Color.

These abbreviations work with other backends, but not with sane-airscan in my setup.

## Environment
archlinux, with sane-airscan 0.99.16-2 and sane 1.0.31-1

Apologies if I should file this against the archlinux package but I couldn't see any issues reporting it being fixed already here.

## Logs
[scanimage-Canon-MG3600-series.log](https://github.com/alexpevzner/sane-airscan/files/5387422/scanimage-Canon-MG3600-series.log)
[scanimage-Canon-MG3600-series.tar.zip](https://github.com/alexpevzner/sane-airscan/files/5387426/scanimage-Canon-MG3600-series.tar.zip)

alexpevzner, 2020-10-15:
> These abbreviations work with other backends, but not with sane-airscan in my setup.

Could you please tell me a name of backend there abbreviations work with, I'll try to investigate this issue

alexpevzner, 2020-10-15:
BTW, is there any way to report issue against the archlinux package? I want to tell Arcj maintainers to update list of dependencies (for example, libsoup is not needed anymore), but have no idea how to do it.

2e0byo, 2020-10-16:
> BTW, is there any way to report issue against the archlinux package?

I've never actually done it, but [the wiki page on it](https://wiki.archlinux.org/index.php/Bug_reporting_guidelines#Opening_a_bug) looks fairly straightforward.  

> Could you please tell me a name of backend

Oops, good point.  It works with both escl and pixma (the other two backends which support my canon MG3600, but sane-airscan is faster and more reliable :) ).  I had a brief dig around in the source but I've not got a clue how the sane architecture works, sorry...

alexpevzner, 2020-10-16:
> It works with both escl and pixma

I've looked to the escl and pixma code. Both of them use `strcasecmp` instead of `strcmp` to compare option name against known values. OK, I also can do so (though many other backends perform case-sensitive search).

None of them do anything to support abbreviations. sane-escl compares colour mode against `Gray` and `Lineart` and if none match, considers mode to be `Color`. It may look as abbreviation support in some cases, but actually it's just a simple bug.

So I think I will make `--mode` and `--source` case-insensitive, while complain about abbreviation support, promised but missed, should be reported against `simplescan` (probably here: https://gitlab.com/sane-project/backends/-/issues)

2e0byo, 2020-10-16:
Okay, I feel a bit daft for not actually checking the images produced.  I've [opened an issue](https://github.com/alexpevzner/sane-airscan/issues/86) against sane-backends.

I also feel a bit daft for not noticing `strcasecmp` this morning, but I hadn't had any coffee yet!

alexpevzner, 2020-10-17:
Fixed

closed 2020-10-17 by alexpevzner