alexpevzner/sane-airscan #89: Version 0.99.17 breaks WSD protocol on Canon TS6220

chenhongqiao, 2020-10-20:
After updating sane-airscan from v0.99.16 to v0.99.17, WSD protocol seems to stop working. On Fedora 32 Workstation x86_64 with Canon TS6220, after starting the scanner, "airscan:w0:CANON INC. TS6200 series" appears shortly when executing "scanimage -L", then disappears after less than 30 seconds. I tried to scan in that 30 seconds time period and I got "unable to connect to scanner". Downgrading to v0.99.16 solved this. Thought it might be an upstream issue so report here.

chenhongqiao, 2020-10-20:
log attached
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5406347/scanimage-zeroconf.log)

chenhongqiao, 2020-10-20:
I got WSD back as an option by adding `protocol = manual` into the config file, so closing this issue.

closed 2020-10-20 by chenhongqiao

alexpevzner, 2020-10-20:
Hi Harry,

sane-airscan follows the principle, that one physical device should appear once in the list, even if supports both eSCL and WSD protocols. Versions 0.99.16 and before were not able to correctly merge eSCL and WSD instances of some Canon scanners, and it was fixed in 0.99.17. For most of other devices this merge worked even before this fix.

If you want to see both variants in the list, `protocol = manual` option, that you've discovered by yourself, exist especially for this purpose.

So it's not a bug, but desired behaviour. I believe, for most users the need to choose manually between eSCL and WSD only adds a bit of headache.

Anyway, thank you for your feedback and have a nice day! Hope, sane-airscan will work perfectly for you.