alexpevzner/sane-airscan #91: HP Deskjet 3762

debiantriage, 2020-10-26:
https://forums.linuxmint.com/viewtopic.php?f=51&t=332709&sid=1fe3caf380a611a5444f958567160350

> ty=HP DeskJet 3700 series

Alex, is my advice to power cycle and replug the device merely cargo cult computing? It does seem to produce results.

Cheers,

Brian.

closed 2020-10-26 by debiantriage

alexpevzner, 2020-10-26:
Brian,

Device may be out of sync with the host after previous unsuccessful attempts to scan with ippusbxd. Although ipp-usb performs some effort to put device into the known state, it was not investigated very deeply (I can't reproducibly put  my own device into the out-of-sync state), so in some cases ipp-usb may fail to start immediately after failed ippusbxd.

So power cycle and/or replugging the device may make some sense.

debiantriage, 2020-10-26:
Thank you Alex. I have experienced this "out of sync" myself and got very frustrated when I was initially testing ippusbxd and ipp-usb one after the other. I generally tell users that ippusbxd may leave the device in an inconsistent state (when I remember at the time, that is).