alexpevzner/sane-airscan #92: Oki-MC853: Error during device I/O

roro60, 2020-10-27:
When trying to scan with an Oki_MC853 on Debian testing, I always get the error "Error during device I/O" with sane. The eScl/ScanJobs always returns 400 Bad request. But if I try the same with curl, the scan succeeds.

I tried it again with self-built sane-airscan and sane-airscan-unstable with no change.

[xscanimage-OKI-MC853-FEA6D2.log](https://github.com/alexpevzner/sane-airscan/files/5446838/xscanimage-OKI-MC853-FEA6D2.log)

alexpevzner, 2020-10-27:
I don't see scan attempt it the log that you've submitted. Only opening device and setting some options, but no attempt to scan,

Could you please rerun the test and resubmit the log?

roro60, 2020-10-27:
I deleted the old files before retrying.

[xscanimage-OKI-MC853-FEA6D2.log](https://github.com/alexpevzner/sane-airscan/files/5446905/xscanimage-OKI-MC853-FEA6D2.log)

alexpevzner, 2020-10-27:
It seems that IP address of computer, allowed to scan, needs to be registered on a printer:

https://okiprinting-en-gb.custhelp.com/app/answers/detail/a_id/8222/~/i-want-to-scan-to-a-networked-mac.%28mc853%2C-mc873%2C-mc883%29

Please look to the `1-3. Register the destination computer for "Scan To Computer" to the device` section

roro60, 2020-10-28:
That is necessary for the scan-to-computer button on the Oki, using the TWAIN driver, not for the eScl interface. 

I tried something different: I commented out the "dirty hack" for HP LaserJet MFP M630 in escl_scan_query in airscan-escl.c and rebuilt sane-airscan. Then it works fine! :-) 

I don't know how to detect a device needs this hack or not. Maybe it would be better to use the real IP address of the computer, not localhost, or to define an configuration option.

alexpevzner, 2020-10-28:
> I tried something different: I commented out the "dirty hack" for HP LaserJet MFP M630 in escl_scan_query

It was my second idea. But seems you read my though :-)

> I don't know how to detect a device needs this hack or not

This is easy. I have `MakeAndModel` parameter in device capabilities. I will fix it today

> Maybe it would be better to use the real IP address of the computer

That was initial behaviour, and this is how HTTP client should actually work, But it doesn't work with that HP

Interesting that VueScan always sets `Host: localhost` for all eSCL requests. In theory, it should not work with your OKI

alexpevzner, 2020-10-28:
Please, test the fix

roro60, 2020-10-28:
Test is ok! :-) Thank you!

closed 2020-10-29 by alexpevzner