alexpevzner/sane-airscan #96: Potential race condition with http_data_ref and unref

fletcherw, 2020-11-09:
The ref counting logic for `http_data` (https://github.com/alexpevzner/sane-airscan/blob/master/airscan-http.c#L1702-L1731) is concerning. If an `http_data*` is being held by threads A and B, what prevents thread B from calling `http_data_unref()` and freeing the `http_data` before thread A is able to call `http_data_ref()` to increase the ref count?

alexpevzner, 2020-11-09:
There are actually no multithreading access to these structures. There is only one worker thread, and most of the code runs on its context. Few data structures are accessed from SANE API calls, but access from these calls is protected with a global mutex (see `eloop_mutex_lock/eloop_mutex_unlock` in nearly all API functions, in the `airscan.c` file)

The only piece of the airscan code that can be used from multiple threads is logger, logger has its own mutex.

fletcherw, 2020-11-09:
If it's never used in multithreaded code, then could the calls to `__sync_fetch_and_sub` and `__sync_fetch_and_add` just be replaced with `--` and `++`?

alexpevzner, 2020-11-10:
`__sync_fetch_and_sub` there mostly for convenience/out of habit.

alexpevzner, 2020-11-10:
BTW, this code is safe even in multithreaded environment.

If `http_data` is being held by threads A and B, reference count is at least 2. If thread B calls `http_data_unref`, reference count becomes 1, and `http_data` still not freed, so thread A can continue to safely work with it.

fletcherw, 2020-11-10:
The use of the atomics immediately makes me as a reader think that the code is meant to be safe in multi-threaded contexts. If that's not the intention, using them increases the difficulty of understanding the code since I am now concerned about data races.

I agree with you that as long as both threads hold strong references, the ref-counting is safe. But that means that you can never send a weak reference between threads. With something like http_query_get_request_data() saying that you must http_data_ref() in order to preserve the data after query end of life, what prevents me from calling that in a thread while the data is owned by another thread? That would introduce a chance for this race.

Even with the ref counting, it's also not safe to modify http_data when references are held by multiple threads, since the only memory barriers occur during the atomic operations, and it's not guaranteed that a barrier would occur after a write or before a read.

alexpevzner, 2020-11-10:
Yes, I agree, presence of atomic operations switches on "extra attention mode". `http_data` is very old data structure in this project, it was nearly from the very beginning, and it is hard to remember now, why I decided to implement thread-safe reference counting for it. Probably because I'm trying to be as generic as possible in any particular piece of code, especially if cost is not very high.

The only place when atomic operations are really needed now, is the synchronization between `sane_cancel()`, which can be called, according to SANE spec, from signal handler context and the worker thread, which really implements cancel logic. This is in the `airscan-device.c` file.

Regarding other concerns, well, it's C, not Rust. We can only document our agreements, but can't enforce them :-)

The life cycle of the `http_data` is simple and observable. On a transmit path it is allocated by the `http_query_new_len` function, released when `http_query` is being released and never exposed outside `airscan-http.c` except that for tracer, which writes its content to the trace file synchronously, on a context of `trace_http_query_hook`. On receive path is is allocated by response reader (`http_query_on_body_callback`, called from `http-parser`) transformed only within `airscan-http.c` (synchronously, by MIME multipart decoder), then passed to the eSCL/WSD protocol handler by `http_query` completion callback, where queued for image reader and released when image reading is completed.

 It begins its life in the `airscan-http.c`, when response is being

fletcherw, 2020-11-11:
Thanks for the explanation Alex. It helped me to understand the logic structure a bit better.

Since it sounds like there's no issue here, I'm going to close this.

closed 2020-11-11 by fletcherw