alexpevzner/sane-airscan #97: Add an ip address to the device name

motorBMW, 2020-11-12:
Hi, Alexander

I have a request to make. Could you kindly add the IP address to the scanner name? I am sure that this is true for many users who use a large number of devices of the same name. Otherwise, I have to determine the device by brute force, which is very inconvenient. It might even be better to add an option (device description) so that I can pick up the IP address when using your backend. I would really appreciate your help!

Example of a subnet with a large number of devices of the same name:
[image]

alexpevzner, 2020-11-12:
Hi,

I understand you problem, but good solution needs to be discussed.

You probably have DNS-SD (Avahi) disabled, aren't you? If DNS-SD is enabled, name comes from DNS-SD, even if chosen protocol is WSD, and all DNS-SD names are unique, at least within a single LAN. Only if DNS-SD name is not available, sane-airscan fallbacks to model name, which is not unique, as in your case.

Please note, in more complex setup device may have multiple IP addresses (IP4/IP6, Ethernet/WiFi), and actual address that will be used for communication, is determined only when device is opened, not at discovery time. IP6 addresses are long and look especially beautiful :-)

alexpevzner, 2020-11-12:
P.S. I'm sorry, I want to ask just out of curiosity, what large company are you working for? BMW?

motorBMW, 2020-11-12:
> P.S. I'm sorry, I want to ask just out of curiosity, what large company are you working for? BMW?

No, I don't work for BMW, I work for a financial organization.

> You probably have DNS-SD (Avahi) disabled, aren't you?

I tried enabling DNS-SD and got a relatively positive result. The names have really become unique, which allows you to identify devices. The negative result is that I now need to add each unique name to the exceptions so that the device doesn't appear in the list.
[image]

alexpevzner, 2020-11-12:
I remember you were asking for blacklisting facilities. I was thinking some time about it and can probably add something like this:
```
[blacklist]
model = "Brother*"
name = "*LaserJet*"
ip = 192.168.0.1/24
```

motorBMW, 2020-11-12:
> I remember you were asking for blacklisting facilities.

Yes, that was also my request. Of course, the extension of the blacklist parameters will allow to configure the use of the backend with different devices more flexibly.

Alexander, what do you think about still adding the device options the ability to view the IP address (ipv4)? As I understand it, the backend knows these parameters. If run airscan-discover, it clearly shows the device address. Is it possible to display the data similar to the hpaio or smfp? By the way, backends from manufacturers also return information if the devices are connected via USB. Perhaps this option will also help someone.

[image]

The bottom line is that the final call of scanners is made from the terminal server. The user can select a device available next to it in the interface. User looking at the built-in scanner interface (MFP) and see the ipv4 address there.
This is how the user sees the list of scanners:
[image]

alexpevzner, 2020-11-14:
Hi,

I've exposed scanner IP addresses to UI. If you can build by yourself, you can try it now

motorBMW, 2020-11-16:
Hi, Alexander

> I've exposed scanner IP addresses to UI. If you can build by yourself, you can try it now

Thank you very much for your work. Everything works flawlessly! I have two small questions for you.

1. Will your changes be added in the next release?
2. Do you have any idea where this device address for Xerox 3225 might come from?
It is not available locally, and there are no such addresses in the network either.

[image]

alexpevzner, 2020-11-16:
1. Yes. I'm currently trying to get Canon i-SENSUS MF411dw work properly. As soon as I'll succeed, I'll create a new release

2. No, I have no idea. Scanner announces it for some reason. It can be an address of the WiFi interface, for example.

motorBMW, 2020-11-17:
Alexander, 

I tried to understand the code and move the output of the ip address to another place in the string, but without success. Could you output the ip address as in my example? Usually in 'SANE-Device' structure the field 'name' is made alike 'airscan:wsd:10.22.33.44:Xerox' or 'airscan:wsd:Xerox?ip=10.22.33.44'.

Можем ли мы прейти на русский?))

alexpevzner, 2020-11-17:
Можем и на русский :-)

В поле `name` положить такую строку не получится. `name` - это уникальный идентификатор устройства. У меня идентификатором устройства является его DNS-SD name, а IP-адресов у устройства может быть несколько, и их набор может динамически меняться.

Более-менее произвольно назначаются поля `vendor, model и type`.

motorBMW, 2020-11-17:
Я обратил внимание, что у вас присваивается порядковый номер e0, w6 и так далее. Возможно ли после него вставить?
device `airscan:**e0**:Xerox VersaLink B405 (69:7c:8c)' is a eSCL Xerox VersaLink B405 (69:7c:8c) eSCL ip=10.34.31.50

Например вот так:
device `airscan:e0:ip=10.34.31.50:Xerox VersaLink B405 (69:7c:8c)' is a eSCL Xerox VersaLink B405 (69:7c:8c) eSCL

или вот так
device `airscan:e0:Xerox VersaLink B405 (69:7c:8c)?ip=10.34.31.50' is a eSCL Xerox VersaLink B405 (69:7c:8c) eSCL

alexpevzner, 2020-11-17:
Порядковый номер я приписываю потому, что существует некоторая вероятность, пусть и маленькая,  что в двух независимых локальный сетях, к торорым может быть подключена машина, найдутся два разных устройства с одинаковым именем (в одной сети это невозможно, потому, что устройства сами следят, чтобы их имена не совпадали).

Если я впишу адреса в имя устройства, найдутся пользователи, которые будут недоумевать, почему они попытались открыть устройство, передав IP-адрес в имени, а ничего не работает.

Тут есть существенное отличие между моим драйвером и драйвером от того же Xerox'а. Если Xerox имеет два адреса, и оба они видны с компьютера, то в списке устройств Xerox будет упомянут 2 раза. А у меня - только один раз, сколько бы у него адресов и вариантов подключения не было (их, на самом деле, может быть много: WiFi/Ethernet, HTTP/HTTPS, eSCL/WSD и т.д.). Поэтому у Xerox'а действительно точка подключения закодирована в имени, а у меня - нет.

В общем, мне не хотелось бы идти этим путем, чтобы избежать таких вопросов.

Так же нет смысла подстраиваться под то, чтобы все красиво выглядело с каким-то определенным UI, потому что в другой программе список будет выглядеть по-другому

motorBMW, 2020-11-17:
> Так же нет смысла подстраиваться под то, чтобы все красиво выглядело с каким-то определенным UI, потому что в другой программе список будет выглядеть по-другому

Александр, тут я с вами абсолютно согласен и прекрасно понимаю ваш ход мыслей.

У меня к вам одна просьба, немного изменить ваш код, что бы появился маркер "ip=". Я вчера протестировал и получил вот такой результат:
[image]

Строки которые я исправил, возможно не так деликатно :-) , но тем не менее:
[image]

Еще раз огромное спасибо за проделанную работу! Ваш дайвер решает большое количество проблем с совместимостью устройств. Могу ли я рассчитывать на внедрение изменения?

alexpevzner, 2020-11-17:
Это я могу, да (попробуйте с последним коммитом).

alexpevzner, 2020-11-19:
ping

дело движется к релизу. Хотелось бы, если можно получить обратную связь по поводу последних изменений,

motorBMW, 2020-11-19:
Все работает с вашими изменениями, огромное спасибо! Александр, будет ли реализована фильтрация устройств по производителю или модели в этом релизе? Или мне надо открыть еще одну проблему?

[image]

alexpevzner, 2020-11-20:
Все-таки в следующем, для этого уже достаточно накопилось изменений

alexpevzner, 2020-12-17:
Fixed

closed 2020-12-17 by alexpevzner