alexpevzner/sane-airscan #98: Canon i-SENSYS MF411dw. The driver is not working.

aleksey1831981, 2020-11-12:
This is the continuation of the story
https://gitlab.com/sane-project/backends/-/issues/378

Just in case, I'll describe the problem here.

So. I have:
Multifunction Printer Canon i-SENSYS MF411dw.
The printer is connected via a local network.
OS - Linux Mint 20 Ulyana Cinnamon. All updates have been installed.
Frontend - SimpleScan.

With the SANE driver, I was able to set up scanning for the flatbed scanner only.

As recommended by @alexpevzner, I installed sane-airscan.

I did as indicated here:
https://software.opensuse.org//download.html?project=home%3Apzz&package=sane-airscan

`echo 'deb http://download.opensuse.org/repositories/home:/pzz/xUbuntu_20.04/ /' | sudo tee /etc/apt/sources.list.d/home:pzz.list`
`curl -fsSL https://download.opensuse.org/repositories/home:pzz/xUbuntu_20.04/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/home_pzz.gpg> / dev / null`
`sudo apt update`
`sudo apt install sane-airscan`

Rebooted the computer.

Also, in the device settings:
[3]: by default, WS-Scan is disabled on this device and needs to be enabled before use: open web console, Click [Settings / Registration], Click [Network Settings] -> [TCP / IP Settings], Click [ Edit] in [WSD Settings], enable [Use WSD Scanning] checkbox, Click [OK]

Rebooted the device.

After these steps, I began to see more devices.
Some devices came to be identified as WSD.

My scanner (and several other scanners) was identified twice in the list at the same time.
In one case, it is identified as `ESCL - flatbed scanner`. (with the SANE driver, I only saw this entry.)
In the second case, `eSCL  - network scanner`.
I also stopped seeing the IP addresses of the scanners.

I chose eSCL - network scanner
I have tried the automatic feed tray. Also a flatbed scanner. One-sided and two-sided scanning. But nothing happens when scanning.

I attach the output of `simple-scan --debug`:  

- user@pc-138:~$ simple-scan --debug
[+0,00s] DEBUG: simple-scan.vala:1720: Starting simple-scan 3.36.3, PID=1472
[+0,00s] DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
[+0,03s] DEBUG: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
[+0,08s] DEBUG: app-window.vala:1906: Loading state from /home/user/.cache/simple-scan/state
[+0,08s] DEBUG: app-window.vala:1885: Restoring window to 600x400 pixels
[+0,15s] DEBUG: scanner.vala:1539: sane_init () -> SANE_STATUS_GOOD
[+0,15s] DEBUG: scanner.vala:1545: SANE version 1.0.29
[+0,15s] DEBUG: scanner.vala:1606: Requesting redetection of scan devices
[+0,15s] DEBUG: scanner.vala:828: Processing request
[18:11:27.994931] [bjnp] add_scanner: Scanner imageRUNNER1133 series is not supported, model is unknown! Please report upstream
[+0,31s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+2,34s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+9,97s] DEBUG: scanner.vala:341: sane_get_devices () -> SANE_STATUS_GOOD
[+9,97s] DEBUG: scanner.vala:353: Device: name="escl:http://192.168.11.149:80" vendor="ESCL" model="Canon MF410 Series" type="flatbed scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="escl:http://192.168.11.179:80" vendor="ESCL" model="Canon MF410 Series (e5:98:e7)" type="flatbed scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="escl:http://192.168.11.142:80" vendor="ESCL" model="Canon MF410 Series (e7:93:f3)" type="flatbed scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:e0:Canon MF410 Series" vendor="eSCL" model="Canon MF410 Series" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:e1:Canon MF410 Series (e5:98:e7)" vendor="eSCL" model="Canon MF410 Series (e5:98:e7)" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:e2:Canon MF410 Series (e7:93:f3)" vendor="eSCL" model="Canon MF410 Series (e7:93:f3)" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:ef:Canon MF420 Series (72:ed:43)" vendor="eSCL" model="Canon MF420 Series (72:ed:43)" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:e3:Canon MF440 Series" vendor="eSCL" model="Canon MF440 Series" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:e5:Canon MF440 Series (78:6d:61)" vendor="eSCL" model="Canon MF440 Series (78:6d:61)" type="eSCL network scanner"
[+9,97s] DEBUG: scanner.vala:353: Device: name="airscan:w14:HP LaserJet 400 MFP M425dn" vendor="WSD" model="HP LaserJet 400 MFP M425dn" type="WSD network scanner"
[+10,28s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+74,02s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+77,17s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[... 44 lines cut ...]
[+110,18s] DEBUG: scanner.vala:571: sane_control_option (3, SANE_ACTION_SET_VALUE, "Color") -> (SANE_STATUS_GOOD, "Color")
[+110,18s] DEBUG: scanner.vala:483: sane_control_option (2, SANE_ACTION_SET_VALUE, 300) -> (SANE_STATUS_GOOD, 300)
[+110,18s] DEBUG: scanner.vala:553: sane_control_option (8, SANE_ACTION_SET_VALUE, option.range.max=215,899994) -> (SANE_STATUS_GOOD)
[+110,18s] DEBUG: scanner.vala:553: sane_control_option (9, SANE_ACTION_SET_VALUE, option.range.max=355,599991) -> (SANE_STATUS_GOOD)
[+110,52s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+115,29s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+117,98s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+122,66s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+126,45s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+127,27s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state
[+140,21s] DEBUG: scanner.vala:1285: sane_start (page=0, pass=0) -> SANE_STATUS_IO_ERROR
[+140,21s] WARNING: scanner.vala:1302: Unable to start device: Error during device I/O
[+140,21s] DEBUG: scanner.vala:790: sane_cancel ()
[+140,21s] DEBUG: scanner.vala:793: sane_close ()
[+140,61s] DEBUG: app-window.vala:1981: Saving state to /home/user/.cache/simple-scan/state

This is the content of the file /home/user/.cache/simple-scan/state:

[window]
width=600
height=400
is-maximized=false
is-fullscreen=false
[scanner]
scan-type=adf-both

alexpevzner, 2020-11-12:
Hi!

I need a protocol trace. Please, uncomment the following lines in the `/etc/sane.d/airscan.conf` file:
```
[debug]
trace = ~/airscan/trace
enable = true
```
The traces will be in the `~/airscan/trace` directory. `.tar` files are not needed for now, please attach only `.log` files.

aleksey1831981, 2020-11-12:
OK. Access to a work computer will be available only tomorrow.
I will do it.

aleksey1831981, 2020-11-13:
Hi Alexander.

I have attached the logs. I also inform the commands scanimage -L and airscan-discover

> $ airscan-discover
[devices]
Canon LBP214 = http://192.168.11.164/active/msu/scan, WSD
Canon LBP252 = http://192.168.11.155/active/msu/scan, WSD
Canon MF410 Series (e5:98:e7) = http://192.168.11.179:80/eSCL/, eSCL
Canon MF410 Series (e5:98:e7) = http://192.168.11.179/active/msu/scan, WSD
Canon MF410 Series (e7:93:f3) = http://192.168.11.142:80/eSCL/, eSCL
Canon MF410 Series (e7:93:f3) = http://192.168.11.142/active/msu/scan, WSD
Canon MF410 Series (e8:05:91) = http://192.168.11.158:80/eSCL/, eSCL
Canon MF410 Series (e8:05:91) = http://192.168.11.158/active/msu/scan, WSD
Canon MF420 Series = http://192.168.11.163:80/eSCL/, eSCL
Canon MF420 Series (72:ed:43) = http://192.168.11.146:80/eSCL/, eSCL
Canon MF440 Series = http://192.168.11.141:80/eSCL/, eSCL
Canon MF440 Series (78:6d:61) = http://192.168.11.140:80/eSCL/, eSCL
Supply Canon MF411 = http://192.168.11.149:80/eSCL/, eSCL
Supply Canon MF411 = http://192.168.11.149/active/msu/scan, WSD
Supply Canon MF411 = http://192.168.11.149:80/wsd/scan, WSD

> $ scanimage -L
[12:13:09.529706] [bjnp] add_scanner: Scanner imageRUNNER1133 series is not supported, model is unknown! Please report upstream
[12:13:09.536951] [bjnp] add_scanner: Scanner imageRUNNER1133 series is not supported, model is unknown! Please report upstream
[12:13:09.542472] [bjnp] add_scanner: Scanner imageRUNNER1133 series is not supported, model is unknown! Please report upstream
[12:13:09.547436] [bjnp] add_scanner: Scanner imageRUNNER1133 series is not supported, model is unknown! Please report upstream
[12:13:09.552363] [bjnp] add_scanner: Scanner Canon iR2002/2202 is not supported, model is unknown! Please report upstream
device `escl:http://192.168.11.142:80' is a ESCL Canon MF410 Series (e7:93:f3) flatbed scanner
device `escl:http://192.168.11.179:80' is a ESCL Canon MF410 Series (e5:98:e7) flatbed scanner
device `escl:http://192.168.11.158:80' is a ESCL Canon MF410 Series (e8:05:91) flatbed scanner
device `escl:http://192.168.11.149:80' is a ESCL Supply Canon MF411 flatbed scanner
device `escl:http://192.168.11.163:80' is a ESCL Canon MF420 Series flatbed scanner
device `escl:http://192.168.11.140:80' is a ESCL Canon MF440 Series (78:6d:61) flatbed scanner
device `escl:http://192.168.11.141:80' is a ESCL Canon MF440 Series flatbed scanner
device `escl:http://192.168.11.146:80' is a ESCL Canon MF420 Series (72:ed:43) flatbed scanner
device `airscan:w1f:Canon LBP214' is a WSD Canon LBP214 WSD network scanner
device `airscan:w1c:Canon LBP252' is a WSD Canon LBP252 WSD network scanner
device `airscan:e6:Canon MF410 Series (e5:98:e7)' is a eSCL Canon MF410 Series (e5:98:e7) eSCL network scanner
device `airscan:e7:Canon MF410 Series (e7:93:f3)' is a eSCL Canon MF410 Series (e7:93:f3) eSCL network scanner
device `airscan:e5:Canon MF410 Series (e8:05:91)' is a eSCL Canon MF410 Series (e8:05:91) eSCL network scanner
device `airscan:e3:Canon MF420 Series' is a eSCL Canon MF420 Series eSCL network scanner
device `airscan:e0:Canon MF420 Series (72:ed:43)' is a eSCL Canon MF420 Series (72:ed:43) eSCL network scanner
device `airscan:e1:Canon MF440 Series' is a eSCL Canon MF440 Series eSCL network scanner
device `airscan:e2:Canon MF440 Series (78:6d:61)' is a eSCL Canon MF440 Series (78:6d:61) eSCL network scanner
device `airscan:e4:Supply Canon MF411' is a eSCL Supply Canon MF411 eSCL network scanner

[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5535989/simple-scan-zeroconf.log)
[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5535990/simple-scan-Supply-Canon-MF411.log)

aleksey1831981, 2020-11-13:
We're talking about the Supply Canon MF411 scanner.
We configure this scanner.
After all the changes, this scanner is not recognized as a WSD scanner for some reason.

alexpevzner, 2020-11-13:
Предлагаю перейти на русский :-)

> After all the changes, simple scan this scanner is not recognized as a WSD scanner for some reason.

Он, как выяснилось, eSCL тоже умеет, а если сканер умеет оба протокола, sane-airscan предпочитает eSCL.

А теперь следующий вопрос. При попытке сканирования, со сканером вообще ничего не происходит, или он начинает жужжать, но картинку не отдает?

aleksey1831981, 2020-11-13:
Не происходит ничего. Не жужжит. Не пищит. Не щелкает. Картинки нет. Просто тишина. Через 10-15 секунд появляется сообщение об ошибке.

alexpevzner, 2020-11-13:
Я вижу по логам, что драйвер делает HTTP POST, чтобы запустить сканирование, а устройство просто не отвечает. Через 30 секунд запрос отваливается по таймауту.

А вы сможете драйвер самостоятельно пересобрать из исходников с целью экспериментов?

aleksey1831981, 2020-11-13:
Думаю, что смогу.

alexpevzner, 2020-11-13:
Хорошо. Его можно даже не переставлять, а сделать симлинк /usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1 -> libsane-airscan.so.1 в сборочной директории.

Тогда я попрошу вас в файле `airscan-escl.c`, начиная со строки 646, найти такой код:
```
    if (!escl->oki) {
        http_query_set_request_header(query, "Host", "localhost");
        http_query_onredir(query, escl_scan_fix_location);
    }
```

И две строки в фигурных скобках закоментировать, потом пересобрать и попробовать еще раз.

aleksey1831981, 2020-11-13:
Вы дадите ссылку на скачивание исходников?

alexpevzner, 2020-11-13:
git clone https://github.com/alexpevzner/sane-airscan.git

aleksey1831981, 2020-11-13:
Это работает.  
Автоподатчик работает. Двухстороннее сканирование работает. Планшетный сканер работает. 
Но, в авторежиме, в simple scan, область сканирования несколько больше.
Позже скину сканы.
Ещё, если запустить сканирование из лотка автоподачи, сканер выдает ошибку, что замята бумага.

alexpevzner, 2020-11-13:
> Это работает.

OK, значит будет в следующем релизе, который я выпущу на днях

> Но, в авторежиме, в simple scan, область сканирования несколько больше.

Да, я по умолчанию просто ставлю максимум, что сканер может. Программа (simple-scan, в данном случая) должна выбирать правильный размер. с учетом возможностей устройства и предпочтений пользователя. Мне кажется, simple-scan это даже умеет.

> Ещё, если запустить сканирование из лотка автоподачи, сканер выдает ошибку, что замята бумага.

С этим хотелось бы разобраться, нужны логи.

aleksey1831981, 2020-11-13:
> С этим хотелось бы разобраться, нужны логи.

Извиняюсь за путаницу.  Правильно будет так:
Ещё, если запустить сканирование из лотка автоподачи **без бумаги**, сканер выдает ошибку, что замята бумага.

aleksey1831981, 2020-11-13:
Прикрепил результаты сканирования.

[А4.pdf](https://github.com/alexpevzner/sane-airscan/files/5536719/4.pdf)
[Автоформат.pdf](https://github.com/alexpevzner/sane-airscan/files/5536720/default.pdf)

aleksey1831981, 2020-11-13:
Прикрепил логи при сканировании с пустым лотком автоподачи.
[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5536753/simple-scan-Supply-Canon-MF411.log)
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5536754/simple-scan-zeroconf.log)

aleksey1831981, 2020-11-13:
Александр, есть еще пожелание на отдельные функции, если, конечно, это входит в рамки реализации этого пакета.
А именно:
1. Сделать возможность указать адрес сканера (статический) где нибудь в  настройках. Или хранить список ранее найденных сканеров в неком кэше, до перезагрузки компьютера. Потому что при запуске программы сканирования, поиск сканеров происходит достаточно длительное время.
2. Сделать возможность исключить из результатов поиска ненужные сканеры. Потому как список получается достаточно длинным. И я представляю ситуацию, когда сканеры в разных кабинетах начинают сканировать сами по себе, потому что кто то выбрал не тот сканер.

Если это необходимо, я оформлю новый вопрос.

Прикрепляю картинки того, как сейчас у меня выглядит меню выбора сканера.
[image]
[image]

alexpevzner, 2020-11-13:
> Прикрепил логи при сканировании с пустым лотком автоподачи.

Тут хотелось бы получить еще и `.tar`-файлы. Поскольку устройство не проставляет в ответе Content-Type, трассировщик не понимает, что получил XML, и вместо того, чтобы красиво распечатать его в лог, кладет его, как есть, в `.tar`

> есть еще пожелание на отдельные функции

Это все есть. Надо прописать устройства в `/etc/sane/airscan.conf` в секцию `[devices]` в том формате, в котором их `airscan-discover` выдает (но без повторов по именам), и в секции `[options]` выключить `discovery`

В самом конфиге написано вкратце, как им пользоваться

aleksey1831981, 2020-11-13:
Тут не поддерживается этот формат файлов.
Куда их отправить?

aleksey1831981, 2020-11-13:
> /etc/sane/airscan.conf

Нет такого пути.
Есть `/etc/sane.d/airscan.conf`
Это подойдет?

alexpevzner, 2020-11-13:
> Куда их отправить?

pzz@apevzner.com

Не уверен, что туда пролезет .tar, но если зазиповать, то точно пролезет

> Есть /etc/sane.d/airscan.conf

Да-да, я просто опечатался.

aleksey1831981, 2020-11-13:
Готово.
зип тут проходит.

[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5537306/simple-scan-Supply-Canon-MF411.log)
[simple-scan-Supply-Canon-MF411.zip](https://github.com/alexpevzner/sane-airscan/files/5537307/simple-scan-Supply-Canon-MF411.zip)
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5537308/simple-scan-zeroconf.log)
[simple-scan-zeroconf.zip](https://github.com/alexpevzner/sane-airscan/files/5537309/simple-scan-zeroconf.zip)

alexpevzner, 2020-11-13:
Спасибо.

`<scan:AdfState>ScannerAdfJam</scan:AdfState>`

Боюсь, с этим я ничего сделать не могу, ошибка firmware, сканер говорит мне, что у него замятие в ADF, и нет никаких признаков, по которым я мог бы догадаться, что это не так.

aleksey1831981, 2020-11-13:
В понедельник, я попрошу у коллег запустить сканирование с пустым лотком  с виндозного компьютера. Посмотрим как будет вести себя сканер.
Александр, может вы подскажете, где у SANE отключается поиск сканеров? Или может мне удалить SANE совсем, в таком случае?

alexpevzner, 2020-11-13:
>В понедельник, я попрошу у коллег запустить сканирование с пустым лотком с виндозного компьютера

Windows будет использовать WSD, или кенноновский драйвер. С WSD вы можете и sane-airscan попробовать, если aiscan-discover видит это устройство в WSD-режиме. Можно либо его WSD-инкарнацию вручную добавить в список устройств, либо в секции `[options]` прописать `protocol = manual`

>Александр, может вы подскажете, где у SANE отключается поиск сканеров?

Надо в `/etc/sane.d/dll.conf` закоментировать все лишние драйвера (можно вообще все, если кроме sane-airscan никакие другие не используются)

aleksey1831981, 2020-11-13:
> Можно либо его WSD-инкарнацию вручную добавить в список устройств

Это попробую. Отпишусь, какие будут ощущения.

> Надо в /etc/sane.d/dll.conf закоментировать все лишние драйвера

Понял. Спасибо.  В понедельник попробуем.

И последний вопрос. Сейчас сделан симлинк на файл   `libsane-airscan.so.1` из сборки. 
После того как придет обновление, нужно будет в ручном режиме удалять симлинк? Или все произойдет автоматом?

alexpevzner, 2020-11-13:
> Это попробую. Отпишусь о результатах.

Хорошо.

> После того как придет обновление, нужно будет в ручном режиме удалять симлинк? Или все произойдет автоматом?

Не знаю, если честно. Скорее всего, само произойдет.

aleksey1831981, 2020-11-16:
Итак.
Я попробовал сканировать WSD с пустым лотком автоподачи.
Я указал в `/etc/sane.d/airscan.conf`:
`Supply Canon MF411 = http://192.168.11.149:80/wsd/scan, WSD`
В самом начале сканер жу-жу и после этого тишина. Специфичного продолжительного жужжания, как во время сканирования, не было.
Если с бумагой сканировать, то все ОК.

SimleScan никаких ошибок не выдает.  Просто висит статус: "Сканирование".
На самом сканере, на жк дисплее, отображается информационное сообщение:
**_Удаленный сканер.
Ожидание... (Подключено)
Начать сканирование с использованием устройства в месте сохранения._**

Логи прикрепил. Может что то будет.
[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5545135/simple-scan-Supply-Canon-MF411.log)
[simple-scan-Supply-Canon-MF411.zip](https://github.com/alexpevzner/sane-airscan/files/5545136/simple-scan-Supply-Canon-MF411.zip)
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5545137/simple-scan-zeroconf.log)
[simple-scan-zeroconf.zip](https://github.com/alexpevzner/sane-airscan/files/5545138/simple-scan-zeroconf.zip)

Также я пробовал сканировать с пустым лотком автоподачи.
`Supply Canon MF411 = http://192.168.11.149/active/msu/scan, WSD`
Тут сканирование даже не запускается. SimleScan пишет "Не удалось подключиться к сканеру".
Вероятно это порт для беспроводного сканирования.

ИЗ Windows компьютера если начать сканировать с пустым лотком автоподачи, На мониторе пишет: "Сканеру не удалось обнаружить документы. Проверьте лоток подачи." Сканер при этом никаких звуков не издает. Сканировать даже не начинает.

Когда вкладываешь бумагу в лоток подачи, обчно сканер пищит однократно, давая понять, что он бумагу увидел. Предполагаю, что информацию о наличии бумаги сканер не передает корректно. Или мы  не можем это корректно прочитать.

По моему мнению, в режиме WSD ошибка об отсутствии бумаги в лотке подачи обрабатывается более корректно. 
Т.е. в eSCL устройство зависает, и последующие задания (в том числе с других компьютеров)  на печать или сканирование, не обрабатывает. Приходится перезагружать.
В WSD, просто ничего не происходит. Если завершить сканирование в SimpleScan, устройство остается в рабочем состоянии. Все дальнейшие задания обрабатываются корректно.

Еще я провел сравнительный тест на скорость сканирования в режимах WSD и eSCL.
WSD - 12 листов за 69 сек.
eSCL - 12 листов за 73 сек.

aleksey1831981, 2020-11-16:
Еще я закомментировал все строки в `/etc/sane.d/dll.conf`.
Теперь я не вижу никаких сканеров, кроме нужного мне. И запуск программы происходит достаточно быстро.

alexpevzner, 2020-11-16:
Про WSD. При сканировании из пустого лотка мне сканер возвращает `ServerErrorNotAcceptingJobs`. Это неспецифическая ошибка, сканер может возвращать такую в разных случаях. Например, моя Kyocera в WSD сканирует только, если перед запуском сканирования нажать соответствующие кнопки на передней панели, а иначе возвращает эту ошибку (а в eSCL сканирует без всяких кнопок).

Я эту ошибку превращаю в `SANE_STATUS_DEVICE_BUSY`. Далее, simple-scan, получив `SANE_STATUS_DEVICE_BUSY`, делает в цикле повторные попытки сканирования, пока ему не нажмут кнопку Stop или он чего-нибудь не добъется. Другие клиенты так не делают, xsane выдаст окошко с ошибкой сразу.

Интересно понять, как Windows догадывается, что дело в пустом лотке. Вы можете с помощью Wireshark поймать трафик между компьютером под Windows и сканером, или это слишком сложно?

aleksey1831981, 2020-11-16:
Я не являюсь системным администратором. У меня есть отступ только к Линукс компьютеру. 
Нужны инструкции, которые я мог бы передать админки, с просьбой сделать это.

alexpevzner, 2020-11-16:
Хорошо, тогда пойдем другим путем. Я там поковырял немного в драйвере, пожалуйста, обновитесь, пересоберите, попробуйте еще раз WSD-сканирование из пустого лотка, и пришлите мне логи.

Да, для WSD можно с .tar-ом не заморачиваться, для WSD сканер нормально прописывает Content-Type, и все, что надо, попадает в лог.

aleksey1831981, 2020-11-16:
> обновитесь, пересоберите

Я должен удалить текущий air-sane. Затем пересобрать и установить. Все верно? Из репо обновление делать нет смысла?

alexpevzner, 2020-11-16:
Почему? Просто git pull, затем make

Если у вас установка заключалась в прописывании симлинка, то этого вообще достаточно

alexpevzner, 2020-11-16:
Windows, в отличии от меня, посылает сначала запрос ValidateScanTicketRequest (проверка запроса на сканирование), а потом уже CreateScanJobRequest. Я никогда не понимал, зачем они это делают, ведь если есть какие-то проблемы с запросом, то CreateScanJobRequest сообщит о них не хуже, чем ValidateScanTicketRequest, а если с запросом все хорошо, то непонятно, что добавляет этот лишний шаг.

Но возможно в некоторых случаях (например, как с вашим устройством) ValidateScanTicketRequest вернет более конкретный код ошибки, чем ServerErrorNotAcceptingJobs.

Я временно сделал посылку ValidateScanTicketRequest перед CreateScanJobRequest. Если это поможет, то я немного окультурю и все так и оставлю, а если нет, буду думать, что еще можно сделать.

aleksey1831981, 2020-11-16:
Мой рабочий компьютер отказывается работать со мной. Какие то проблемы с железом. Сейчас он в ремонте у наших IT. Как только мне вернут его в рабочем состоянии, сообщу о результатах тестов и выложу логи. Надеюсь это произойдет завтра до обеда.

alexpevzner, 2020-11-16:
Хорошо

aleksey1831981, 2020-11-17:
Готово.

Правда есть сомнения, что все корректно обновилось из исходников.

`git pul https://github.com/alexpevzner/sane-airscan.git` не сработал. Полезли какие то ошибки.
В итоге просто  удалил все файлы из директории `sane-airscan`.
Затем выполнил
```
git clone https://github.com/alexpevzner/sane-airscan.git
cd sane-airscan
make
make install
```

После этого сканирование с пустым лотком.

Логи прилагаются.
[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5552841/simple-scan-Supply-Canon-MF411.log)
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5552844/simple-scan-zeroconf.log)

aleksey1831981, 2020-11-17:
Александр, удалось получить необходимую информацию из этих логов? Или нужно что то еще с моей стороны?

alexpevzner, 2020-11-17:
Да, спасибо. Я понял, что та идея не сработала. Завтра, наверное, приду с просьбой проверить следующую идею, сегодня уже не успею.

aleksey1831981, 2020-11-17:
Хорошо.

alexpevzner, 2020-11-18:
Пожалуйста, обновитесь и попробуйте. Пустой ADF в WSD режиме должен теперь корректно определяться

aleksey1831981, 2020-11-18:
Пишет: 
Сканирование не удалось. Лоток подачи документов пуст.

alexpevzner, 2020-11-18:
Ну т.е., то, что надо?

Теперь хотелось бы с eSCL тоже разобраться. Я правильно понимаю, что в режиме eSCL при сканировании из пустого лотка есть две проблемы:
1. Неправильное сообщение об ошибке
2. Сканер после такого сканирования зависает, и без перезагрузки в себя не приходит?

aleksey1831981, 2020-11-18:
> Ну т.е., то, что надо?

Да. Идеально работает.
В чем секрет?
`ServerErrorNotAcceptingJobs` теперь обрабатывается как "Лоток подачи документов пуст"?

> Я правильно понимаю, что в режиме eSCL при сканировании из пустого лотка есть две проблемы:
> 
> Неправильное сообщение об ошибке
> Сканер после такого сканирования зависает, и без перезагрузки в себя не приходит?

Да, все верно.
После последнего обновления в eSCL сканировать не пробовал.

alexpevzner, 2020-11-18:
> ServerErrorNotAcceptingJobs теперь обрабатывается как "Лоток подачи документов пуст"?

Не всегда, а если в отдельно запрошеном scanner status'е ScannerState == Idle.

> Да, все верно.

А он совсем зависает, или через минуту-другую приходит в себя?

aleksey1831981, 2020-11-18:
Сейчас еще раз попробовал сканировать с пустым лотком. Но, после некоторого простоя. 
Т.е. устройство, видимо переходит в спящий режим.

Итак , при сканировании, сканер сначала сделал жу-жу (простите меня за такой оборот речи).  И потом сообщение об ошибке.

SimpleScan после простоя. 
Сканер сначала сделал жу-жу (простите меня за такой оборот речи). Затем:
"Сканирование не удалось. Не удалось начать сканирование."

gscan2pdf после простоя.
Сканер сначала сделал жу-жу. Затем:
gscan2pdf: sane_start: Scanner cover is open

XSane после простоя.
Сканер сначала сделал жу-жу. Затем:
Невозможно инициализировать сканирование. Крышка сканера открыта.

Windows компьютер после простоя сразу выдает сообщение, что нет бумаги. Причем жу-жу не делает.

А вот если сканировать, пока устройство не спит.

SimpleScan
Сканирование не удалось. Лоток подачи документов пуст.

gscan2pdf 
сообщений вообще никаких нет.

XSane
Невозможно инициализировать сканирование. Автоподатчик документов не может найти бумагу.

Логи на сканирование из спящего режима сделаю чуть позже.

aleksey1831981, 2020-11-18:
Логи при сканирование из спящего режима прикрепил.
[simple-scan-Supply-Canon-MF411.log](https://github.com/alexpevzner/sane-airscan/files/5559243/simple-scan-Supply-Canon-MF411.log)
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/5559244/simple-scan-zeroconf.log)

alexpevzner, 2020-11-18:
Попробуйте еще раз из спящего режима, должно полегчать

alexpevzner, 2020-11-18:
(1) Про WSD. Windows, видимо, запрашивает состояние сканера перед сканированием (чего я не делаю), что-то там видит, и на основании этого делает вывод, что нет бумаги, еще до начала сканирования. А я сразу инициирую сканирование, в результате чего сканер просышается и делает жу-жу.

Делать как Windows я не хочу, потому что по моему опыту это самое состояние сканера до попытки сканирования - вещь ненадежная, и разные сканеры его возвращают по-разному. А вот состояние после неудавшегося сканирования, когда я **уже** знаю, что какая-то ошибка произошла, оно гораздо надежнее.

Microsoft может себе это позволить, потому что они после каждого существенного изменения могут перетестировать все на сотне устройств, а я не могу.

Но жу-жу - проблема чисто косметическая, поэтому я не буду ее решать, чтобы не ухудшить поддержку тех сканеров, которые сейчас работают, а после соответствующих изменений могут и перестать.

Так что тему с WSD я считаю закрытой. Кстати, это первый сканер, с которым мне довелось иметь дело, у которого WSD работает лучше, чем eSCL, обычно наоборот.

(2) Про eSCL. Я там пока ничего не менял, кроме того самого раннего изменения, с которого мы начали. Так что если поведение изменилось, это может быть связано с тем, что после предыдущего WSD-сканирования у сканера в голове остались какие-то мысли.

Вообще, это очень похоже на ошибку firmware, похоже, сканер не считывает вовремя состояние датчика наличия бумаги в автоподатчике, а потом у него концы с концами не сходятся, и он решает, что бумага замялась.

С этим бы как-то хотелось разобраться, потому, что нехорошо оставлять сканер в состоянии, когда он нуждается в реанимационных мероприятиях.

Не хотите firmware проапдейтить?

aleksey1831981, 2020-11-18:
> Не хотите прошивку проапдейтить?

Если это модифицированная прошивка, то это вряд ли.
Если оригинальная, тогда нужно разговаривать с админами. Гипотетически возможно.

alexpevzner, 2020-11-18:
Конечно, оригинальная.

https://www.canon-europe.com/support/consumer_products/products/fax__multifunctionals/laser/laserbase_mf_series/i-sensys-mf411dw.html?type=firmware

Еще обещают какую-то уязвимость в WiFi заткнуть...

alexpevzner, 2020-11-18:
Можно еще попробовать посмотреть, воспроизводится ли ошибка с eSCL в VueScan (https://www.hamrick.com/linux-scanner-software.html). Он не бесплатный, но у него есть бесплатный preview.

Правда, есть некоторая вероятность, что VueScan с вашим сканером сговорится, используя Canon'овский протокол, а не eSCL. Но если вы попробуете, я вам скажу, где у него логи, по логам видно, что он делает.

Еще, если у вас андройдный смартфон, можно попробовать Mopria Scan (https://play.google.com/store/apps/details?id=org.mopria.scan.application&utm_source=www.apk4fun.com), он точно использует eSCL

aleksey1831981, 2020-11-18:
> Конечно, оригинальная.

Напишу письмо нашим IT. Думаю, ответ будет завтра.

> воспроизводится ли ошибка с eSCL в VueScan 

VueScan я пробовал еще до нашего с Вами общения. У меня VueScan вообще не увидел ни одного сканера. Может я что  то не так сделал. Сегодня попробую еще раз.
 

> если у вас андройдный смартфон, можно попробовать Mopria Scan 

С удовольствием попробую. Только надо опять админов просить, что бы включили WIFI на устройстве. Доступа в веб интерфейс у меня нет.

Сообщите, что для Вас предпочтительней. Туда и будем копать.

alexpevzner, 2020-11-18:
Ну, что из этого получится, то и стоит попробовать.

Если апгрейд firmware поможет, я просто задокументирую это в README.md, и буду считать проблему решенной. Особенно если со старой прошивкой проблема возпроизводится с VueScan/Mopria, а с новой - нет.

aleksey1831981, 2020-11-18:
Я понял. Попробуем все.
Сейчас пришлось уйти с работы не надолго. Думаю, часов в 17 по МСК доберусь до рабочего места.

alexpevzner, 2020-11-18:
Хорошо

aleksey1831981, 2020-11-18:
Я посмотрел. На европейском сайте доступны:

**MF416dw / MF411dw Firmware Update Tool V09.01 [Windows]
Версия файла: V09.01 Дата выхода: 09 ноября 2020 г.**

**Прошивка главного контроллера (Windows)
Версия файла: 6.0.1 Дата выпуска: 08 марта 2017 г.**
Желательно обе прошивки обновить?

На Российском сайте только 
**[Windows] Инструмент обновления встроенного ПО MF411dw/MF416dw (V08.06)
Версия файла: V08.06 Дата выпуска: 21 мая 2018**
Эту нежелательно, я понимаю?

alexpevzner, 2020-11-18:
Ну, 8 марта 2017 г, это какое-то совсем уж старье. Я бы поставил эту, которая от 9 ноября 2020 г.

aleksey1831981, 2020-11-18:
**Прошивка главного контроллера**

Я предположил, если есть главный контроллер, то есть и не главный ).
И что это дополнительно в куда нибудь заливается. Думал Вы знаете.

Если нет, тогда пусть будет только:
**MF416dw / MF411dw Firmware Update Tool V09.01 [Windows]
Версия файла: V09.01 Дата выхода: 09 ноября 2020 г.**

[32 more replies]

closed 2020-11-20 by alexpevzner