alexpevzner/sane-airscan #100: License of the fuzzer/* files?

OdyX, 2020-11-23:
Between 0.99.19 and 0.99.20, `fuzzer/*` files were added with this header:
```
// Copyright 2020 The Chromium OS Authors. All rights reserved.
// See LICENSE for license terms and conditions.
```

It's not clear whether a `fuzzer/LICENSE` was forgotten, or if that points to `./LICENSE`. Could you clarify under which License these files are shipped?

alexpevzner, 2020-11-23:
@yetamrra,

could you please look to this issue?

yetamrra, 2020-11-23:
That was meant to refer to the project top-level LICENSE.  I'll send a quick PR to clarify.

alexpevzner, 2020-11-23:
Thank you, @yetamrra,

@OdyX, I've created 0.99.21 release with the only difference in updated license information in the fuzzer files

PR #101 (2020-11-23): Clarify fuzzer license
The LICENSE file mentioned in fuzzer headers was meant to refer to the
main sane-airscan license, not a potentially separate fuzzer license.

Fixes #100.
commit 0e97af03 (2020-11-23): Clarify fuzzer license

The LICENSE file mentioned in fuzzer headers was meant to refer to the
main sane-airscan license, not a potentially separate fuzzer license.

Fixes #100.
closed 2020-11-23 by alexpevzner