alexpevzner/sane-airscan #102: Add supported resolutions

gaubitzer, 2020-11-26:
I have an Epson Workforce WF-7710 and using xsane/sane-airscan I can only choose between 100 and 300 dpi, both work. Directly on the scanner I can choose between 200, 300 and 600 dpi. How are sane resolutions determined (I saw from the log that the backend get it directly from the scanner!?). Why is it different? Can resolutions be manually added?

Thanks for your great backend!!

[xsane-EPSON-WF-7710-Series.log](https://github.com/alexpevzner/sane-airscan/files/5601554/xsane-EPSON-WF-7710-Series.log)

alexpevzner, 2020-11-26:
Via wsd, device only offers 100 and 300:

```
              <wscn:PlatenResolutions>
                <wscn:Widths>
                  <wscn:Width>100</wscn:Width>
                  <wscn:Width>300</wscn:Width>
                </wscn:Widths>
                <wscn:Heights>
                  <wscn:Height>100</wscn:Height>
                  <wscn:Height>300</wscn:Height>
                </wscn:Heights>
              </wscn:PlatenResolutions>
```

And the same for ADF.

It sometimes happens, that device offers different set of resolutions, depending on how device is accessed.

alexpevzner, 2020-11-26:
What is interesting, this device is marked in my table as supporting eSCL, while your log contains WSD. I want to clarify it. May I ask you to show me output of the following 3 commands?

    airscan-discover
    avahi-browse -rt _uscan._tcp
    avahi-browse -rt _uscans._tcp

gaubitzer, 2020-11-26:
Here you go
```
erwin@workhorse:~ 11:01 $ airscan-discover
[devices]
  EPSON WF-7710 Series = http://192.168.2.114:80/WDP/SCAN, WSD
erwin@workhorse:~ 11:01 $ avahi-browse -rt _uscan._tcp
erwin@workhorse:~ 11:01 $ avahi-browse -rt _uscans._tcp
erwin@workhorse:~ 11:01 $
```

alexpevzner, 2020-12-17:
Closed; description updated

closed 2020-12-17 by alexpevzner

gaubitzer, 2022-01-12:
Dear Alex

Due to to recent inability to access my printer/scanner WF7710 I unchecked und checked "Enable Scanning" of "Airprint" in the "Network Scan" menu. After that even the eSCL capabilities were visible. Also all resolutions (100, 200, 300, 600 and 1200) were available. Somehow settings on the device were corrupted and restored by unsetting/setting. Thought this would be interesting for you.

```
erwin@workhorse:~ 15:03 $ airscan-discover 
[devices]
  EPSON WF-7710 Series = http://192.168.2.114:443/eSCL/, eSCL
  EPSON WF-7710 Series = https://192.168.2.114:443/eSCL/, eSCL
  EPSON WF-7710 Series = http://192.168.2.114:80/WDP/SCAN, WSD
```